arendst/Tasmota #6216: Alexa Discovery Problem ?

eastabrooka, 2019-08-13:
Hi All, 

I have this problem yesterday where I deleted all my saved devices in Alexa, and tried to re-add them. Alexa is now saying "Cannot find any new smart devices."
I was running some really old version of Tasmota and it had been working for about a year, with over 1000 reboots. So - In any case, I thought its probably time to pull myself up to the latest version and see if this fixes things. 

So - Here is my write up. 

1) I have flashed the latest tagged release ( Binary ) - No compiling myself. 
2) This was version 6.6.0(release-sonoff)
3) My HW is the Basic type. Nothing fancy. 1 Relay. 
4) I get the web page and everything works ok there. I can toggle etc.  
5) I have tried selecting both the Hue emulation and the Wemos emulation. 
        Alexa cannot find either. 

In continued troubleshooting, I have pulled down a copy of NodeRed and installed a Wemos emulatior node to that as well. 
And Alexa cannot find that either. 

**I am raising this issue to ask if one of the devs wouldnt mind deleting all their devices from Alexa, and asking it to rescan to see if Amazon has pulled the plug on Wemos without it being registered as a Skill ?** 

Additional points,  https://github.com/arendst/Sonoff-Tasmota/issues/596#issuecomment-314104100 
I have left it overnight - and it still wont show up. 

https://github.com/arendst/Sonoff-Tasmota/issues/596#issuecomment-357546354 I cannot change the wemos port in a compiled binary. 

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X ] Device used (i.e. Sonoff Basic) : Sonoff Basic - 1x Channel Relay. 
- [X ] Tasmota binary firmware version number used : Precompiled. 6.6.0 / (pre-compiled or self-compiled ?)
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` :

```
STATUS 0 OUTPUT HERE:

07:15:09 CMD: status 0
07:15:09 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Tube Light"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
07:15:09 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T09:01:06","StartupUTC":"2019-08-12T21:14:03","Sleep":50,"CfgHolder":4617,"BootCount":3,"SaveCount":11,"SaveAddress":"F9000"}}
07:15:09 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
07:15:09 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["**Redacted**",""],"TelePeriod":300,"Resolution":"55A180C0","SetOption":["00008009","280500000100060000000000000000000000","00000000"]}}
07:15:09 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashMode":2,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
07:15:09 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"Tube Light","IPAddress":"192.168.1.97","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"5C:CF:7F:E7:5A:CA","Webserver":2,"WifiConfig":4}}
07:15:09 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E75ACA","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
07:15:09 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Aug 13 06:15:09 2019","Local":"Tue Aug 13 07:15:09 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:40","Sunset":"20:10"}}
07:15:09 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-08-13T07:15:09"}}
07:15:09 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-08-13T07:15:09","Uptime":"0T09:01:06","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"**Redacted**","BSSId":"40:F2:01:41:A9:E4","Channel":11,"RSSI":72,"LinkCount":1,"Downtime":"0T00:00:08"}}}
```

Many thanks for anyone looking into this ticket. 

Alex

s-hadinger, 2019-08-13:
Can you please set log level to 4, trigger an Alexa discovery and paste here the logs collected?

eastabrooka, 2019-08-13:
Hi, Sure - I can do this tomorrow and append the logs to this ticket. Thanks.

eastabrooka, 2019-08-14:
[image]
[image]

Wow, Thats wierd. 
Last night I was scanning for devices with the light off. 
Tonight I am doing this test with the light on. 

It flicks off and on when i run the test. 

00:00:00 CFG: Loaded from flash at F4, Count 16
00:00:00 SRC: Restart
00:00:00 Project sonoff Tube Light Version 6.6.0(release-sonoff)-2_3_0
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 Dovenet in mode 11N as Tube Light...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Checking connection...
00:00:06 WIF: Attempting connection...
00:00:07 WIF: Checking connection...
00:00:07 WIF: Connected
00:00:07 HTP: Web server active on Tube Light with IP address 192.168.1.97
00:00:07 UPP: Multicast (re)joined
22:28:39 NTP: Drift 0, (UTC) Wed Aug 14 21:28:39 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
22:28:39 APP: Boot Count 8
22:28:39 CFG: Saved to flash at FB, Count 17, Bytes 3584

Its very much like when discovery comes out, it crashes on my sonoff. 
Any thoughts ?

eastabrooka, 2019-08-14:
I think the crash may be related to log level. When I turn the log level back down, it does not flicker again.

ascillato, 2019-08-14:
Please, try again but with latest Tasmota (6.6.0.4) and with core 2.5.2 :

http://thehackbox.org/tasmota/020502/sonoff.bin

meingraham, 2019-08-14:
You shouldn't need to elevate the log level on all logging channels. Unless you have a terminal connected to the serial interface, you won't see those messages. Syslog messages go to a remote syslog server. If you have that turned on, you're going to get the log details there... and it's redundant to set the log level high for the web console. Besides, Unless you have the Console open and are watching the Console, those messages are going to roll off... and even faster with logging levels set high. Also note that all of this is extra work the software is having to accomplish so it may be bogging down your device... perhaps even to the point of crashing it.

s-hadinger, 2019-08-15:
I cannot explain why it's flickering and crashing. For what we need here, weblog 4 should be enough. Please set all logs to 0 expcept WebLog to 4.

ascillato2, 2019-08-16:
@eastabrooka 

Hi, any news on this?

eastabrooka, 2019-08-16:
Hi @ascillato2  - Yes, I have an Update, 
    And the problem has been resolved for me. I will detail what steps I believe i can attiribute to the 'Fix' 

1) I am still running 6.6.0. 

2) I flashed this using the NodeMCU and did a full flash erase during this process. 
[image]

3) As I was used to a very old build 3.1? - Before updating - I was used to using the Wemos type emulation. 

4) I changed this to be <Type Hue> But this did not resolve my issue. 

5) This is the step I attribute to the fix - 
[image]

My speculation is - Maybe by default not all settings are initialised as 'Active' so by changing the template, and then back, it writes something into the SPI Flash, and then it works properly. 

Steps to take this forward: 

Would you like me to flash 6.6.0.4 - With a Full SPI Erase, and see if this is still a problem for me / the release, or would you like me to mark this ticket as closed ? 
Many thanks to all,  @s-hadinger @meingraham @ascillato

ascillato, 2019-08-16:
> Would you like me to flash 6.6.0.4 - With a Full SPI Erase

That would be great. If you can, please check also that. Thanks.

closed 2019-08-17 by ascillato2

eastabrooka, 2019-08-17:
I tried to repeat the issue. I could not replicate. 
Thanks so much everyone.