arendst/Tasmota #6217: Wakeup ignores SetOption20

Mr-iX, 2019-08-13:
When the Wakup is started and the CT value is changed while the wakeup is happening, the wakeup is seen as done before the wakeupduration has ended.

To reproduce:
set wakeupduration --> command wakeup --> command ct 300 --> wakeup is canceled and scheme is reset to 0

See attached output

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : H801
- [x] Tasmota binary firmware version number used : 6.6.0(release-sonoff) / (pre-compiled)
- [x] Development IDE - Compiler / Upload tools used : ____ / esptool
- [x] Provide the output of command ``status 0`` :
```
11:07:11 MQT: stat/h801_1/STATUS = {"Status":{"Module":0,"FriendlyName":["H801"],"Topic":"h801_1","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:07:11 MQT: stat/h801_1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Exception","Uptime":"1T00:19:09","StartupUTC":"2019-08-12T08:48:02","Sleep":50,"CfgHolder":4617,"BootCount":94,"SaveCount":2626,"SaveAddress":"F8000"}}
11:07:11 MQT: stat/h801_1/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
11:07:11 MQT: stat/h801_1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WIFI",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["80108009","280500000100060000000000000000000000","00000200"]}}
11:07:11 MQT: stat/h801_1/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440EF","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
11:07:11 MQT: stat/h801_1/STATUS5 = {"StatusNET":{"Hostname":"h801_1","IPAddress":"192.168.0.4","Gateway":"192.168.0.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"DE:ES:RS:23:34:64","Webserver":2,"WifiConfig":4}}
11:07:11 MQT: stat/h801_1/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.local","MqttPort":1883,"MqttClientMask":"h801_1","MqttClient":"h801_1","MqttUser":"user","MqttCount":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
11:07:11 MQT: stat/h801_1/STATUS7 = {"StatusTIM":{"UTC":"Tue Aug 13 09:07:11 2019","Local":"Tue Aug 13 11:07:11 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"06:05","Sunset":"20:32"}}
11:07:11 MQT: stat/h801_1/STATUS10 = {"StatusSNS":{"Time":"2019-08-13T11:07:11"}}
11:07:11 MQT: stat/h801_1/STATUS11 = {"StatusSTS":{"Time":"2019-08-13T11:07:11","Uptime":"1T00:19:09","Heap":13,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":53,"POWER":"OFF","Dimmer":100,"Color":"000000DC23","HSBColor":"0,0,0","Channel":[0,0,0,86,14],"CT":200,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"WIFI","BSSId":"CC:CE:1E:34:T3:67","Channel":6,"RSSI":74,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```

### OUTPUT
```
11:18:44 CMD: wakeupduration 20
11:18:44 MQT: stat/h801_1/RESULT = {"WakeUpDuration":20}
11:18:54 CMD: dimmer
11:18:54 MQT: stat/h801_1/RESULT = {"Dimmer":100}
11:18:59 MQT: tele/h801_1/STATE = {"Time":"2019-08-13T11:18:59","Uptime":"1T00:30:57","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Dimmer":100,"Color":"00000049B6","HSBColor":"0,0,0","Channel":[0,0,0,29,71],"CT":400,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"WIFI","BSSId":"CC:CE:1E:34:T3:67","Channel":6,"RSSI":90,"LinkCount":1,"Downtime":"0T00:00:04"}}
11:19:02 CMD: wakeup 100
11:19:02 MQT: tele/h801_1/STATE = {"Time":"2019-08-13T11:19:02","Uptime":"1T00:31:00","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":19,"POWER":"ON","Dimmer":100,"Color":"00000049B6","HSBColor":"0,0,0","Channel":[0,0,0,29,71],"CT":400,"Scheme":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"WIFI","BSSId":"CC:CE:1E:34:T3:67","Channel":6,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:04"}}
11:19:02 MQT: stat/h801_1/RESULT = {"POWER":"ON"}
11:19:02 MQT: stat/h801_1/POWER = ON
11:19:02 MQT: stat/h801_1/RESULT = {"Wakeup":"Started"}
11:19:07 CMD: ct 300
11:19:07 MQT: tele/h801_1/STATE = {"Time":"2019-08-13T11:19:07","Uptime":"1T00:31:05","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":29,"POWER":"ON","Dimmer":22,"Color":"0000002018","HSBColor":"0,0,0","Channel":[0,0,0,13,9],"CT":300,"Scheme":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"WIFI","BSSId":"CC:CE:1E:34:T3:67","Channel":6,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:04"}}
11:19:07 MQT: stat/h801_1/RESULT = {"POWER":"ON","Dimmer":22,"Color":"0000002018","HSBColor":"0,0,0","Channel":[0,0,0,13,9],"CT":300}
11:19:08 MQT: tele/h801_1/WAKEUP = {"Wakeup":"Done"}
```

ascillato2, 2019-08-13:
Sorry, that behaviour is intended.

closed 2019-08-13 by ascillato2

Mr-iX, 2019-08-13:
@ascillato2 Why is that behavior intended? What’s the purpose?

Is there an event published when the wake up is finished so that I can trigger a rule?
So then I could disable my automatic color temperature rule (which sends ct commands) and after the wake up automatically enable it again through another rule.

robertwoodhead, 2020-03-16:
I think if you set up code to listen for tele/%topic%/wakeup -> %topic%/cmnd/ct "300" You'll get the CT change when wakeup finished is sent by the light.