arendst/Tasmota #6219: Can't set DisplayModel 4 for ILI9341

MaceMoneta, 2019-08-13:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
Attached an ILI9341 to a Wemos D1 Mini Lite. Confirmed proper operation with Arduino example software. Used esptool.py to erase the firmware, then upload sonoff-display 6.6.0. After configuring the module, I tried setting the Displaymode to 4, but it stays at zero:

15:00:31 CMD: DisplayModel 4
15:00:31 RSL: RESULT = {"DisplayModel":0}
15:00:47 CMD: display
15:00:47 RSL: RESULT = {"Display":{"Model":0,"Mode":1,"Dimmer":1,"Size":1,"Font":1,"Rotate":0,"Refresh":2,"Cols":[16,8],"Rows":2}}

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Wemos D1 Mini Lite
- [x] Tasmota binary firmware version number used : sonoff-display 6.6.0 / (pre-compiled)
- [x] Development IDE - Compiler / Upload tools used : esptool 2.6
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
15:12:35 CMD: status 0
15:12:35 RSL: STATUS = {"Status":{"Module":0,"FriendlyName":["display1"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:12:35 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-display.bin","RestartReason":"Software/System restart","Uptime":"0T00:12:23","StartupUTC":"2019-08-13T14:00:12","Sleep":50,"CfgHolder":4617,"BootCount":8,"SaveCount":24,"SaveAddress":"F4000"}}
15:12:35 RSL: STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-display)","BuildDateTime":"2019-07-06T13:13:37","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
15:12:35 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["moneta","moneta"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","280500000100060000000000000000000000","00000000"]}}
15:12:35 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":481,"Free":520,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FAAA39E","001787A0","22B617C1","010028C0"]}}
15:12:35 RSL: STATUS5 = {"StatusNET":{"Hostname":"display1","IPAddress":"135.82.8.105","Gateway":"135.82.8.1","Subnetmask":"255.255.255.0","DNSServer":"135.82.8.1","Mac":"60:01:94:A6:7D:FD","Webserver":2,"WifiConfig":4}}
15:12:35 RSL: STATUS7 = {"StatusTIM":{"UTC":"Tue Aug 13 14:12:35 2019","Local":"Tue Aug 13 15:12:35 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:40","Sunset":"20:10"}}
15:12:35 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-08-13T15:12:35"}}
15:12:35 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-08-13T15:12:35","Uptime":"0T00:12:23","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"moneta","BSSId":"78:8A:20:21:81:20","Channel":6,"RSSI":60,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
15:13:31 CMD: display
15:13:31 SRC: WebConsole from 135.82.8.18
15:13:31 RSL: Received Topic /display, Data Size 0, Data 
15:13:31 RSL: Group 0, Index 1, Command DISPLAY, Data 
15:13:31 RSL: RESULT = {"Display":{"Model":0,"Mode":1,"Dimmer":1,"Size":1,"Font":1,"Rotate":0,"Refresh":2,"Cols":[16,8],"Rows":2}}
15:13:37 WIF: Checking connection...
15:13:37 WIF: Connected
15:13:46 CMD: displaymodel 4
15:13:46 SRC: WebConsole from 135.82.8.18
15:13:46 RSL: Received Topic /displaymodel, Data Size 1, Data 4
15:13:46 RSL: Group 0, Index 1, Command DISPLAYMODEL, Data 4
15:13:46 RSL: RESULT = {"DisplayModel":0}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Install sonoff-display.
Configure SPI connections.
Try to set DisplayMode 4 in console. 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Try to set DisplayMode 4 in console. 
Displaymode 4 set.
Display becomes usable.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-08-13:
The driver only supports hardware SPI so your configuration should look like this:

[image]

MaceMoneta, 2019-08-13:
That was it, thanks.

closed 2019-08-13 by MaceMoneta

bakman2, 2020-03-07:
@arendst are the settings in the screenshot you posted still valid ? Screen remains empty for me, or should I open a new topic ?

arendst, 2020-03-07:
Try Discord first as it is not clear what version you are using.

bakman2, 2020-03-07:
@arendst  on discord i did not get response.

I am using this shield on a wemos d1 mini
https://wiki.wemos.cc/products:d1_mini_shields:tft_2.4_shield

| D1				| mini Shield |
| ------------------|-------------|
| NC*(D1/D2/D3/D4)	| TFT_LED     |
| RST*(D1/D3/D4))	| TFT_RST     |
| D8				| TFT_DC      |
| D7				| MOSI        |
| D6				| MISO		  |
| D5				| SCK		  |
| D0				| TFT_CS	  |
| D3*(D1/D2/D4)		| TS_CS		  |
*default

Tried on both 
Tasmota v8.1.0 Doris
Tasmota v7.2.0 Constance (currently running)

```
16:59:11 DSP: Power 1
16:59:11 RSL: stat/tasmota_DD784D/RESULT = {"POWER":"ON"}
16:59:11 RSL: stat/tasmota_DD784D/POWER = ON
16:59:13 HTP: Console
16:59:17 CMD: status 0
16:59:17 SRC: WebConsole from 10.0.0.227
16:59:17 CMD: Group 0, Index 1, Command "STATUS", Data "0"
16:59:17 RSL: stat/tasmota_DD784D/STATUS = {"Status":{"Module":18,"FriendlyName":["tasmo_display"],"Topic":"tasmota_DD784D","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:59:17 RSL: stat/tasmota_DD784D/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T00:00:22","StartupUTC":"2020-03-07T15:58:55","Sleep":50,"CfgHolder":4617,"BootCount":34,"SaveCount":118,"SaveAddress":"F4000"}}
16:59:17 RSL: stat/tasmota_DD784D/STATUS2 = {"StatusFWR":{"Version":"7.2.0(display)","BuildDateTime":"2019-12-22T11:42:36","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"292/1151"}}
16:59:17 RSL: stat/tasmota_DD784D/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["APAC02",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
16:59:17 RSL: stat/tasmota_DD784D/STATUS4 = {"StatusMEM":{"ProgramSize":551,"Free":452,"Heap":28,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640C8","FlashMode":3,"Features":["00000809","8FAAA19F","04049FA0","22B617C1","010008C0","00004080"],"Drivers":"1,2,4,5,8,9,10,13,16","Sensors":"1,2,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
16:59:17 RSL: stat/tasmota_DD784D/STATUS5 = {"StatusNET":{"Hostname":"tasmo_display","IPAddress":"10.0.0.243","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.1","Mac":"A4:CF:12:DD:78:4D","Webserver":2,"WifiConfig":4}}
16:59:17 RSL: stat/tasmota_DD784D/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DD784D","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
16:59:17 RSL: stat/tasmota_DD784D/STATUS7 = {"StatusTIM":{"UTC":"Sat Mar 07 15:59:17 2020","Local":"Sat Mar 07 16:59:17 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"07:19","Sunset":"18:43"}}
16:59:17 RSL: stat/tasmota_DD784D/STATUS10 = {"StatusSNS":{"Time":"2020-03-07T16:59:17"}}
16:59:17 RSL: stat/tasmota_DD784D/STATUS11 = {"StatusSTS":{"Time":"2020-03-07T16:59:17","Uptime":"0T00:00:22","UptimeSec":22,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"APAC02","BSSId":"44:D9:E7:91:E1:29","Channel":9,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:07"}}}
16:59:23 WIF: Checking connection...
16:59:23 WIF: Connected
16:59:24 CMD: weblog 4
16:59:24 SRC: WebConsole from 10.0.0.227
16:59:24 CMD: Group 0, Index 1, Command "WEBLOG", Data "4"
16:59:24 RSL: stat/tasmota_DD784D/RESULT = {"WebLog":4}
16:59:27 CMD: display
16:59:27 SRC: WebConsole from 10.0.0.227
16:59:27 CMD: Group 0, Index 1, Command "DISPLAY", Data ""
16:59:27 RSL: stat/tasmota_DD784D/RESULT = {"Display":{"Model":4,"Width":240,"Height":320,"Mode":0,"Dimmer":15,"Size":1,"Font":1,"Rotate":0,"Refresh":2,"Cols":[16,8],"Rows":2}}
16:59:43 WIF: Checking connection...
16:59:43 WIF: Connected
16:59:50 CMD: displaymode
16:59:50 SRC: WebConsole from 10.0.0.227
16:59:50 CMD: Group 0, Index 1, Command "DISPLAYMODE", Data ""
16:59:50 RSL: stat/tasmota_DD784D/RESULT = {"DisplayMode":0}
17:00:03 WIF: Checking connection...
17:00:03 WIF: Connected
```

I thought I had to map the GPIOs according to the table, but seeing the screenshots it is a bit different ?

bakman2, 2020-03-08:
Ok it the display is actually working, but only with displaymode 2. with 0 and DisplayText, i don't see anything. Will search further

scargill, 2020-04-27:
Hi @arendst 

I don't use CS or screen brightness as only one SPI device - but I've put them on some pins in theESP on the assumption they need to be there but are flexible..- 

[image]

I'm running DisplayModel 4
DisplayMode 0
DisplayText [s1l1c1]Hello how you

All of this is on latest tasmota-display.bin from DEVELOPMENT so 8.2.0.5 

DC is obviously needed, my PCB  has it on GPIO16 - I put backlight on TX after disabling serial monitoring. I get no errors - but the display rmains grey - no text.

DS18b20 works. Do I absolutely HAVE to rewire and use GPIO4 and 5 for unused CS and backlight?

and have I mised anything above - please - I can get little sense on this out of Discord. I do know how to use the ILI board and this one IS wired and normally working but not with Tasmota (tasmota-display). Any pointers would possibly save me HOURS of blind checking.

A fellow voluntary creator - Pete... don'y expect support but if you could tell me if I'm missing hte point somewhere would be marvellous.

arendst, 2020-04-28:
Pete,

your config seems right to me.

In the assumption you turned on backlight (``power 1``) the correct command would be ``DisplayText [C0s1l1c1]Hello how are you``. Notice the inverted color command ``C0`` in front.

Over time the color became inverted and to get it going again just execute ``C0`` once.

scargill, 2020-04-28:
> Pete,
>
> your config seems right to me.
>
> In the assumption you turned on backlight (`power 1`) the correct command would be `DisplayText [C0s1l1c1]Hello how are you`. Notice the inverted color command [...]

Ok, with your mod, no change, still grey display (remember I know it works generally - just not with Tasmota). Te output from power up is below inc commands to console and the responses.. and you have my configuration. No idea why the info I've put below in BOLD shows all 255 in configuration as I checked before and after that the config is as my picture I sent higher up.

One thing that might be useful - here is my pin usage
//  GPIO2 is DS18b20
// GPIO4 and  are inverted touch buttons - so in Tasmota - experiment using button2i and button3i 
// WEMOS D7 ESP8266 GPIO13 is MOSI (dout)
// WEMOS D5 ESP8266 GPIO14 is clock
// WEMOS D0 ESP8266 GPIO16 as the D/C line.
// WEMOS D8 ESP8266 GPIO15 not used – it is SPI CHIP select but not needed here.
// (Can’t re-use as a button input as GPIO15 must be LOW at power up) 

I don't connect MISO at all as commands are going into the ILI), I have the display reset fastened to processor reset, Display backlight hardwired to 3v3,  display CS fastened to power or ground - can't remember but it is correct as I had the same board running on ESP-GO 2 days ago and another running for over a year on ESP-GO) - is any of that needed for the Tasmota installation other than the definitions in "CONFIGURE MODULE" ?

Throughout all of my tests with tasmota-display.bin, my DS18b20 works fine on GPIO2 and my 2 inverted touch buttons work on GPIO4 and 5.  This is not a bodge wiring job, it's a working quality  PCB with the display hardwired in.

Anything I'm missing or have mis-understood about GPIO requirements for Tasmota? I want to use Tasmota for this as I use it for lights and power plugs...

and in case that mess of console responses below doesnt make it obvious, yes I remembered "power 1" :-)

Pete

**[11:30:24] cmnd/ili/seriallog 0**
[11:30:24] stat/ili/RESULT {"SerialLog":{"0":{"Active":"0"}}}
[11:32:54] tele/ili/STATE {"Time":"2020-04-28T11:32:49","Uptime":"0T00:05:10","UptimeSec":310,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"office","BSSId":"00:1B:29:66:D1:B8","Channel":6,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:04"}}
[11:32:54] tele/ili/SENSOR {"Time":"2020-04-28T11:32:49","DS18B20":{"Id":"01186BC05FFF","Temperature":33.3},"TempUnit":"C"}
**[11:33:00] cmnd/ili/DisplayModel 4**
[11:33:00] stat/ili/RESULT {"DisplayModel":4}
[11:33:02] tele/ili/LWT Offline
[11:33:07] tele/ili/LWT Online
[11:33:07] cmnd/ili/POWER 
[11:33:07] tele/ili/INFO1 {"Module":"Generic","Version":"8.2.0.5(c9149b5-display)","FallbackTopic":"cmnd/DVES_4B5430_fb/","GroupTopic":"cmnd/tasmotas/"}
[11:33:07] tele/ili/INFO2 {"WebServerMode":"Admin","Hostname":"ili","IPAddress":"192.168.14.160"}
[11:33:07] tele/ili/INFO3 {"RestartReason":"Software/System restart"}
[11:33:07] stat/ili/RESULT {"POWER":"ON"}
[11:33:07] stat/ili/POWER ON
[11:33:07] cmnd/ili/status 0
[11:33:07] cmnd/ili/modules 
[11:33:07] cmnd/ili/gpios 255
[11:33:07] cmnd/ili/switchdebounce 
[11:33:07] cmnd/ili/blinktime 
[11:33:07] cmnd/ili/mqttlog 
[11:33:07] cmnd/ili/pulsetime2 
[11:33:07] cmnd/ili/pulsetime4 
[11:33:07] cmnd/ili/pulsetime6 
[11:33:07] cmnd/ili/pulsetime8 
[11:33:07] stat/ili/STATUS {"Status":{"Module":18,"FriendlyName":["ili"],"Topic":"ili","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
[11:33:07] stat/ili/STATUS1 {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:06","StartupUTC":"2020-04-28T10:32:56","Sleep":50,"CfgHolder":4617,"BootCount":31,"BCResetTime":"2020-04-26T19:26:44","SaveCount":238,"SaveAddress":"F6000"}}
[11:33:07] stat/ili/STATUS2 {"StatusFWR":{"Version":"8.2.0.5(c9149b5-display)","BuildDateTime":"2020-04-26T18:01:59","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"351/699"}}
[11:33:07] stat/ili/STATUS3 {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["office",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","01000000","00000000"]}}
[11:33:07] stat/ili/STATUS4 {"StatusMEM":{"ProgramSize":544,"Free":456,"Heap":25,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164068","FlashMode":3,"Features":["00000809","8FAAA59F","04049FA0","000000C1","010000C0","00004080","00000020"],"Drivers":"1,2,4,5,8,9,10,13,16","Sensors":"1,2,5,6"}}
[11:33:07] stat/ili/STATUS5 {"StatusNET":{"Hostname":"ili","IPAddress":"192.168.14.160","Gateway":"192.168.14.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.14.1","Mac":"CC:50:E3:4B:54:30","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
[11:33:07] stat/ili/STATUS6 {"StatusMQT":{"MqttHost":"192.168.14.71","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4B5430","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
[... cut ...]
[11:33:12] tele/ili/STATE {"Time":"2020-04-28T11:33:06","Uptime":"0T00:00:10","UptimeSec":10,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"office","BSSId":"00:1B:29:66:D1:B8","Channel":6,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:04"}}
[11:33:12] tele/ili/SENSOR {"Time":"2020-04-28T11:33:06","DS18B20":{"Id":"01186BC05FFF","Temperature":33.4},"TempUnit":"C"}
**[11:33:21] cmnd/ili/Displaymode 0**
[11:33:21] stat/ili/RESULT {"DisplayMode":0}
**[11:34:56] cmnd/ili/DisplayText [C0s1l1c1]hello how are you**
[11:34:56] stat/ili/RESULT {"DisplayText":"[C0s1l1c1]hello how are you"}

arendst, 2020-04-28:
Just uploaded the latest tasmota-display bin on my ILI9341 which still works fine.

Pls verify your display settings with the command ``display``. Should partly look like this:
```
14:17:55 CMD: display
14:17:55 MQT: stat/tft2/RESULT = {"Display":{"Model":4,"Width":240,"Height":320,"Mode":5,"Dimmer":1,"Size":2,"Font":0,"Rotate":1,"Refresh":1,"Cols":[26,8],"Rows":14}}
```
Your mode and rotate will probably be different.

Your GPIO config seems fine too compared to mine:
```
14:21:16 CMD: gpio
14:21:16 MQT: stat/tft2/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"68":"Backlight"},"GPIO5":{"4":"DS18x20"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"66":"SPI CS"},"GPIO16":{"67":"SPI DC"}}
```
The important ones ar SPI CS and SPI DC as they are used by Tasmota to configure the other three hardware SPI GPIO's like GPIO12 = SPI MISO, GPIO13 = SPI MOSI and GPIO14 is SPI CLK.

If all hardware SPI's are needed depends on the used library. I use the Adafruit ILI9341 library which seems to be very relaxed regarding unconfigured pins.

scargill, 2020-04-28:
You won't believe this. I put ESP-GO back on that board as I didn't want to re-wire it.... and took another less polished board with Wemos D1, a DHT22 and the ILI display, exactly the same configuration  - put Tasmota-display onto it with no setup apart from the config module- and the display worked IMMEDIATELY - it's in display mode 1 -  I'll need to change to get landscape - otherwise I'm off to a flying start - unbelievable.  I just need to change the orientation and increase the text size now... and thanks for your responses - really appreciated.

scargill, 2020-04-28:
Having said that after  reboot I get  black screen - displaymode 1 brings up some red/green material, initially most but not all of the time and date in green on red at one side, version and miscellaneous data on the other side.- but the parameters in square braces show up in the text. displaymode 0 gives me just a blank screen and it ignored my text.  Could I possibly trouble you once more? i need landscape, clear then any text in say red on black - I should be able to take it from there.... it is clearly working in one way or another... I don;t know why the status info.

I've put a line in that should format the text

DisplayText [zf2p7l1c1]Some text here[u128:20:3] but the parameters just show in the image.

[image]

bakman2, 2020-07-04:
@arendst would it be possible to add touch button support for this display ? (can open a new feature request if needed).