arendst/Tasmota #6220: DisplayText not working correctly with SSD1306/SH1106

kaennchenstruggle, 2019-08-13:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
Using (rotate:0) DisplayText is just blinking for 10ms or so.
Using (rotate:2) DisplayText is displayed at the bottom of the screen along with a lot of garbage. But upon each time issuing the command it's being displayed somewhere (in the same line) see screenshots.
This is independent of Cols/Rows/Size settings.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : ESP-01
- [x] Tasmota binary firmware version number used : 6.6.0.4 / compiled using TasmoCompiler; USE_DISPLAY + USE_DISPLAY_SSD1306 defined
- [x] Development IDE - Compiler / Upload tools used : ____ / NodeMCU PyFlasher 4.0
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
16:32:54 CMD: status 0
16:32:54 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:32:54 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:35:14","StartupUTC":"2019-08-13T14:57:40","Sleep":50,"CfgHolder":4617,"BootCount":12,"SaveCount":68,"SaveAddress":"F8000"}}
16:32:54 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0.4(sonoff)","BuildDateTime":"2019-08-11T08:43:19","Boot":31,"Core":"2_5_2","SDK":"2.2.2-dev(c0eb301)"}}
16:32:54 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["DraussenNurKaennchen",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000005A00000000000000","00000000"]}}
16:32:54 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":612,"Free":388,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashMode":3,"Features":["00000809","87FAE397","043F8121","23B617CD","01003BC0","00000000"]}}
16:32:54 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-8163","IPAddress":"192.168.142.179","Gateway":"192.168.142.48","Subnetmask":"255.255.255.0","DNSServer":"192.168.142.48","Mac":"60:01:94:00:FF:E3","Webserver":2,"WifiConfig":4}}
16:32:54 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_00FFE3","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
16:32:54 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Aug 13 15:32:54 2019","Local":"Tue Aug 13 16:32:54 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:40","Sunset":"20:10"}}
16:32:54 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-08-13T16:32:54","Epoch":1565710374}}
16:32:54 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-08-13T16:32:54","Epoch":1565710374,"Uptime":"0T00:35:14","UptimeSec":2114,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"DraussenNurKaennchen","BSSId":"98:9B:CB:2C:A6:5E","Channel":6,"RSSI":74,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
```

16:37:48 CMD: i2cscan
16:37:48 RSL: stat/sonoff/RESULT = {"I2CScan":"Device(s) found at 0x3c"}
16:37:51 CMD: display
16:37:51 RSL: stat/sonoff/RESULT = {"Display":{"Model":2,"Width":128,"Height":64,"Mode":0,"Dimmer":15,"Size":1,"Font":2,"Rotate":2,"Refresh":2,"Cols":[16,8],"Rows":17}}
16:37:58 CMD: DisplayText [s1l1c1]Hello how are you?
16:37:58 RSL: stat/sonoff/RESULT = {"DisplayText":"[s1l1c1]Hello how are you?"}
```

### TO REPRODUCE
SH1106 connected on GPIO0/SDA GPIO2/SCL
Link to module: [(https://www.ebay.de/itm/1-3-OLED-Arduino-Display-SH1106-128x64-I2C-Modul-Raspberry-Pi-Blau/162572748094?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m2749.l2649)]

configuration commands:
```
module 18
gpio4 06
gpio5 05
gpio16 162
power1 on
displaymodel 2
displaymode 1
```

### EXPECTED BEHAVIOUR
Expect to display text at l1/c1

### SCREENSHOTS
![IMG_7881](https://user-images.githubusercontent.com/54073894/62956303-fdd76d80-bdf2-11e9-8ef0-608cf72a4f22.JPG)
![IMG_7880](https://user-images.githubusercontent.com/54073894/62956308-016af480-bdf3-11e9-88fc-5a8ff4ff14aa.JPG)

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-08-13:
Cannot reproduce. My SSD1306 is working just fine using the SSD1306 driver.

From your link I see your hardware is a SH1106. That doesn't work correctly with the standard SSD1306 driver.

kaennchenstruggle, 2019-08-13:
> 
> 
> Cannot reproduce. My SSD1306 is working just fine using the SSD1306 driver.
> 
> From your link I see your hardware is a SH1106. That doesn't work correctly with the standard SSD1306 driver.

Oh my mistake then. I was mislead by the comment on https://github.com/arendst/Sonoff-Tasmota/wiki/Displays
`SH1106 = 1,18 k`

Can I set some compiler defines or so to get that working? (Can't compile myself, have to use Tasmo Compiler - great tool btw.)

Jason2866, 2019-08-13:
SH1106 is only supported in this fork https://github.com/gemu2015/Sonoff-Tasmota
Gemu did a big part of the display drivers of Tasmota.
I edited the wrong wiki entry which suggests SH1106 is supported in orig. Tasmota

closed 2019-08-13 by ascillato2