arendst/Tasmota #6225: Tasmota 6.6.0 dosent work with mosquitto 1.6.4

rikray02, 2019-08-14:
#4755 Tasmota 6.6.0 wont work with mosquitto 1.6.4

Tonight, i upgrade Mosquitto Broker(on my MAC) from 1.5.8 to 1.6.4 version.  After that, all my Sonoff with tasmota 6.6.0 refuse to connect to Mosquitto.

**Console of Sonoff Tasmota**
05:55:36 MQT: Attempting connection...
05:55:42 MQT: Connect failed to 10.0.1.15:1883, rc -2. Retry in 10 sec
05:55:53 MQT: Attempting connection...
05:55:59 MQT: Connect failed to 10.0.1.15:1883, rc -2. Retry in 10 sec
05:56:10 MQT: Attempting connection...
05:56:16 MQT: Connect failed to 10.0.1.15:1883, rc -2. Retry in 10 sec
05:56:27 MQT: Attempting connection...
05:56:33 MQT: Connect failed to 10.0.1.15:1883, rc -2. Retry in 10 sec
05:56:44 MQT: Attempting connection...
05:56:50 MQT: Connect failed to 10.0.1.15:1883, rc -2. Retry in 10 sec
05:57:01 MQT: Attempting connection...
05:57:07 MQT: Connect failed to 10.0.1.15:1883, rc -2. Retry in 10 sec
05:57:18 MQT: Attempting connection...
05:57:24 MQT: Connect failed to 10.0.1.15:1883, rc -2. Retry in 10 sec
05:57:35 MQT: Attempting connection...
05:57:41 MQT: Connect failed to 10.0.1.15:1883, rc -2. Retry in 10 sec
05:57:52 MQT: Attempting connection...
05:57:58 MQT: Connect failed to 10.0.1.15:1883, rc -2. Retry in 10 sec
05:58:09 MQT: Attempting connection...
05:58:15 MQT: Connect failed to 10.0.1.15:1883, rc -2. Retry in 10 sec  

**Console Mosquitto Server (Broker)**
1565757596: mosquitto version 1.6.4 starting
1565757596: Config loaded from /usr/local/etc/mosquitto/mosquitto.conf.
1565757596: Opening ipv6 listen socket on port 1883.
1565757596: Opening ipv4 listen socket on port 1883.
1565757596: Warning: Mosquitto should not be run as root/administrator.
1565757597: New connection from 10.0.1.15 on port 1883.
1565757597: New connection from 10.0.1.15 on port 1883.
1565757597: New connection from 10.0.1.15 on port 1883.
1565757597: New connection from 10.0.1.15 on port 1883.
1565757597: New connection from 10.0.1.15 on port 1883.
1565757597: New connection from 10.0.1.15 on port 1883.
1565757597: New connection from 10.0.1.15 on port 1883.
1565757597: New connection from 10.0.1.15 on port 1883.
1565757597: New connection from 10.0.1.15 on port 1883.
1565757597: New connection from 10.0.1.15 on port 1883.
1565757597: New connection from 10.0.1.15 on port 1883.
1565757597: New connection from 10.0.1.15 on port 1883.
1565757597: New connection from 10.0.1.15 on port 1883.
1565757597: New connection from 10.0.1.15 on port 1883.
1565757597: New client connected from 10.0.1.15 as mqttjs_5adfbd8d (p2, c1, k10).
1565757597: New client connected from 10.0.1.15 as mqttjs_b8225d43 (p2, c1, k10).
1565757597: New client connected from 10.0.1.15 as mqttjs_8a629022 (p2, c1, k10).
1565757597: New client connected from 10.0.1.15 as mqttjs_eaa7334e (p2, c1, k10).
1565757597: New client connected from 10.0.1.15 as mqttjs_853219cc (p2, c1, k10).
1565757597: New connection from 10.0.1.15 on port 1883.
1565757597: New client connected from 10.0.1.15 as mqttjs_7bef2e77 (p2, c1, k10).
1565757597: New client connected from 10.0.1.15 as mqttjs_1d42090f (p2, c1, k10).
1565757597: New client connected from 10.0.1.15 as mqttjs_acdaf756 (p2, c1, k10).
1565757597: New client connected from 10.0.1.15 as mqttjs_b653d467 (p2, c1, k10).
1565757597: New client connected from 10.0.1.15 as mqttjs_8861da00 (p2, c1, k10).
1565757597: New client connected from 10.0.1.15 as mqttjs_14871dfd (p2, c1, k10).
1565757597: New client connected from 10.0.1.15 as mqttjs_f653309a (p2, c1, k10).
1565757597: New client connected from 10.0.1.15 as mqttjs_31ab21e9 (p2, c1, k10).
1565757597: New client connected from 10.0.1.15 as mqttjs_41cc4544 (p2, c1, k10).
1565757597: New client connected from 10.0.1.15 as mqttjs_54bdc0c6 (p2, c1, k10).

Any suggest why Sonoff won't to connect with Broker 1.6.4?

Thanks for your help
Richard

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Sonoff Basic, Sonoff Dual, Sonoff TH16, Sonoff S31
- [ ] Tasmota binary firmware version number used : 6.6.0 pre-compiled
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` :
```
**STATUS 0 OUTPUT HERE:**
06:05:03 CMD: status 0
06:05:03 SRC: WebConsole from 10.0.1.15
06:05:03 RSL: Received Topic /status, Data Size 1, Data 0
06:05:03 RSL: Group 0, Index 1, Command STATUS, Data 0
06:05:03 RSL: stat/Libre/STATUS = {"Status":{"Module":1,"FriendlyName":["Cabanon"],"Topic":"Libre","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
06:05:03 RSL: stat/Libre/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"4T02:14:50","StartupUTC":"2019-08-10T02:50:13","Sleep":50,"CfgHolder":4617,"BootCount":27,"SaveCount":184,"SaveAddress":"F8000"}}
06:05:03 RSL: stat/Libre/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
06:05:03 RSL: stat/Libre/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Airport_Maison",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["80008009","280500000100060000000000000000000000","00000000"]}}
06:05:03 RSL: stat/Libre/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
06:05:03 RSL: stat/Libre/STATUS5 = {"StatusNET":{"Hostname":"Libre-5736","IPAddress":"10.0.1.207","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.1","Mac":"DC:4F:22:DF:F6:68","Webserver":2,"WifiConfig":4}}
06:05:03 RSL: stat/Libre/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.15","MqttPort":1883,"MqttClientMask":"Libre","MqttClient":"Libre","MqttUser":"DVES_USER","MqttCount":5,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
06:05:03 RSL: stat/Libre/STATUS7 = {"StatusTIM":{"UTC":"Wed Aug 14 05:05:03 2019","Local":"Wed Aug 14 06:05:03 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:41","Sunset":"20:08"}}
06:05:03 RSL: stat/Libre/STATUS10 = {"StatusSNS":{"Time":"2019-08-14T06:05:03"}}
06:05:03 RSL: stat/Libre/STATUS11 = {"StatusSTS":{"Time":"2019-08-14T06:05:03","Uptime":"4T02:14:50","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":631,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Airport_Maison","BSSId":"70:73:CB:BD:43:57","Channel":1,"RSSI":56,"LinkCount":2,"Downtime":"0T00:00:43"}}}
06:05:07 WIF: Checking connection...
06:05:07 WIF: Connected
06:05:14 MQT: Attempting connection...
06:05:20 MQT: Connect failed to 10.0.1.15:1883, rc -2. Retry in 10 sec

```
CONSOLE OUTPUT HERE:

```
**(Please, remember to close the issue when the problem has been addressed)**

curzon01, 2019-08-14:
Can not be reproduced:

```
1565768708: mosquitto version 1.6.4 starting
1565768708: Config loaded from /etc/mosquitto/mosquitto.conf.
1565768708: Opening ipv4 listen socket on port 8883.
1565768708: Opening ipv6 listen socket on port 8883.
1565768708: Opening ipv4 listen socket on port 1883.
1565768708: Opening ipv6 listen socket on port 1883.
...
1565769088: Received PUBLISH from Libre (d0, q0, r0, m0, 'stat/tasmota/phesp08/RESULT', ... (134 bytes))
```

Issue seems be reasoned by your mosquitto configuration - find out using more debug output on your mosquitto configuration (`log_type debug` or `log_type all`)

ascillato, 2019-08-14:
I have the same setup as you and it is working fine. Please check credentials and permissions of your setup.

closed 2019-08-14 by ascillato2