arendst/Tasmota #6226: Add MAC address to Home Assistant Discovery Message

caphm, 2019-08-14:
**Have you looked for this feature in other issues and in the wiki?**
Yes, nothing found.

**Is your feature request related to a problem? Please describe.**
Home Assistant is using identifiers like the MAC address or serial numbers to match up devices which are reported by multiple integrations.
I have a device tracker integration set up (Unifi), which creates Home Assistant devices for each device found on the network, identified by their MAC address. Because the MAC address is not provided in the MQTT discovery message by Tasmota, these Home Assistant devices cannot be matched with the devices created by the MQTT integration.
As a result, I now have two entries in my Home Assistant device_registry for every Sonoff on my network.

**Describe the solution you'd like**
The MAC address of Tasmota devices should be included in the payload of the Home Assistant MQTT discovery message as an additional attribute `cns` (see https://www.home-assistant.io/docs/mqtt/discovery/ section _Supported abbreviations for device registry configuration_).

JSON: `"cns": [ ["mac", "$DEVICE_MAC_ADDRESS"] ]`
where $DEVICE_MAC_ADDRESS is the Sonoff's MAC address in FF:FF:FF:FF:FF:FF form (lowercase is also possible).
It needs to be a nested list, because the attribute is itself a listing of identifiable connections and each connection is given as a list with its type as first and value as second element.

**Describe alternatives you've considered**
Leave it like it is and live with multiple device_registry entries for the same device.

Jason2866, 2019-08-14:
With command Status 5 you get it
```
13:01:33 CMD: Status 5
13:01:33 RSL: stat/sonoff-13D92F/STATUS5 = {"StatusNET":{"Hostname":"sonoff-13D92F-6447","IPAddress":"192.168.2.119","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"EC:FA:BC:13:D9:2F","Webserver":2,"WifiConfig":5}}
```

ascillato2, 2019-08-14:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-08-14 by ascillato2

caphm, 2019-08-14:
> With command Status 5 you get it
>
> ```
> 13:01:33 CMD: Status 5
> 13:01:33 RSL: stat/sonoff-13D92F/STATUS5 = [...]

This does not remotely relate to my request. Did you even read it?
I'm interested in having the MAC address included in the payload of the initial MQTT message for Home Assistant MQTT discovery.

Please reopen this issue as it has not been answered. And it's not a question, it's a feature request for a functionality that is currently not implemented.

meingraham, 2019-08-14:
@caphm 

How are other HA users who use auto-discovery addressing this issue?

Jason2866, 2019-08-14:
Tasmota is NOT designed to statisfy all Autodiscovery features availiable for different Home Automation Hubs.
It is Open Source, feel free to fork and do all the changes you want/need.

caphm, 2019-08-15:
> @caphm
> 
> How are other HA users who use auto-discovery addressing this issue?

They are not, at least I wasn't able to find any info on this. Since the device registry was only recently introduced, I assume that not many have stumbled across this.

> Tasmota is NOT designed to statisfy all Autodiscovery features availiable for different Home Automation Hubs.
> It is Open Source, feel free to fork and do all the changes you want/need.

I have done so and will create a PR when tested successfully.

PR #6236 (2019-08-15): Add MAC address to Home Assistant Discovery Message
## Description:

**Related issue (if applicable):** fixes #6226

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.3.0, 2.4.2 and 2.5.2
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).
commit 7e0ae16c (2019-08-15): Add MAC address to Home Assistant Discovery Message, fixes #6226