arendst/Tasmota #6231: simple get strange response

Hennsarv, 2019-08-15:
/ay 
/ay?o=x 

produces following response - this might be incorrect?

{t}</table>{t}<tr><td style='width:25{c}bold;font-size:38px'>ON</div></td><td style='width:25{c}normal;font-size:38px'>OFF</div></td><td style='width:25{c}normal;font-size:38px'>OFF</div></td><td style='width:25{c}normal;font-size:38px'>OFF</div></td></tr></table>

Hennsarv, 2019-08-15:
(html tags are removed - there is incomplete part of html actually

ascillato2, 2019-08-15:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-08-15 by ascillato2

meingraham, 2019-08-15:
@Hennsarv 

the `/ay` directive was an undocumented Tasmota capability. I believe ioBroker (?) used that to encode commands for Tasmota devices. Perhaps they did so before some http capabilities in Tasmota were transitioned from experimental and into the released feature set. In any case, `ay` is gone and has been for some time.

You will need to go into your home automation hub command definitions and change the directive to the correct directives per the methods documented in the Tasmota wiki.

Mike