arendst/Tasmota #6235: one device constantly loses connection (all other devices work fine)

prankousky, 2019-08-15:
Hi,

I have multiple devices running `tasmota`, including `sonoff` and `Gosund` outlets; this one is about a `nodeMCU` though. I also have multiple nodeMCUs, all of which run without any trouble as far as I can tell. But there is one that keeps losing WiFi connecten, reconnects, loses connection again, reconnects again, etc. etc.

* Firmware `Sonoff-Tasmota 6.6.0.2 von Theo Arends`
* Device `nodeMCUv2`
* used as `WS2812` controller
* `Powerretain on`
* `Teleperiod 10`

`status 0`

```
10:26:58 MQT: stat/tasmota-5909/STATUS = {"Status":{"Module":18,"FriendlyName":["tasmota"],"Topic":"tasmota-5909","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":1,"SwitchRetain":1,"SensorRetain":0,"PowerRetain":1}}
10:26:58 MQT: stat/tasmota-5909/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Exception","Uptime":"0T12:40:36","StartupUTC":"2019-08-14T20:46:22","Sleep":0,"CfgHolder":4617,"BootCount":78,"SaveCount":623,"SaveAddress":"F8000"}}
10:26:58 MQT: stat/tasmota-5909/STATUS2 = {"StatusFWR":{"Version":"6.6.0.2(classic)","BuildDateTime":"2019.08.09 12:26:42","Boot":31,"Core":"2_5_2","SDK":"2.2.2-dev(c0eb301)"}}
10:26:58 MQT: stat/tasmota-5909/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ssid",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["000208E9","280500000100060000005A00000000000000","00000001"]}}
10:26:58 MQT: stat/tasmota-5909/STATUS4 = {"StatusMEM":{"ProgramSize":498,"Free":504,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000407","6000A192","048003A8","200217C6","010000C0"]}}
10:26:58 MQT: stat/tasmota-5909/STATUS5 = {"StatusNET":{"Hostname":"tasmota-5909-5909","IPAddress":"192.168.100.145","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.1","Mac":"60:01:94:2B:D7:15","Webserver":2,"WifiConfig":3}}
10:26:58 MQT: stat/tasmota-5909/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.van","MqttPort":1883,"MqttClientMask":"tasmota-5909","MqttClient":"tasmota-5909","MqttUser":"mqttuser","MqttCount":607,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
10:26:58 MQT: stat/tasmota-5909/STATUS7 = {"StatusTIM":{"UTC":"Thu Aug 15 09:26:58 2019","Local":"Thu Aug 15 10:26:58 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00"}}
10:26:58 MQT: stat/tasmota-5909/STATUS10 = {"StatusSNS":{"Time":"2019-08-15T10:26:58","Epoch":1565861218}}
10:26:58 MQT: stat/tasmota-5909/STATUS11 = {"StatusSTS":{"Time":"2019-08-15T10:26:58","Epoch":1565861218,"Uptime":"0T12:40:36","UptimeSec":45636,"Vcc":3.022,"Heap":22,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"OFF","Dimmer":100,"Color":"255,0,255","HSBColor":"300,100,100","Channel":[100,0,100],"Scheme":0,"Width":1,"Fade":"ON","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"ssid","BSSId":"bssid","Channel":6,"RSSI":74,"LinkCount":1,"Downtime":"0T00:00:04"}}}
10:27:00 MQT: tele/tasmota-5909/STATE = {"Time":"2019-08-15T10:27:00","Epoch":1565861220,"Uptime":"0T12:40:38","UptimeSec":45638,"Vcc":3.022,"Heap":24,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"OFF","Dimmer":100,"Color":"255,0,255","HSBColor":"300,100,100","Channel":[100,0,100],"Scheme":0,"Width":1,"Fade":"ON","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"ssid","BSSId":"bssid","Channel":6,"RSSI":74,"LinkCount":1,"Downtime":"0T00:00:04"}} (beibehalten)
```

I have connected a 60 LED WS2812 to `D7`, `VIN` and `GND`. It works fine (schemes, color control, on/off switching, fading, dimmer, ...). But it keeps disconnecting from the WiFi a lot. Right after that it will re-establish the connection. This is not an infrastructure issue. I have a `UniFi AC AP Pro` on the first floor in the room next to the nodeMCU. RSSI seems fine. Even when I take this nodeMCU downstairs to the very room my other AP is in (decent RSSI as well), it will constantly reconnect.

Every one to four minutes I get something like this 

```
10:27:45 MQT: Verbindungsversuch...
10:27:45 MQT: verbunden
``` 

Which translates roughly to `attempting connection` and `connected`. 

`sleep` ==> `stat/tasmota-5909/RESULT = {"Sleep":"0 (0)"}`. 

Since I have multiple devices running tasmota without this problem, I doubt that [this](https://github.com/arendst/Sonoff-Tasmota/issues/1256) has anything to do with it. Especially because each device has it's own personal fallback topic depending on it's name (which depends on it's mac address).

What can I do to fix this? Do you need any additional output besides `status 0` and everything else I have already provided? 

Thank you for your support.

fribse, 2019-08-15:
Do you have a powersupply directly to the WS2812, they can eat a lot of juice?

prankousky, 2019-08-15:
Thank you @fribse . I have not tried that, but will now and report back.

closed 2019-08-15 by ascillato2