arendst/Tasmota #6237: Adafruit MQTT (AIO) - can't connect

m-jurczyszyn, 2019-08-15:
Hi!
Let me just say that TASMOTA is a blessing for my sonoff devices, and I love it. 

I'm using latest 6.6 sonoff-PL release and my device fails to connect to Adafruit IO, with this error (PL language):
> MQT: Nie udało się nawiązać połączenia io.adafruit.com:1883, rc 7.
In english: Failed to connect to io.adafruit.com:1883, rc 7.
Double checked the server url, port, username and AIO key.
Adafruit connection details:
> Host | io.adafruit.com
> Insecure Port | 1883
> Username | Your Adafruit IO Username
> Password | Your Adafruit IO Key

Device connects to CloudMQTT just fine.
I've read the previous issues regarding Adafruit IO, but the answer isn't there.
The AIO key is 32 characters in length.

ascillato2, 2019-08-15:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato, 2019-08-15:
Anyway, a public MQTT server is not recommended due to it is very easy to hack and take control of your devices, replace the firmware of your devices and take control of your network.

That is why Tasmota is designed to be used in a local network. There are plenty of home automation software options in the wiki.

ascillato, 2019-08-15:
you can try modifying the max amount of available characters for the password

https://github.com/arendst/Sonoff-Tasmota/blob/6d5ffdf068669c0f6b04bcb3656f53a478440b4c/sonoff/settings.h#L222

and compile by yourself

Remember that modifying the settings memory makes Tasmota not being backwards compatible, so you will need to erase all flash and flash your custom version.

m-jurczyszyn, 2019-08-16:
Does the 33 in the brackets mean it's current length is 33 chars?

s-hadinger, 2019-08-16:
As it is a C string the last byte is zero so you have only 32 characters available.

ascillato2, 2019-08-17:
Hi, any news on this?

m-jurczyszyn, 2019-08-17:
I can confirm, that the problem was in the topic. Adafruit is very specific when it comes to the MQTT topic:
> Any one of the following topic forms is valid for a feed:
>     (username)/feeds/(feed name or key)
>     (username)/f/(feed name or key)

I'm still having issues subscribing/ publishing to specific feed. Currently testing various configurations.

m-jurczyszyn, 2019-08-17:
Why does tasmota try to subscribe to topic ending with /cmnd/ ?

meingraham, 2019-08-17:
@m-jurczyszyn 

Have you set Home Assistant auto-discovery [`SetOption19`](../wiki/Commands#setoption19)? This [changes the structure of Tasmota topics](../wiki/Home-Assistant#warning-important).

The topic structure will be `%topic%/cmnd/+`  rather than the default `cmnd/%topic%/+`

Our ability to provide assistance is __greatly__ hampered since you did not take a few minutes to complete the  [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) with the requested information. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

Please close this issue and address any further questions to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2019-08-17:
Great to know that now you could connect to the broker.

If you need further assistance on configuring, please address this to the Tasmota support chat.

Anyway, I still not recommending a public MQTT broker with insecure ports. Those are very easy to hack, take control of your devices and do an OTA and change the firmware off all your devices and control all your network. Really not recommended.

closed 2019-08-17 by ascillato2

tech-freak, 2020-07-30:
> I can confirm, that the problem was in the topic. Adafruit is very specific when it comes to the MQTT topic:

hi there
any update on this ?