arendst/Tasmota #6238: Subtle differences with the Hue API

suhajda3, 2019-08-15:
Hello,

I have turned on Hue Bridge emulation on my Sonoff-Tasmota 6.6.0. I thought since Alexa works great with this emulation it should work with [homebridge-hue](https://github.com/ebaauw/homebridge-hue) as well. After a lot of tries I couldn't make it work so I reported [this issue](https://github.com/ebaauw/homebridge-hue/issues/512). As it turns out Sonoff-Tasmota has some subtle differences with the Hue API.

> 1. Missing attributes from the /config, notably modelid and apiversion;
> 2. It still exposes pointsymbols in the /lights;
> 3. state is missing from the /groups resources;
> 4. The /sensors resource seems to be missing.

It would be great if Sonoff-Tasmota would be compatible with homebrdige-hue.

Thank you!

meingraham, 2019-08-15:
@suhajda3 

There has been additional effort for the Hue emulation and lights features since 6.6.0 master was released. Please load the latest stable version of Tasmota (6.6.0.4) from http://thehackbox.org/tasmota and report back whether the issues you outlined still exist.

Thanks.

Mike

s-hadinger, 2019-08-15:
I will take a look.

s-hadinger, 2019-08-16:
Troubleshooting happening here: https://github.com/ebaauw/homebridge-hue/issues/512

closed 2019-08-22 by arendst

Krocko, 2019-08-27:
@s-hadinger is it possible for you to get the `resourcelinks` working?
So we can expose the device as outlet in homebridge-hue?

s-hadinger, 2019-08-27:
It's already in the latest version v6.6.0.8. Please download the latest binaries.

Krocko, 2019-08-27:
I am on v6.6.0.8 but i can't create a resourcelink.
Can you tel me, how to create a resourcelink on a Sonoff Dual R2?

s-hadinger, 2019-08-27:
There are no plans to implement `resourcelinks`. They would require persistent storage and would impact too much the code size. This type of advanced features are better implemented at MQTT gateway level.

PR #6281 (2019-08-22): Fix Hue Emulation not responding on /resourcelinks, issue #6238
## Description:

Responding to `/resourcelinks` Hue API to make homebridge-hue work with Tasmota. See https://github.com/ebaauw/homebridge-hue/issues/512

**Related issue (if applicable):** fixes #6238 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.3.0, 2.4.2 and 2.5.2
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
   [... cut]