arendst/Tasmota #6239: Support Albohes PC-1606 Outdoor Plug

arzaman, 2019-08-15:
Hi

need to control via tasmota a dual plug for **OUTDOOR** usage
my choice was for Albohes PC-1606, very well done, robust waterproof with tactlile switch

[image]

I try to upgrade via Tuya-Convert with no success so I decide to open it and flash the firmware but not sure about the TX/RX/GPIO0 mapping on the ESP8285  (Vcc and GND are marked)

any one can help me in finding the PINout ?

[image]

[image]

[image]

any alternative dual plug for outdoor (Easy to disassemble or compatible with OTA upgrade)

thanks
Davide

thxthx0, 2019-08-16:
maybe this will help:
https://blakadder.github.io/templates/albohes-pc-1606.html
https://forum.creationx.de/forum/index.php?thread/1767-albohes-sp-1606/

ascillato2, 2019-08-16:
Closing this issue as the device requested is already supported by Tasmota. 

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-08-16 by ascillato2

arzaman, 2019-08-22:
I know that the template and support for this outdoor switch is OK but still don't know what are the PIN for flashing ESP8285

does anyone has idea of the TX/RX/GPIO0 poistion for this device ?

thanks
Davide

arzaman, 2019-08-29:
Anyone has idea where are the PIN for flashing tasmota on this device ?
it's supported as a template so I expect that someone has flashed it

thanks
Davide

meingraham, 2019-08-29:
@arzaman It may have been flashed with Tuya-Convert. Give that a try first... but that may no longer work.

psi-4ward, 2020-12-09:
Anyone had success flashing this device? Its flagged supported but I cant get it work with SonOTA or Tuya-Convert

ronny130286, 2021-08-01:
Hello, i have problem with the same plug, but i cant flash ist via tuya-convert.
I thing its a problem with the orgi firmeware on plug.

The plug create an AP with IP 10.10.7.1 (in pairing mode) so that the tuja convert cant find it.
Is there a soloution for it?

Thx for Help :)