arendst/Tasmota #6241: Link error on Platformio with 6.6.0

mikeww1, 2019-08-16:
I downloaded 6.6.0. Put it in a directory.Started Platformio.Pointed to that directory and just edited
platformio.ini to use der DE Version. Then I tried to compile and received the following error.

Linking .pio\build\sonoff-DE\firmware.elf
c:/users/mikeww1/.platformio/packages/toolchain-xtensa@1.40802.0/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\sonoff-DE\firmware.elf section `.text' will not fit in region `iram1_0_seg'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\sonoff-DE\firmware.elf] Error 1

After that I tried it with several older versions that compiled fine 3 months ago but always the same error. What is the problem ?

ascillato2, 2019-08-16:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-08-16 by ascillato2

mikeww1, 2019-08-16:
I adressed it to the chat and received the following tips that did not help.

**So, wipe out your libraries folder and then grab the libraries from the Tasmota package
Also, make sure you aren't trying to use an old my_user_config.h
If you are making changes to that file, the recommended file to change instead is user_config_override.h**

As I wrote I used a clean folder from github and did not edit anything except in platformio.ini
So I have to assume that this is a more general problem.

ascillato, 2019-08-16:
Tasmota is compiling fine, so the problem is on your building system. Try to uninstall everything and install again.

For more help and support, please use the Tasmota support chat. Github issues are for Tasmota Software Bug reporting with the template completed. Thanks

meingraham, 2019-08-16:
@mikeww1 

As Adrian suggests, let's continue to have this dialog on Discord. I have to agree with him that even if it was a clean install, something was not set up according to the procedure outlined in the wiki. It is quite specific and if you miss even one detail, it can quickly go off the rails. Also, editing platformio.ini can introduce errors if the wrong settings are changed.

I recommend using Gitpod even if you want to customize your compiled firmware. Gitpod provides a known working environment so eliminates any installation issues. The [Gitpod article](https://github.com/arendst/Sonoff-Tasmota/wiki/Compiling-Tasmota-on-Gitpod) in the wiki describes the procedure for using it as well as being able to add your custom settings.

Regards.

Mike

Jason2866, 2019-08-16:
Your Platformio is updated to v.4.0. You need latest platformio.ini from development branch.
Build dir is changed in platformio v.4.0 and this results in the error you get.

mikeww1, 2019-08-17:
Thank you so much Jason !!!!!!!
I took the ini file from the developer branch and now it works perfectly.
I am glad that my environment works again !

Smart0811, 2019-08-24:
Hi Jason, I exactly have the same issue. Can you send to me the Link to the platformio.ini developer Branch.

Cheers,
Helmut

ascillato, 2019-08-24:
Github.com/arendst/Sonoff-Tasmota/platformio.ini