arendst/Tasmota #6242: Integrate Rule/Events with Timers and Hue Emulation

mainLiveBox, 2019-08-16:
Hi, 

I've been a fond user of Tasmota for some time now, and I really welcomed the addition of Timers (Allowed me to set garden watering on odd days at times that wouldn't drench the cats or neighbors), and Events, allowing me to simplify multi relay control of Rolling Shutters (Basically I wanted to retain both manual and remote control, so two switches cooperate to provide Manual/Override and Direction control). 
Initially I handed this by Rule saved events, that would execute the power and delay commands to handle the shutters. event dropshutters and the like :) and loved it.

I recently purchased an Alexa device, and thought...
How nice, I could just shout "Alexa, drop the shutters" and bang. So I enabled the suggested Hue Emulation on the various Tasmotas, and added them to Alexa. 
Then I fought the Alexa Routine menus, and faced both lack of control issues (can't set single second delays and the like), dramatic drag and drop and order issues. Amazon's fault granted. 

Would it be possible to make the Tasmota devices expose set rule/events to Alexa? Current controls are clunky in Alexa menus. 

Would it be possible to expose rules/events to Tasmota Timers? If I want to drop the blinds why should I have to use 4 timers instead of 1 event trigger?

**Describe the solution you'd like**
The list of set rule triggers to be shown as toggles to Alexa devices (or what is most appropriate ofc)
The list of set rule triggers to be shown in the Timer settings on Tasmota. 

**Describe alternatives you've considered**
Not really sure, will welcome alternatives or things I haven't considered.

Thanks in advance,

ascillato, 2019-08-16:
>The list of set rule triggers to be shown as toggles to Alexa devices

Hue emulation only has ON and OFF and some light controls. Sorry.

>The list of set rule triggers to be shown in the Timer settings on Tasmota

You want a rule trigger like ON SWITCH1#STATE=1 to turn on a timer? if that is the case, you have the command timer (to be used inside a rule) to enable/disable timers.
Or you want a timer to trigger a rule? That case is also available in the timers menu.

closed 2019-08-16 by ascillato2

s-hadinger, 2019-08-16:
There is no out-of-the-box solution for this. As Adrian mentioned, only on/off/dimmer/color is natively supported between Alexa and Tasmota.

To do what you want will require an Alexa skill (quite easy to do), and a gateway (local or cloud).

You can have a local gateway (Raspberry PI for ex) with pre-coded Alexa skills. Although I don't know them well.

Second option is to use a cloud MQTT Gateway like AWS IoT. When your Tasmota device is connected to AWS IoT, it's easy to control it from an Alexa skill. I'm working on a tutorial for this, but it's too early stage to publish. AWS IoT for Tasmota doc is here: https://github.com/arendst/Sonoff-Tasmota/wiki/AWS-IoT

mainLiveBox, 2019-08-19:
Sorry for the delay, yes I know I can expose something over a Pi, and the proposed solutions will be what I will go with I suppose. 
To clarify, what I meant regarding timers:
Say that I want to raise the blinds at dusk on office days: 
This in my setup would be switch manual to override on a relay, then the direction to move the shutters, then the set delay then the reverse. 
This fits into a single Rule so I can do something like: 
on event#raiseshutters do backlong; power1 on ; delay 5; power2 on; delay 300; power1 off; delay 5; power2 off; endon;
As a Timer this requires extensive work as it is... I have seen the rule option in Timers, but it doesn't seem to allow trigger selection, does it just run all the saved rules, or how does one specificy a given trigger?
Regarding Hue: 
What I meant was this: I suppose that if a single channel sonoff exposes a single hue device and  4 channel relay exposes four different controls over hue, it must be software emulated. 
So what I meant was this: Could there be the option to Set the Hue emulation to expose triggers? 
this sort of result is what i have in mind... Relay1, Relay2, event#raise, event#drop of course exposed as switches or buttons.

meingraham, 2019-08-19:
@mainLiveBox 

Not sure if this is what you were asking

Set a timer (@sunset?) and when it expires Clock#Timer=N (N is the number of the timer you set) is the trigger.
`ON Clock#Timer=1 DO Event raiseshutters ENDON`

cbelloua, 2023-02-19:
Hello,
I need some help with Hue Emulation and rules. Is it possible to have a trigger for a Hue event and what would be the syntax? E.g when a Hue command is received, to execute an HTTP get through a WebQuery. I also would like to know if it's possible for the Tasmota to advertise more than one Hue device. 
Thanking you in advance.