arendst/Tasmota #6244: LedLink(i) doesn't work when Mqtt or Wifi is disconnected

shantur, 2019-08-16:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
When Mqtt or Wifi is disconnected LedLink(i) doesn't follow relays.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Tuya Double Socket https://blakadder.github.io/templates/makegood_MG-AUWF01.html
- [x] Tasmota binary firmware version number used : 6.6.0.5 / (pre-compiled or self-compiled ?)
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [x] Provide the output of command ``status 0`` :
```

00:17:06 RSL: stat/sonoff/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:17:06 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.10.22:8000/firmware.bin","RestartReason":"Software/System restart","Uptime":"0T00:09:21","StartupUTC":"2019-08-16T23:07:45","Sleep":50,"CfgHolder":4617,"BootCount":16,"SaveCount":75,"SaveAddress":"F6000"}}
00:17:06 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0.5(sonoff)","BuildDateTime":"2019-08-17T00:01:28","Boot":31,"Core":"2_5_2","SDK":"2.2.2-dev(c0eb301)"}}
00:17:06 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":5,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SKY91125",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000005A00000000000000","00000000"]}}
00:17:06 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":554,"Free":448,"Heap":28,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00000001"],"Sensors":[1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34]}}
00:17:06 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-2991","IPAddress":"192.168.10.30","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.1","Mac":"BC:DD:C2:90:2B:AF","Webserver":2,"WifiConfig":4}}
00:17:06 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_902BAF","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
00:17:06 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Fri Aug 16 23:17:06 2019","Local":"Sat Aug 17 00:17:06 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:45","Sunset":"20:03"}}
00:17:06 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-08-17T00:17:06","Epoch":1565997426}}
00:17:06 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-08-17T00:17:06","Epoch":1565997426,"Uptime":"0T00:09:21","UptimeSec":561,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"SKY91125","BSSId":"A0:BD:CD:5E:19:1A","Channel":1,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```

### TO REPRODUCE
1. Setup a device with LedLink
2. Disconnect Mqtt or Wifi
3. Observe Led won't follow relay

### EXPECTED BEHAVIOUR
Led shows relay state

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-08-17:
What's in a name. A ledlink that follows link must be a ledlink led.

If you want a led to follow both link and relay then do not define it as ledlknk. Simply use the old led definition.

shantur, 2019-08-17:
I agree to that. LedLink should always show link status.

I had defined LedLink and 2 other leds (LedX) to follow relays.
In the above scenario the LedLink should keep blinking to notify issue with Mqtt / Wifi but other 2 Ledx should still show power state. They are ignored due to the blinking logic.

This fixes the other 2 leds. https://github.com/shantur/Sonoff-Tasmota/commit/46eeceb6dca0d1cd138b911957629f20b879a5d7

closed 2019-08-17 by arendst

ramanraja, 2021-03-03:
My requirement is even simpler than this: I just want the Link LED to blink whenever the connectivity is lost. But that was not happening. Details are below:
Program Version	    9.2.0(tasmota)
Build Date & Time   2021-02-26T22:40:18
Core/SDK Version    2_7_4_9/2.2.2-dev(38a443e)

Problem:
The Link LED is not blinking when WiFi or MQTT connectivity is lost. The relevant settings seem to be correct.

Experiment:

Take the MQTT broker offline; Tasmota is unable to connect to MQTT. Now GET the status of ledpower and setoption31 :

$ Backlog LedPower; SetOption31 

20:25:07 RSL: stat/mytopic/RESULT = {"LedPower1":"OFF"}
20:25:07 RSL: stat/mytopic/RESULT = {"SetOption31":"OFF"}

Both the settings seem to be correct, but the Link LED is not blinking !

Solution:
EXPLICITLY SET both ledpower and setoption31 to 0 again, manully:

19:53:56 CMD: Backlog LedPower 0; SetOption31 0
19:53:56 MQT: stat/mytopic/RESULT = {"LedPower1":"OFF"}
19:53:56 MQT: stat/mytopic/RESULT = {"SetOption31":"OFF"}

Now the Link LED starts blinking whenever wifi or MQTT connectivity is lost.

ascillato, 2021-03-03:
Please, update to latest Tasmota.

hem-bhagat-tark, 2023-05-12:
What is the meaning of link status?

hem-bhagat-tark, 2023-05-12:
What is the meaning of "link status" here?

sfromis, 2023-05-13:
See https://tasmota.github.io/docs/Lights/#status-leds

PR #6246 (2019-08-17): LedX should show power status even if there is no wifi or mqtt.
Fixes: https://github.com/arendst/Sonoff-Tasmota/issues/6244

## Description:

**Related issue (if applicable):** fixes #6244 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [] The code change is tested and works on core 2.3.0, 2.4.2 and 2.5.2
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Sonoff-Tasmota/blob/devel [... cut]