arendst/Tasmota #6245: Advanced template / configuration support

shantur, 2019-08-16:
Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

**Have you looked for this feature in other issues and in the wiki?**
Yes
**Is your feature request related to a problem? Please describe.**
There are many devices (almost all Tuya) which don't work very will unless `SetOptions` or other params like `LedState` isn't set correctly.
Things like power calibration and other params need to set manually everytime
Just setting up Template isn't enough, the user also needs to find and set options accordingly.

**Describe the solution you'd like**
Implement a way to add these options as part of initial configuration. This was discussed in depth in DiscordChat. There were couple of suggestions

1. Use decode-config.py to generate a config without Wifi and Mqtt to generate a JSON file. This JSON file will be published as part of Device Templates and then while initial setup we can choose one of the device from template repo and the settings gets copied over to the new device.

Pros: Tool is already built
Cons: The Json file will contain every setting even default ones and will make it difficult to find out whats minimum changes needed to make this device work.
Also restoring config without Wifi and Mqtt might clear these settings.

2. Use `Backlog` command chain to initialise device instead of decode-config.py. This would also needed to be implemented in Template repository and the magic of copying settings over.

Pros: Functionality already built
Cons: `Backlog` commands can become really long and needs some setup to be able to go to console.

3. Implement something like `init-script` which can read commands from a file / url and do the inital setup. This file will have commands in every line with a possibility of comments. This would also need to be implemented in Template repository.

Pros: Scripts can be very simple and easy to understand.
Cons: Needs some work to implement in Tasmota.

For all the above, a way of providing script in initial Wifi setup screen would make things much easier when you are trying to setup many devices. User can copy paste a script with all Wifi and Mqtt details in here instead of making it a manual process.

**Describe alternatives you've considered**
Already described above

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2019-08-17:
i have implemented a way to setup devices with very variant settings.
i developed a driver for SML OBIS power meters combined with optional gas, water meters. there a dozens of variants.
i initially had them defined per ifdef in the source code.
having scripts i found a much better solution. i simply extract a section of a script to setup a device,
in script you can have arbitrary sections like >M. as long as they are not already defined in scripter they are ignored by the scripting language.
my sml driver checks for that >M section and if found, extracts the section to define the device.
below are some examples of the descriptors.
the user simply has to copy and paste the >M section of his meter type from a library of meters.
he then can adapt, if wanted the descriptor to his needs 
the descriptor defines the SML signals,  the look of the WEBUI as well as the MQTT messages 

// script for combined water,gas,powermeter
>D

; define 3 meters
>M 3
; describes the physical interface
+1,1,c,0,10,H20
+2,4,c,0,50,GAS
+3,3,s,0,9600,SML

; describes the signals extracted, web ui and mqtt desginators
-1,1-0:1.8.0*255(@10000,Zählerstand,cbm,Count,4|
-2,=h==================|
-2,1-0:1.8.0*255(@100,Zählerstand,cbm,Count,3|
-3,77070100010800ff@1000,Verbrauch,KWh,Total_in,3|
-3,=h==================|
-3,77070100100700ff@1,Aktueller Verbrauch,W,Power_curr,2|
-3,=h -------------------------------|
-3,=m 10+11+12 @100,Ströme L1+L2+L3,A,Curr_summ,2|
-3,=m 13+14+15/#3 @100,Spannung L1+L2+L3/3,V,Volt_avg,2|
-3,=h==================|
-3,77070100240700ff@1,Verbrauch P1,W,Power_p1,2|
-3,77070100380700ff@1,Verbrauch P2,W,Power_p2,2|
-3,770701004c0700ff@1,Verbrauch P3,W,Power_p3,2|
-3,=h -------------------------------|
-3,770701001f0700ff@100,Strom L1,A,Curr_p1,2|
-3,77070100330700ff@100,Strom L2,A,Curr_p2,2|
-3,77070100470700ff@100,Strom L3,A,Curr_p3,2|
-3,=h -------------------------------|
-3,77070100200700ff@100,Spannung L1,V,Volt_p1,2|
-3,77070100340700ff@100,Spannung L2,V,Volt_p2,2|
-3,77070100480700ff@100,Spannung L3,V,Volt_p3,2|
-3,=h==================|
-3,77070100000009ff@#,Service ID,,Meter_id,0|
-3,=h--------------------------------
#

; define on OBIS type power meter
;EHZ161
>D

>M 1
+1,0,o,0,9600,OBIS

-1,1-0:1.8.0*255(@1,Verbrauch,KWh,Total_in,4|
-1,1-0:2.8.0*255(@1,Einspeisung,KWh,Total_out,4|
-1,1-0:21.7.0*255(@1,Verbrauch P1,W,Power_p1,0|
-1,1-0:41.7.0*255(@1,Verbrauch P2,W,Power_p2,0|
-1,1-0:61.7.0*255(@1,Verbrauch P3,W,Power_p3,0|
-1,=m 3+4+5 @1,Aktueller Verbrauch,W,Power_curr,0|
-1,1-0:0.0.0*255(@#),Zähler Nr,,Meter_number,0
#

; define one SML type power meter
;EHZ363
>D

>M 1
+1,0,s,0,9600,SML

-1,77070100010800ff@1000,Verbrauch,KWh,Total_in,4|
-1,77070100020800ff@1000,Einspeisung,KWh,Total_out,4|
-1,77070100100700ff@1,Aktueller Verbrauch,W,Power_curr,0|
-1,77070100000009ff@#,Zähler Nr,,Meter_number,0
#

arendst, 2019-08-17:
Some of the proposed solutions will lead to excessive code increase so I suggest using the backlog feature and provide additional config commands at the template website.

curzon01, 2019-08-17:
@shantur - decode-config.py only changes the settings you have defined within the json input - this can also be a subset, other settings will be kept as is.
The way the tool works is always pull/load the binary config from device/file, patch the settings from json and push/write back the changed binary - so you can use a subset of settings as input (json) without impacting the existing device settings like Wifi or MQTT.

closed 2019-08-17 by ascillato2