arendst/Tasmota #6247: 6.6.0.5 Not rep[orting the Power Status to Home Assistant

DavidFW1960, 2019-08-17:
Loaded new 6.6.0.5 today and the noticed the status info (switch on or off) not being reported in Home Assistant correctly.

See log lines from a 6.6.0.4 and 6.6.0.5 console below.

I just noticed that my Home Assistant wasn't showing a switch as off that I had turned off even though the switch was actually off.

16:20:47 MQT: sonoff1/tele/STATE = {"Time":"2019-08-17T16:20:47","Epoch":1566022847,"Uptime":"0T00:13:49","UptimeSec":829,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WILLIAMS","BSSId":"E0:28:6D:76:2B:AE","Channel":6,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:05"}}
16:20:47 MQT: sonoff4/tele/STATE = {"Time":"2019-08-17T16:20:47","Epoch":1566022847,"Uptime":"0T00:05:58","UptimeSec":358,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"WILLIAMS","BSSId":"E0:28:6D:76:2B:AE","Channel":6,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:07"}}

16:20:47 MQT: sonoff1/tele/HASS_STATE = {"Version":"6.6.0.4(sonoff)","BuildDateTime":"2019-08-10T10:44:18","Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Module":"Sonoff Basic","RestartReason":"Software/System restart","Uptime":"0T00:13:49","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:05","MqttCount":1,"BootCount":22,"SaveCount":434,"IPAddress":"10.90.11.24","RSSI":"100","LoadAvg":19}
16:20:47 MQT: sonoff4/tele/HASS_STATE = {"Version":"6.6.0.5(sonoff)","BuildDateTime":"2019-08-17T12:18:10","Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Module":"Sonoff Basic","RestartReason":"Power on","Uptime":"0T00:05:58","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:07","MqttCount":1,"BootCount":36,"SaveCount":77,"IPAddress":"10.90.11.35","RSSI":"100","LoadAvg":19}

16:20:47 MQT: sonoff1/stat/RESULT = {"Version":"6.6.0.4(sonoff)","BuildDateTime":"2019-08-10T10:44:18","Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Module":"Sonoff Basic","RestartReason":"Software/System restart","Uptime":"0T00:13:49","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:05","MqttCount":1,"BootCount":22,"SaveCount":434,"IPAddress":"10.90.11.24","RSSI":"100","LoadAvg":19}
16:20:47 MQT: sonoff4/stat/RESULT = {"Version":"6.6.0.5(sonoff)","BuildDateTime":"2019-08-17T12:18:10","Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Module":"Sonoff Basic","RestartReason":"Power on","Uptime":"0T00:05:58","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:07","MqttCount":1,"BootCount":36,"SaveCount":77,"IPAddress":"10.90.11.35","RSSI":"100","LoadAvg":19}

KPK75, 2019-08-17:
I noticed the same error.

s-hadinger, 2019-08-17:
Sorry, bug introduced in latest commit. Power would not be reported if no PWM is present. Will be corrected shortly.

arendst, 2019-08-17:
pls provide a full status 0 output as i cannot reproduce.

s-hadinger, 2019-08-17:
@arendst This is my fault. I wanted to call `LightState()` only once in the telemetry loop, so only when `device == light_device` discarding any value with `device > light_device`.

I just forgot that when you have only relays, `light_device == 0` so this would discard the readings of all relays if there is no light, i.e. no PWM present.

PR #6250 (2019-08-17): Fix #6247 and bug introduced in #6243
## Description:

Two bugs introduced by #6243

- Comma missing in JSON output between channel outputs
- Power would not be reported in telemetry if no PWM is present

**Related issue (if applicable):** fixes #6247 and #6191 via #6243

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.3.0, 2.4.2 and 2.5.2
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass [... cut]
commit 8084c18c (2019-08-17): Merge pull request #6250 from s-hadinger/fix_6247

Fix #6247 and bug introduced in #6243
commit 859156c3 (2019-08-17): Fix #6247
closed 2019-08-17 by arendst