arendst/Tasmota #6248: SONOFF T4EU1C - 1 Gang EU Wi-Fi Smart Single Wire Wall Switch

FVM00, 2019-08-17:
https://www.itead.cc/sonoff-t4eu1c-wi-fi-smart-single-wire-wall-switch.html

Anybody already seen this device?
Is it ESP8266 based and flasable?
Very exited and curious. Living in an old house with very strange wiring, So have not been able to route an extra wire

[image]

ascillato2, 2019-08-17:
It is already supported.

If you need further assistant, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-08-17 by ascillato2

AS137430, 2019-08-23:
1. Scanned the development changelog and commits. Nothing on Sonoff T4EU1C.
2. Nothing on templates for this model as well.
3. Nothing in issues as well.

@ascillato2: Why did you say this Sonoff model is supported? This product was released in August only.

effinG123, 2019-09-01:
I'm interested in this also!  Agreed can't find info on this device as it has only just been released.

arendst, 2019-09-01:
This device is new indeed.

I expect that module 28 (Sonoff T1 1CH) will do the job just fine. Let me know if it doesn't.

effinG123, 2019-09-01:
Will do.  Thanks...It's on order now.

FVM00, 2019-09-01:
Still looking for (extra) GPIO0 spot :-) .

[image]
[image]
[image]

arendst, 2019-09-01:
Looking at the small amount of logic AND the fact that the device doesn't need both power connections it might well be that Tasmota won't work because it uses a lot of cpu resources.

From the pictures it seems the push button is connected to GPIO0 via the additional microcontroller.

Once you're able to flash I suggest you make a backup first.

Hellcube, 2019-09-05:
So how can i put this into flashing mode?

MCrissDev, 2019-09-05:
I am interested in this too :)

FVM00, 2019-09-10:
Tried to find a connection for GPIO0, but did not find anything on the board. 
Gave up hope and tried to connect directly on the chip.
This worked, flashing mode was enabled (WiFi led not blinking)
Flashing Tasmota without a problem.
Operates also exactly the same as T1.

Currently operating with Homey through MQTT 😁😁😁

SelcouthGuy, 2019-09-10:
@FVM00 thank you very much for this information.
 Please inform us if you find any abnormalities/inconvenience. I'm very interested.

Hellcube, 2019-09-11:
[image]
Is this pin you are referring to?

meingraham, 2019-09-11:
@Hellcube 

In the diagram, yes. In the photo, no.

Note in the diagram the black dot in the corner of pin 1 & pin 33. The dot is also on the physical chip itself. It tells you were pin 1 is. The GPIO0 pin is actually in the diagonally opposite corner in the photo from what you indicated.

arendst, 2019-09-11:
@meingraham I think you're wrong. Both picture and diagram are correct. Remember, you turned the picture upside down.

meingraham, 2019-09-11:
Or likely I'm blind and need glasses :wink:. That looked like a dot in the photo.

Nevertheless, @Hellcube orient using the dot.

arendst, 2019-09-11:
There's also the dot on the PCA next to the almost invisible dot on the chip :-)

FVM00, 2019-09-11:
[image]
I made a new, better, photo. Now you can see clearly the marking spot on the chip and on the board :smiley:

arendst, 2019-09-12:
I think R10 on the flip side is connected to GPIO00 too. So it may be easier to wire it up there.

Hellcube, 2019-09-19:
R10 works. Sucesfully flashed. I have a problem with the relay, it switches on and off all the time when pluged in main power (not from serial adapter).

Snipercaine, 2019-09-19:
I'm in on this too.   
I got it flashed with R10 as my GPIO-0 connection.  I'm not getting a relay toggle with the cap-touch button.  I had the relay toggle working via the UI with the Module as Sonoff Basic, but not the cap-touch switch.

effelle, 2019-09-19:
I had a look to the board, since @Snipercaine jump on our channel to get some info. Looking at the pic I have, seems GPIO0 is the one controlling the touch, but passing trough the MCU (U2).
Is an SOT23-6, Touch I/O Flash MCU model BS83A02A-4.
ON Pin2 (VSS) is connected GPIO0 trough R10 and on Pin5 (PA1/KEY1) there is the touch out.

mB-PiBox, 2019-09-24:
Is the diagram in the Original Post accurate? It indicates a (N) Neutral Wire, but this switch doesn't require one.

jonoxer, 2019-09-25:
> It indicates a (N) Neutral Wire, but this switch doesn't require one.

Neutral is still required to the load, it just doesn't need to go to the switch. The diagram is correct.

mB-PiBox, 2019-09-25:
> > It indicates a (N) Neutral Wire, but this switch doesn't require one.
> 
> Neutral is still required to the load, it just doesn't need to go to the switch. The diagram is correct.

Except in the diagram, it has the "Out" connection of the Switch connected to the (N) Natural Wire.

d4vew557, 2019-09-25:
The out looks to be connected to the live side of the bulb and then the neutral to the other side of the bulb to me, then the anti flicker module connected across both.

mB-PiBox, 2019-09-26:
> The out looks to be connected to the live side of the bulb and then the neutral to the other side of the bulb to me, then the anti flicker module connected across both.

Oi, you're right.. dumba$$ me didn't take notice of the circle/X representing the bulb.

jonoxer, 2019-09-30:
I've done a description of how this board works, and the process that I found to be successful for installing Tasmota. Some of it is documented in the video, some is on the page: https://www.superhouse.tv/34-science-or-sorcery-sonoff-t4eu1c-no-neutral-wifi-touch-switch/

nolimitek, 2019-10-02:
There must be a problem with tasmota because esphome works fine flashed on the"normal" way

effelle, 2019-10-02:
And we are aware of it. The problem is being investigated.

nolimitek, 2019-10-02:
Thats fine. Hope you find the problem. Love tasmota on the sonoff units

s-hadinger, 2019-10-04:
Quick question for current users. Is this device immune to Ghost switching? Thx

effelle, 2019-10-04:
@s-hadinger I'm not sure. 
As for now we tested three units, each one showed different behavior:

- One refuses to flash without using `reset` pin to enter in program mode and flashed successfully just using 74880bps .
- A Second one was flashed as any other ESP but didn't sense the touch button until an additional resistor was added on R10.
- A third one was flashed as any other ESP and works just fine. 

As you can imagine is hard to reply until we get more units with Tasmota on board. 
Seems on EspHome this particular model is working just fine out of the box. 
My fear is we are facing a power issue here: @jonoxer told me the stock firmware was draining about 70-80mA and will test the actual draining under Tasmota. Until then can't say what's going on.

vampik, 2019-10-08:
Successfully flashed without any difficulties. Used R10 as a connection point for GPIO0. Additional GND for easy connection is available at some of J2 pins. Switch works just fine with a couple of fluorescent bulbs and the anti flicker module.

effelle, 2019-10-08:
@vampik thanks for your feedback. 
What is your Voltage configuration? 110v/220v? 50/60Hz?

vampik, 2019-10-08:
> What is your Voltage configuration? 110v/220v? 50/60Hz?
220V 50Hz.
> And what `.bin` you flashed?
I flashed sonoff.bin.

moccajava, 2019-10-12:
Hi guys,

Timely this came out. So my home wiring does not have a N to the switch and I find it next to impossible to run the lines.

So I do have the Sonoff T1 UK switches. Will the anti-flicker module work with this? And can I make my own anti-flicker module? Any help on how to make that will save me the trouble and hassle of snake wiring all my points.

Thanks guys.

vzorglub, 2019-10-13:
No the T1 will not work without neutral. With on without the anti flicker device. See @jonoxer video above

nolimitek, 2019-10-26:
Any news for this device ?

meingraham, 2019-10-26:
Yes. Google it and Tasmota. Several successful outcomes.

nolimitek, 2019-10-26:
Can you send me a link. I read everywhere that after tasmota is flashed the Hardware button is not working anymore

meingraham, 2019-10-26:
https://www.youtube.com/watch?v=4f34SPqP4-A
https://www.youtube.com/watch?v=VNYcD7MEp4A
https://www.youtube.com/watch?v=_q96xx9sZ8o

nolimitek, 2019-10-26:
The first link Hardware Button does not Work
Second link to strong way not doable for me.
Third link if you read the comments the hardware button does not work.

meingraham, 2019-10-26:
Them's your choices at this time.

nolimitek, 2019-10-28:
FYI:
Flashed three T4EU1C with an older sonoff.bin from februar 2019 and all works. After flashing it does not work, But after setting up the wifi and set the device to T1 alls is working fine

Hellcube, 2019-10-31:
Tried new version v6.7.1 does not work. Can You provide me the link to that firmware from february? i will test it...

nolimitek, 2019-10-31:
What exactly does not work?

Hellcube, 2019-11-01:
same deal. When pluged

> R10 works. Sucesfully flashed. I have a problem with the relay, it switches on and off all the time when pluged in main power (not from serial adapter).

Hellcube, 2019-11-01:
> R10 works. Sucesfully flashed. I have a problem with the relay, it switches on and off all the time when pluged in main power (not from serial adapter).

nolimitek, 2019-11-01:
> > R10 works. Sucesfully flashed. I have a problem with the relay, it switches on and off all the time when pluged in main power (not from serial adapter).

Do you have attached the anti flicker module ?

Hellcube, 2019-11-24:
> > > R10 works. Sucesfully flashed. I have a problem with the relay, it switches on and off all the time when pluged in main power (not from serial adapter).
> 
> Do you have attached the anti flicker module ?

I did not flicker module fixed it. Thanks.

ivohulsman, 2019-11-26:
For everyone with a working Tasmota flash but without the actual button working, I replaced R10 with a 1K resistor and everything works fine on my end now.

donseelim, 2020-01-07:
I made a video on how to flash this for those who need it.  https://www.youtube.com/watch?v=RNQXRko5mPU

kreso975, 2020-01-11:
@donseelim Thanks a lot!! :) :) Very easy and fast. 
note: tried with DIO (not working) , DOUT works like a charm. Flash it with my code including SPIFFS.

donseelim, 2020-01-11:
> @donseelim Thanks a lot!! :) :) Very easy and fast.
> note: tried with DIO (not working) , DOUT works like a charm. Flash it with my code including SPIFFS.

You are welcome. Great to know my video helped!

askrejans, 2020-01-27:
Flashed successfully with 8.1.0 version straight after unpacking through the R10 resistor. All worked right away with no modifications, touch button and everything. T1 config preset works fine. Wifi led was blinking all the time, but that could be disabled through the normal config options. Ordered two more, will check if its the same with those.

meingraham, 2020-01-27:
@askrejans 

Link LED blinking is either Wi-Fi (probably not your case) or MQTT not connected. If the latter and you don't use MQTT, you can disable it (`SetOption3 0`).

vwtune, 2020-02-12:
I managed to flash the T4eu with tasmota 8.1.0 but after flashing, the touch is not responding very good, i have to press hard and/ ore multiple times. Over mqtt it is switching instantly. Before flashing it was just fine. What to do?

JulesBousema, 2020-05-07:
Does anyone know if two of the t4eu1c in a row will work? As in the live out from the first goes to the live in from the second? Im worried about enough power being aviable.

pmknowles, 2020-05-26:
@JulesBousema I would say almost certainly not. The first one being switched off would restrict the current to the next. I presume this is for a 2 way switch.  If so you could try using the 'spare' wire (in UK it's yellow) from the output of the first switch to the output of the second and then connect the live feed (UK red or brown) to the L of both switches.

galsalman, 2020-06-09:
Is there a json config available ?

[4 more replies]