arendst/Tasmota #6249: SP111: Tasmota crahes after several toggles

heikoh81, 2019-08-17:
Hi everyone,

I widely use Gosund and Blitzwolf plugs of various form factors without problems.

I've recently bought several Jisusu SP111 (form factor identical to Gosund SP111 and Blitzwohl SHP6).
Due to the small form factor I used tuya-convert for flashing, I don't want to open them for flashing.
I tried 6.6.0.4 with core 2.3.0 and 2.5.2, and also an older version 6.4.1
I tried profile for Gosund SP1 (doesn't work) and Blitzwolf SHP (works). I also used the template for SHP6 from the repository.

However, these Jisusu plugs crash on switching, no matter if I use the Tasmota toggle button,
or if I send power commands with MQTT (tried FHEM built broker as well as mosquitto running on a standalone raspberry Buster testing machine).
They also crash if I disable MQTT and only use the physical button.

There is no pattern when they crash, sometimes I can switch 10x without crash, sometimes they crash immediately.
On crashing, the relay switches several times (probably restoring state before crash).

I don't have these problems with Gosund SP111.

Any help appreciated.
Heiko

meingraham, 2019-08-17:
All of your Jisusu plugs exhibit this issue?

External appearance cannot be relied on. We ha found that sometime appearance and markings are identical yet the internals are different. Try following the new device configuration procedure (link can be found at the top of the Templates article) to confirm all of the GPIO connections.

Our ability to provide further  assistance is greatly hampered since you did not take a few minutes to complete the  [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) with the requested information. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

Please address any further troubleshooting to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-08-17 by ascillato2

heikoh81, 2019-08-17:
The problem happens to all of the recently bought SP111.
I have several Jisusu SP111 from March 2019 that work without problems.

### ISSUE DESCRIPTION - TROUBLESHOOTING
Jisusu SP111 crashes after several relais toggles
  

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [-] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : Jisusu Smart Plug SP111, similar Blitzwolf SHP-6
- [ ] Tasmota binary firmware version number used : 6.6.0 with core 2.3.0, also 6.6.0.5 with core 2.3.0 and core 2.5.1 / (pre-compiled)
- [-] Development IDE - Compiler / Upload tools used : ____ / ____
- [X] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
00:00:00 CFG: Loaded from flash at FB, Count 207
00:00:00 Project sonoff SmartHomeGosund11113 Version 6.6.0.5(b25275f-basic)-2_5_2
00:00:00 WIF: Connecting to AP1 SECRET1 in mode 11N as SmartHomeGosund11113...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on SmartHomeGosund11113 with IP address 192.168.178.87
19:05:25 RSL: INFO1 = {"Module":"BlitzWolf SHP","Version":"6.6.0.5(b25275f-basic)","FallbackTopic":"cmnd/SmartHomeGosund11113_fb/","GroupTopic":"sonoffs"}
19:05:25 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"SmartHomeGosund11113","IPAddress":"192.168.178.87"}
19:05:25 RSL: INFO3 = {"RestartReason":"Power on"}
19:05:25 RSL: RESULT = {"POWER":"OFF"}
19:05:25 RSL: POWER = OFF
19:05:33 RSL: STATE = {"Time":"2019-08-17T19:05:33","Epoch":1566065133,"Uptime":"0T00:00:14","UptimeSec":14,"Vcc":3.386,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":0,"Wifi":{"AP":1,"SSId":"SECRET1","BSSId":"02:9F:XX:XX:XX:XX","Channel":11,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:05"}}
19:05:33 RSL: SENSOR = {"Time":"2019-08-17T19:05:33","Epoch":1566065133,"ENERGY":{"TotalStartTime":"2019-08-15T20:43:21","Total":0.004,"Yesterday":0.002,"Today":0.002,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
19:05:39 CMD: status 0
19:05:39 RSL: STATUS = {"Status":{"Module":0,"FriendlyName":["SmartHomeGosund11113"],"Topic":"Gosund11113","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:05:39 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/020300/sonoff-basic.bin","RestartReason":"Power on","Uptime":"0T00:00:20","StartupUTC":"2019-08-17T18:05:19","Sleep":50,"CfgHolder":4617,"BootCount":57,"SaveCount":208,"SaveAddress":"FA000"}}
19:05:39 RSL: STATUS2 = {"StatusFWR":{"Version":"6.6.0.5(b25275f-basic)","BuildDateTime":"2019-08-16T19:10:04","Boot":4,"Core":"2_5_2","SDK":"2.2.1(cfd48f3)"}}
19:05:39 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SECRET1","SECRET2"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","280500000100060000005A00000000000000","00000000"]}}
19:05:39 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":460,"Free":540,"Heap":29,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","8F082383","003083A0","20021706","010001C0","00000001"],"Sensors":[3,4]}}
19:05:39 RSL: STATUS5 = {"StatusNET":{"Hostname":"SmartHomeGosund11113","IPAddress":"192.168.178.87","Gateway":"192.168.178.1","Subnetmask":"255.255.254.0","DNSServer":"192.168.178.220","Mac":"B4:E6:XX:XX:XX:XX","Webserver":2,"WifiConfig":4}}
19:05:39 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sat Aug 17 18:05:39 2019","Local":"Sat Aug 17 19:05:39 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:45","Sunset":"20:03"}}
19:05:39 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
19:05:39 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-08-17T19:05:39","Epoch":1566065139,"ENERGY":{"TotalStartTime":"2019-08-15T20:43:21","Total":0.004,"Yesterday":0.002,"Today":0.002,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
19:05:39 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-08-17T19:05:39","Epoch":1566065139,"Uptime":"0T00:00:20","UptimeSec":20,"Vcc":3.384,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":0,"Wifi":{"AP":1,"SSId":"SECRET1","BSSId":"02:9F:XX:XX:XX:XX","Channel":11,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
CONSOLE OUTPUT HERE:
Normal operation:
```
00:00:00 CFG: Loaded from flash at F8, Count 226
00:00:00 Project sonoff SmartHomeGosund11113 Version 6.6.0.5(b25275f-basic)-2_5_2
00:00:00 WIF: Connecting to AP1 SECRET1 in mode 11N as SmartHomeGosund11113...
00:00:05 WIF: Connected
[... cut ...]
19:11:44 RSL: STATE = {"Time":"2019-08-17T19:11:44","Epoch":1566065504,"Uptime":"0T00:00:14","UptimeSec":14,"Vcc":3.399,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"SECRET1","BSSId":"02:9F:XX:XX:XX:XX","Channel":11,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:05"}}
19:11:44 RSL: SENSOR = {"Time":"2019-08-17T19:11:44","Epoch":1566065504,"ENERGY":{"TotalStartTime":"2019-08-15T20:43:21","Total":0.004,"Yesterday":0.002,"Today":0.002,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
19:11:44 HTP: Console
```

ascillato, 2019-08-17:
> 19:11:36 RSL: INFO3 = {"RestartReason":"External System"}

Did you unplug it and plug it again?

If not, the issue is the power regulator of the device. The reason **External System** is that the ESP8266 was left without power. Not a crash.

heikoh81, 2019-08-17:
No, I didn't unplug the Smart Plug. Continous power 230V all the time (German power grid, so very reliable).

Would the log file continue even if the complete system crashes and reboots?

So, to conclude: I bought a bad batch, return to Amazon for refund?

I did some further stress tests with my "original Gosund SP111" and managed to reproduce the above behaviour, but it occured only once after many relais toggles.

ascillato, 2019-08-17:
> So, to conclude: I bought a bad batch, return to Amazon for refund?

Yes, exactly. Sorry about your hardware issue, but sometimes that happens.

> I did some further stress tests with my "original Gosund SP111" and managed to reproduce the above behaviour, but it occured only once after many relais toggles.

If you can't get a refund, you can make a little hack and add an electrolitic capacitor of 1mF or 10mF between 3.3VCC and GND. That should give you more energy for switching several times in a row without losing power. Or a better solution is to replace the power regulator.

heikoh81, 2019-08-17:
Thanks for your reply.
My soldering experience is not that good to add an capacitor in such a small device (-> tuya-convert).

However, I rarely switch several times within seconds, so probably this isn't that big a problem.
On Plugs 1 and 2, I have noticed a crash even after hours of just ideling, so maybe the PSU is defective in those two.

I ordered 2x3 Jisusu SP111 on Amazon and I have 25 days left to return them, so no hurry.
Plug 3 of 6 seems to work more reliable, so probably I won't have to return all of them.

ilgrank, 2019-08-20:
Hi
Just chiming in to tell you I'm having the exact same issue with a genuine Blitzwolf SHP6.
I also tried adding a capacitor (tried with 10mF and 300mF) but nothing changed.

heikoh81, 2019-08-21:
Thanks for your reply. My answer is here: https://github.com/arendst/Sonoff-Tasmota/issues/6209#issuecomment-523491288