arendst/Tasmota #6252: 6.5.0.5 LED Issues

DavidFW1960, 2019-08-18:
The fix for the LED.... Previously, if the switch was on, the red and blue LED were on. With the overnight fix, Home Assistant is controlling the switch and showing status correctly again but the blue LED is off - only the red one was on. I believe the blue one flashes if the switch is turned on but not connected to WiFi.. so now there is only the red LED working indicating if the switch is on or off.

Is this a further bug in the firmware of is there a setoption I need to be aware of?

Both switches have LedMask FFFF, LedPower Off, LedState 1

ascillato, 2019-08-18:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you? Which device are you using? etc etc.

Have you tried ledlink command?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2019-08-18:
Solved in Discord by @meingraham help, setting GPIO as ledlink. Thanks.

closed 2019-08-18 by ascillato2