arendst/Tasmota #6253: BH1750 SHT: Sensor did not ACK command

bakman2, 2019-08-18:
### BUG DESCRIPTION
Hardware:
Wemos D1 mini v3
BME280
BH1750 
Connected on D2 GPIO4 (SDA) and D1 GPIO5 (SCL)

When restarting the Wemos, the BME280 is detected, the BH1750 is not.
Console states: `SHT: Sensor did not ACK command`.
`SENSOR` only shows BME280.

Seriallog is 0.

About 1 minute later (teleperiod?), the console shows `BH1750 found at 0x23` and now shows up in the `SENSOR`

Not a breaking issue as it will always find it, but should show up on boot.

- [x] Provide the output of command ``status 0`` :
```
09:57:47 MQT: devices/balkon/dht/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["wemos-DHT"],"Topic":"devices/balkon/dht","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:57:47 MQT: devices/balkon/dht/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:12","StartupUTC":"2019-08-18T07:57:35","Sleep":250,"CfgHolder":4617,"BootCount":30,"SaveCount":60,"SaveAddress":"F8000"}}
09:57:47 MQT: devices/balkon/dht/stat/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
09:57:47 MQT: devices/balkon/dht/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["APAC02",""],"TelePeriod":60,"Resolution":"D58180C0","SetOption":["00008009","280500000100060000000000000000000000","00000000"]}}
09:57:47 MQT: devices/balkon/dht/stat/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":15,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
09:57:47 MQT: devices/balkon/dht/stat/STATUS5 = {"StatusNET":{"Hostname":"wemos","IPAddress":"10.0.0.30","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.6","Mac":"CC:50:E3:C4:9B:33","Webserver":2,"WifiConfig":4}}
09:57:47 MQT: devices/balkon/dht/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.6","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C49B33","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
09:57:47 MQT: devices/balkon/dht/stat/STATUS7 = {"StatusTIM":{"UTC":"Sun Aug 18 07:57:47 2019","Local":"Sun Aug 18 09:57:47 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+02:00","Sunrise":"06:47","Sunset":"21:01"}}
09:57:47 MQT: devices/balkon/dht/stat/STATUS10 = {"StatusSNS":{"Time":"2019-08-18T09:57:47","BME280":{"Temperature":20.370,"Humidity":70.8,"Pressure":1002.1},"PressureUnit":"hPa","TempUnit":"C"}}
09:57:47 MQT: devices/balkon/dht/stat/STATUS11 = {"StatusSTS":{"Time":"2019-08-18T09:57:47","Uptime":"0T00:00:12","Heap":15,"SleepMode":"Dynamic","Sleep":250,"LoadAvg":3,"Wifi":{"AP":1,"SSId":"APAC02","BSSId":"44:D9:E7:91:E1:29","Channel":9,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:03"}}}
09:57:48 MQT: devices/balkon/dht/tele/STATE = {"Time":"2019-08-18T09:57:48","Uptime":"0T00:00:13","Heap":15,"SleepMode":"Dynamic","Sleep":250,"LoadAvg":7,"Wifi":{"AP":1,"SSId":"APAC02","BSSId":"44:D9:E7:91:E1:29","Channel":9,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:03"}}
09:57:48 MQT: devices/balkon/dht/tele/SENSOR = {"Time":"2019-08-18T09:57:48","BME280":{"Temperature":20.360,"Humidity":70.8,"Pressure":1002.1},"PressureUnit":"hPa","TempUnit":"C"}
```
**Note** this is after a restart, once sensor is detected it shows up here too.

### EXPECTED BEHAVIOUR
Detection of sensor on boot.

### ADDITIONAL CONTEXT
```
00:00:00 CFG: Loaded from flash at FA, Count 58
00:00:00 SRC: Restart
00:00:00 Project sonoff wemos-DHT Version 6.6.0(release-sonoff)-2_3_0
00:00:00 SHT: Sensor did not ACK command
00:00:00 I2C: BME280 found at 0x76
....
09:49:06 MQT: devices/balkon/dht/tele/SENSOR = {"Time":"2019-08-18T09:49:06","BME280":{"Temperature":20.240,"Humidity":71.6,"Pressure":1002.3},"PressureUnit":"hPa","TempUnit":"C"}
...
09:49:19 CMD: i2cscan
...
09:49:19 MQT: devices/balkon/dht/stat/RESULT = {"I2CScan":"Device(s) found at 0x23 0x76"}
...
09:50:06 MQT: devices/balkon/dht/tele/SENSOR = {"Time":"2019-08-18T09:50:06","BME280":{"Temperature":20.260,"Humidity":71.7,"Pressure":1002.2},"PressureUnit":"hPa","TempUnit":"C"}
...
09:50:23 I2C: BH1750 found at 0x23
...
09:51:06 MQT: devices/balkon/dht/tele/SENSOR = {"Time":"2019-08-18T09:51:06","BME280":{"Temperature":20.310,"Humidity":71.6,"Pressure":1002.1},"BH1750":{"Illuminance":2087},"PressureUnit":"hPa","TempUnit":"C"}
```

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : WEMOS 1D Mini V3
- [x] Tasmota binary firmware version number used : 6.6.0(release-sonoff)-2_3_0

ascillato, 2019-08-18:
If you connect just this sensor alone, do you experience the same behavior?

bakman2, 2019-08-20:
> If you connect just this sensor alone, do you experience the same behavior?

I cannot really test as I have soldered it into a package. But before I put it all together/tasmota it worked instantly with arduino code.

ascillato2, 2019-08-21:
Which GPIOs are you using?

ascillato2, 2019-08-22:
Please, can you try latest Tasmota 6.6.0.6 with core 2.5.2 (http://thehackbox.org/tasmota/020502/sonoff.bin) ?

ascillato, 2019-08-24:
@bakman2 

Hi,

Any news on this?

bakman2, 2019-08-24:
> @bakman2
> 
> Hi,
> 
> Any news on this?

No difference.

mike2nl, 2019-09-05:
@bakman2 @ascillato 
When i'm back from the hospital i will look in the sensor driver.
It is possible that the bh1750 needs more time depending on the quality of the sensor itslef.

  if (90 == (uptime %100)) {
    // 1mS
    Bh1750Detect();
  }
Possible is that to short when more i2c sensors are used. I will try to messaure
the senor alone and both together. Hope i will find something.

mike2nl, 2019-09-05:
@ascillato 
can we talk about that issue on discord channel audio?

ascillato, 2019-09-05:
@mike2nl 

Great findings!!!

>can we talk about that issue on discord channel audio?

I would love to but today I'm not at home.

closed 2019-10-05 by stale[bot]