arendst/Tasmota #6254: Support for AM2320 sensor? somebody did it

nicomania, 2019-08-18:
Hello,

could you please merge this driver for the AM2320 (I2C) Sensor? Thank you!
[https://github.com/lrswss/tasmota-am2320-i2c-driver](https://github.com/lrswss/tasmota-am2320-i2c-driver)

Best regards

ascillato, 2019-08-18:
Hi,

Have you tried it? In that repository there is an opened/unresolved issue that said that the driver don't compile.

joba-1, 2019-08-18:
at a first look it seems the errors are only within #defines for domoticz
or knx. Maybe the author just did not care about these.

Looks like very easy to fix if there is interest:

remove the &&(0==i) and replace Dht[i].* with the actual variabls used for
temp and hum...

nicomania, 2019-08-18:
Hi, 
thanks for the quick reply. No, I did not try it, because I am just an enduser. I am not familiar with compiling. 
But I just wanted to point out, that it would probably be a good benefit with relatively less effort to integrate into tasmota. Me and maybe others too, would be happy about it.
I hope the reworking is as simple as my forespeaker wrote.

Best regards!

ascillato2, 2019-08-21:
@joba-1 

If you are interested on supporting this sensor, would be awesome if you could make a PR for this. Thanks.

joba-1, 2019-08-21:
not really. 

this is what Adafruit says to it:

The datasheet mentions it has 3% humidity accuracy and 0.5C temperature accuracy, but we're not very trusting of the specifications. So, while this sensor does seem to work, it's not recommended for anything where you care deeply about guaranteed accuracy. Temperature is probably correct to 2-3 degrees Celsius. Humidity is probably within 5-10%.

and they are selling that thing...

@nicomania I can fix the errors I spotted and give you the result to test. Maybe that already works for you?

ascillato2, 2019-08-21:
Added label requested feature (hold over) for future reference. Thanks

closed 2019-08-21 by ascillato2

nicomania, 2019-08-22:
> not really.
>
> this is what Adafruit says to it:
>
> The datasheet mentions it has 3% humidity accuracy and 0.5C temperature accuracy, but we're not very trusting of the specifications. So, while this [...]

Well, the accurancy is not the point. This sensor is just cheap and does the job for certain cases.
Thank you for the offer, but I do not want to miss the OTA update function.

joba-1, 2019-08-22:
Well, just checked on Aliexpress and found one starting at 2.10€.
A very good BME280 I found for 2.48€ even with excellent pressure measurement.
Both including shipping. I cannot work long for that difference, so I usually use BMEs.

arendst, 2019-08-22:
Totally agree. It's a waste of effort and code space.

Issue closed.

nicomania, 2019-08-23:
Got it for 0.65$ on Ebay. But I canot understand, that you rely on a single statement from Adafruit. Regarding the code space, the Am2320 seems to be more accurate, than other Sensors supported by Tasmota: [http://www.kandrsmith.org/RJS/Misc/Hygrometers/calib_many.html](http://www.kandrsmith.org/RJS/Misc/Hygrometers/calib_many.html)

Thank you anyway for your thoughts.