arendst/Tasmota #6266: Device resets to defaults when configuring sensor MHZ 19

Jason2866, 2019-08-20:
### BUG DESCRIPTION
Device revert to standard settings (GPIO config lost) when selecting and saving
[image]

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Sonoff Basic
- [x] Tasmota binary firmware version number used : 6.6.0.6 / self-compiled
- [x] Development IDE - Compiler / Upload tools used : Platformio / OTA
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
Not usefull since device comes back with standard Sonoff Basic Settings

```

### TO REPRODUCE
Connect and select a MHZ 19 to GPIO1 - MHZ Tx (60) and GPIO3 - MHZ Rx (61) 

### EXPECTED BEHAVIOUR
Working Sensor not resetting device ;-)

arendst, 2019-08-20:
Verified. Will check the day after tomorrow ;-)

ascillato2, 2019-08-21:
Fixed by Theo. Can you test it again?

Jason2866, 2019-08-21:
Does work
So `ToHex_P` is the problem? Maybe on other places too?

closed 2019-08-21 by Jason2866

arendst, 2019-08-21:
tohex_p is the problem when the input data is zero. In the case of mhz19 the initial serial data is zero so tohex_p fails.

I will add the test tomorrow and make a final fix.

Jason2866, 2019-08-22:
Final fix does work.