arendst/Tasmota #6268: Device Switching off by itself after physically turning on. Stays on if toggled through tasmota

blacktirion, 2019-08-20:
Hello,

I recently published the template for the WF15S (Monoprice Stitch) https://blakadder.github.io/templates/monoprice_stitch_paddle.html

I am having issues, as I didn't realize that the switch will turn itself off after a few minutes when the on button is pressed physically. I have confirmed that it will stay on for as long as you want if you toggle it through the Sonoff Interface.

I reviewed the manufacturer's guide and didn't see an auto sleep, so I'm at a loss here. I've factory reset the entire device, tried changing pinouts, but the only way to get it to stop is to not assign the Tuya RX (108) to GPIO 13. However, that is required to be assigned to get signal from the chip that it has been turned on physically.

Here's some log captures that I was able to get:

`22:24:23 TYA: RX Packet: "55aa0007000501010001000e"
22:24:23 TYA: RX - Off State **(Physically Turned OFF, to get this in a proper State)**
22:24:23 SRC: Switch
22:24:23 TYA: Send Dim Level skipped due to 0 or already set. Value=0
22:24:23 RSL: stat/sonoff/RESULT = {"POWER":"OFF"}
22:24:23 RSL: stat/sonoff/POWER = OFF
22:24:23 CFG: Saved to flash at FA, Count 202, Bytes 3584
22:24:32 WIF: Checking connection...
22:24:32 WIF: Connected
22:24:36 TYA: RX Packet: "55aa0007000501010001010f"
22:24:36 TYA: RX - On State **(Physically Turned ON)**
22:24:36 SRC: Switch
22:24:36 TYA: Send Serial Packet Dim Value=145 (id=6)
22:24:36 TYA: TX Packet: "55aa000600080602000400000091aa"
22:24:36 RSL: stat/sonoff/RESULT = {"POWER":"ON"}
22:24:36 RSL: stat/sonoff/POWER = ON
22:24:36 TYA: RX Packet: "55aa000700080602000400000091ab"
22:24:36 TYA: RX Dim State=145
22:24:37 CFG: Saved to flash at F9, Count 203, Bytes 3584
22:26:42 RSL: tele/sonoff/STATE = {"Time":"2019-08-20T22:26:42","Uptime":"0T00:05:19","Vcc":3.217,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":539,"POWER":"ON","Dimmer":57,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"XXXXX","BSSId":"XX:XX:XX:XX:XX:XX","Channel":11,"RSSI":90,"LinkCount":1,"Downtime":"0T00:00:08"}}
22:26:52 WIF: Checking connection...
22:26:52 WIF: Connected
22:27:01 TYA: RX Packet: "55aa0007000806020004000000001a" 
22:27:01 TYA: RX Dim State=0
22:27:01 TYA: RX Packet: "55aa0007000501010001000e"
22:27:01 TYA: RX - Off State **(Turned off automatically)**
22:27:01 SRC: Switch
22:27:01 TYA: Send Dim Level skipped due to 0 or already set. Value=0
22:27:01 RSL: stat/sonoff/RESULT = {"POWER":"OFF"}
22:27:01 RSL: stat/sonoff/POWER = OFF
22:27:01 CFG: Saved to flash at F8, Count 204, Bytes 3584`

I even tried turning off Dynamic Sleep.

I am not sure if this has something to do with the dimmer settings on this template, but I was not able to figure out how to turn off the dimmer settings for it and this seems like the only template that works with my switch. Here is the template on the template database:
https://blakadder.github.io/templates/monoprice_stitch_paddle.html

I have figured out that I don't need assignments on GPIO 2 or 3, as the assignments on GPIO for Tuya RX/TX work on GPIO 13 and 15

blacktirion, 2019-08-21:
Issue appears to be fixed in 6.6.0.6. I believe I still saw it in 6.6.0. I was running 5.5.0 during the initial phases. I built code from https://github.com/arendst/Sonoff-Tasmota/pull/6265 which fixes any dimmer slider issues as well. Will keep monitoring for a few days, and close issue if it does not resurface.

ascillato2, 2019-08-21:
Hi,

Thanks for reporting. Also it is great to know that updating solved your issue. Thanks

closed 2019-08-21 by ascillato2