arendst/Tasmota #6269: Reliability 6.6.0 on Sonoff Basic

scargill, 2019-08-20:
Hi theo...

I've used Tasmota for some time, had a break and come back to it. I'm using NodeMCU to firstly WIPE sonoff BASIC (and Basic R3) and then put Tasmota 6.5.0 on them with an FTDI. This eems reliable, I've done several now, but when I upgrade to 6.6.0 I think they take lkonger to reboot, occasionally need a power cycle after any changes... is there anything about these boards and 6.6.0 I should know about? I'm flashing using 3v3 but then as soon as I get  basic web connection moving to mains power.  Also at one point I woul slways see Tasmota Sonoffs by name on my network (I could not manage this with my own code) but now can ONLY see them by IP address (typical name smart-1305) on my Windows 10 64 bit PC. I've tried mqtt topic cmnd/smart-1305/setoption55  and payload 1 (numeric) to turn on mDNS discovery and that didnt help.

Any thoughts?

ascillato2, 2019-08-20:
Hi, 

>is there anything about these boards and 6.6.0 I should know about?

No, there is no known issue like that. All my devices are workings fine with latest Tasmota 6.6.0.5

For your issue with longer reboots seems to be a network problem but in order to know exactly what is going on in your setup please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2019-08-21:
Also remember that you can address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4) if you like. The chat is a better and more dynamic channel for helping you.

Thanks.

closed 2019-08-22 by ascillato2