arendst/Tasmota #6270: I2C + BME280 + Sonoff Basic

edgiru, 2019-08-21:
Sonoff-Tasmota 6.6.0.6

Help, I already tried everything.
The data from the BME280 sensor is shown when the relay is off.
[image]
[image]

If I turn on the relay, data is lost.
[image]
[image]

ascillato, 2019-08-21:
Seems to be a power issue. Are you powering this sensor from the sonoff basic? If yes, that is not going to work. A sonoff basic is not designed to power anything else than itself.

edgiru, 2019-08-21:
> Seems to be a power issue. Are you powering this sensor from the sonoff basic? If yes, that is not going to work. A sonoff basic is not designed to power anything else than itself.

Thanks for answering.

no decrease in voltage is observed.
mqtt: "Vcc" 3.481v
on contacts bme280: 3.316v

Jason2866, 2019-08-21:
Just measuring static voltage values doesnt help. You wont see short drops or is it a good DC...
There is no known problem. You can try to add a cap. 
Sonoff Basics are known to have a very weak power supply...

edgiru, 2019-08-21:
> Just measuring static voltage values doesnt help. You wont see short drops or is it a good DC...
> There is no known problem with I2C with v.6.6.0.6. and before. You can try to add a cap.
> Sonoff Basics are known to have a very weak power supply...

connected to the BME280 stabilized external power 3.3v
The same problem did not help. I turn off the relay data from the BME280 appear.

Jason2866, 2019-08-21:
Please try release version 6.5.0. There are a lot of changes between 6.5.0 and 6.6.0.6
Maybe code refactoring introduced a error. If 6.5.0. reacts the same way it is your hardware...

edgiru, 2019-08-21:
Work release version 6.2.1

[image]

Jason2866, 2019-08-21:
Please post `Status 0` output

edgiru, 2019-08-21:
> Please post `Status 0` output
where to look?

> And relay is off! You said you get no values when relay is on
Work!!! release version 6.5.0

Relay Enable
[image]
[image]

Relay Disable
[image]

Jason2866, 2019-08-21:
Maybe it is solved too, when Theo fixes the bug #6266 (bug with GPIO 1 and GPIO 3)

ascillato2, 2019-08-21:
@edgiru 

Can you try latest Tasmota from today? Thanks.

ascillato2, 2019-08-21:
remember to set `seriallog` to `0` (you can do that from the tasmota console by command `seriallog 0` or in the webUI)

arendst, 2019-08-22:
Pls provide output of command ``status 0`` when using v6.6.0.6.

> Where to look

You might want to look at the console window and type ``status 0``. This should show what we want.

edgiru, 2019-08-22:
> Pls provide output of command `status 0` when using v6.6.0.6.
Hello!
Work!!  v6.6.0.6

Serial log: off

> 16:37:03 CMD: status 0
16:37:03 SRC: WebConsole from 192.168.1.55
16:37:03 MQT: stat/cabinet/STATUS = {"Status":{"Module":18,"FriendlyName":["cabinet"],"Topic":"cabinet","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:37:03 MQT: stat/cabinet/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:08","StartupUTC":"2019-08-22T13:35:55","Sleep":50,"CfgHolder":4617,"BootCount":10,"SaveCount":59,"SaveAddress":"F8000"}}
16:37:03 MQT: stat/cabinet/STATUS2 = {"StatusFWR":{"Version":"6.6.0.6(944cf38-sonoff)","BuildDateTime":"2019-08-22T13:02:02","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
16:37:03 MQT: stat/cabinet/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["oc",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["01008009","280500000100060000005A00000000000000","00000000"]}}
16:37:03 MQT: stat/cabinet/STATUS4 = {"StatusMEM":{"ProgramSize":524,"Free":476,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","8FDAE397","003683A0","22B617CD","01001BC0","00000001"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
16:37:03 MQT: stat/cabinet/STATUS5 = {"StatusNET":{"Hostname":"cabinet-2218","IPAddress":"192.168.1.81","Gateway":"192.168.1.3","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.4","Mac":"2C:3A:E8:07:A8:AA","Webserver":2,"WifiConfig":4}}
16:37:03 MQT: stat/cabinet/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.96","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_07A8AA","MqttUser":"edgi","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
16:37:03 MQT: stat/cabinet/STATUS7 = {"StatusTIM":{"UTC":"Thu Aug 22 13:37:03 2019","Local":"Thu Aug 22 16:37:03 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+03:00","Sunrise":"07:52","Sunset":"21:53"}}
16:37:03 MQT: stat/cabinet/STATUS10 = {"StatusSNS":{"Time":"2019-08-22T16:37:03","Epoch":1566481023,"BME280":{"Temperature":26.6,"Humidity":47.7,"Pressure":759.5},"PressureUnit":"mmHg","TempUnit":"C"}}
16:37:03 MQT: stat/cabinet/STATUS11 = {"StatusSTS":{"Time":"2019-08-22T16:37:03","Epoch":1566481023,"Uptime":"0T00:01:08","UptimeSec":68,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"oc","BSSId":"D4:CA:6D:1D:05:89","Channel":7,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}}
16:37:09 MQT: tele/cabinet/STATE = {"Time":"2019-08-22T16:37:09","Epoch":1566481029,"Uptime":"0T00:01:14","UptimeSec":74,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"oc","BSSId":"D4:CA:6D:1D:05:89","Channel":7,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}
16:37:09 MQT: tele/cabinet/SENSOR = {"Time":"2019-08-22T16:37:09","Epoch":1566481029,"BME280":{"Temperature":26.6,"Humidity":47.8,"Pressure":759.4},"PressureUnit":"mmHg","TempUnit":"C"}
16:37:19 MQT: tele/cabinet/STATE = {"Time":"2019-08-22T16:37:19","Epoch":1566481039,"Uptime":"0T00:01:24","UptimeSec":84,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"oc","BSSId":"D4:CA:6D:1D:05:89","Channel":7,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}
16:37:19 MQT: tele/cabinet/SENSOR = {"Time":"2019-08-22T16:37:19","Epoch":1566481039,"BME280":{"Temperature":26.6,"Humidity":47.8,"Pressure":759.4},"PressureUnit":"mmHg","TempUnit":"C"}
16:37:19 WIF: Checking connection...

edgiru, 2019-08-22:
another problem!
RSSI: 100
It always shows 100 before this figure has changed. Wifi hotspot across multiple walls. so it should show about 80 approximately.

arendst, 2019-08-22:
So 

> Work!! v6.6.0.6

means all is working fine now?

If so please close this issue.

ascillato, 2019-08-22:
I have just tested latest Tasmota version with core 2.4.2 and RSSI is being measured correctly.

ascillato2, 2019-08-22:
> Hello!
> Work!! v6.6.0.6

Great to know that now your setup is working fine. Thanks.

closed 2019-08-22 by ascillato2