arendst/Tasmota #6272: Accessing sonoff basic wia wifi as AP with no router

jetema, 2019-08-21:
Hello , I love Tasmota, I have several sonoff working perfect, with no error at all, so I have one question/improvement : 
Would it be possible accesing via web interface also activating/deactivating the rele in sonoff with tasmota wia wifi as AP mode ???  
I know  ESP8266 has this working mode, the aim of this idea is using a cell phone as a remote control in rural areas with no internet neither router, just by using a single sonoff flashed with tasmota as point to point control between cell phone and sonoff.
Many thanks in advance

meingraham, 2019-08-21:
@jetema 

Yes, requested before... but it's not going to be added. There are other custom firmware that provide this capability. You can try using one of those on the device(s) you want this capability.

There are many inexpensive "travel routers" that can be deployed to set up a LAN. Then you have the regular Tasmota web UI functionality for each device.

Note that if Tasmota cannot set its time (usually via access to an NTP time server on the Internet), any configured timers will not work. There are solutions, but you need to consider this if you want to have timers.

Mike

ascillato2, 2019-08-21:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-08-21 by ascillato2

mike2nl, 2019-08-26:
@jetema 
then set up your sonoff with fixed ip address or dhcp, set your mobile phne in AP mode and you can access the sonoff via every browser from pc with wifi, so far all arte in the range. We/i have done this in industrial envoiroment to get information.

jetema, 2019-08-26:
Many thanks , but as meingraham said before with tasmota actually it is no possible switching the sonoff's relay accessing sonoff as AP.
The aim is, for instance, opening a garage door in a place where there is no internet available.
I know if I could use a cheap router to create Wlan ( even with no internet ), anyway I wanted point to point as simple as posible, I will have a look at ESP8266 01 and a relay board connected to gpio 0 or 2