arendst/Tasmota #6277: T[wo]o Generic Modules?

joba-1, 2019-08-22:
I was a bit puzzled about finding two Generic modules in the device config list (0 and 18).

One seems to be related to Templates and one to plain ESP8266.

They look very similar and served the purpose of defining my i2c pins.

So, if there really is a difference, I suggest to name them differently so proper wiki documentation can be found by the name.

arendst, 2019-08-22:
The default template IS the Generic module template and they are equal as it is just a copy.

A user can rename the user template to what ever he likes.

Any difference is initiated by a user.

joba-1, 2019-08-22:
so 0 is a copy of 18 and can be changed and 18 is fixed?

Then I would suggest to name it something like 18: „Generic“ or „ESP8266“ and 0 as „Template“.
Maybe even a combi like „Generic ESP8266“ and „ESP8266 Template“

joba-1, 2019-08-22:
my point is, if they are different (like fixed an base for changes) then the name should be different as well, not just the number

arendst, 2019-08-22:
If they are different a user has made them different. He then can also change the name.

They are NOT different initially. It's all under user control.

joba-1, 2019-08-22:
I still think they are different, even with default config. Not regarding pin definitions or name, but 0 can be (and is intended to be) changed, 18 cannot. And that is not obvious for an unwary end user that does not think about or need templates.
It is not dangerous because technically the device will work the same, but it is still confusing and needs wiki study if you want to be sure. If 0 would have had a default name of Template or something like that, I would have chosen Generic for my usecase and be done with it.

I know from own experience this user view can easily elude the developer that fully understands whats going on behind the scenes.
But I‘m sure you are sufficiently aware of this now, so I‘ll close the issue.

Thanks!

closed 2019-08-22 by joba-1