arendst/Tasmota #6279: NodeMCU restart if button inverted

khriss75, 2019-08-22:
### BUG DESCRIPTION
Uploaded latest TASMOTA firmware (.bin, already compiled) on NODEMCU ESP8266.
Configuration:
D3->BUTTON1
D4->BUTTON2
D2->BUTTON3
D1->BUTTON4
D6->RELAY1
D7->RELAY2
D5->RELAY3
D8->RELAY4
No other configuration (except wifi connection).
In this configuration, if I push buttons, the relays state change only when I release button.
Configurate buttons as inverted (Buttonxi). Relays change state when i push button (as I like).
After some seconds Nodemcu reset and enter in AP mode (mode for wifi configuration... 192.168.4.1...). Relay2 blink.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ X] Device used (i.e. Sonoff Basic) : NODEMCU (ESP8266)
- [ X] Tasmota binary firmware version number used : 6.6.0 / PRE-COMPILED
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` : 
```
STATUS 0 OUTPUT HERE:
ESP8266_01/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff","Sonoff2","Sonoff3","Sonoff4"],"Topic":"ESP8266_01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```

### TO REPRODUCE
Upload pre compiled 6.6.0 tasmota on NODEMCU. Set ssid and wifi password. Configure as Generic module and:
D3->BUTTON1i
D4->BUTTON2i
D2->BUTTON3i
D1->BUTTON4i
D6->RELAY1
D7->RELAY2
D5->RELAY3
D8->RELAY4

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
Could be that button inverted on GPIO0 (D3 pin) can rebot Nodemcu?

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-08-22:
Hi,

Not a bug. It is a misconfiguration of your device.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-08-22 by ascillato2

khriss75, 2019-08-22:
Issue closed because not a bug