arendst/Tasmota #6280: Sonoff.bin #define SUPPORT_MQTT_EVENT

VitaliiDolgov, 2019-08-22:
Please post file for downloading Sonoff.bin with active function #define SUPPORT_MQTT_EVENT as it is important function which allows to subscribe topics. I spent one week to change this line using Atom and VSC with platformio plugin but without success. Please call the file as sonoff_MQTT_event.bin. I think there are many people who needs the same. Thanks!

ascillato2, 2019-08-22:
For help on how to compile Tasmota, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-08-22 by ascillato2

meingraham, 2019-08-22:
@VitaliiDolgov is this a request to provide pre-compiled binaries with this feature (SUPPORT_MQTT_EVENT) enabled? It's not a commonly requested feature which is why it is not enabled as a default feature. For those that need something custom, the simplest means to compile one's own firmware is using [Gitpod](https://github.com/arendst/Sonoff-Tasmota/wiki/Compiling-Tasmota-on-Gitpod). There is nothing to install. One compiles the code, online, directly from the Tasmota GitHub repository and then download the resulting binary firmware file.

This Issue is now rightly closed so please direct future questions on this to the Discord chat.

Regards.

Mike

jeylites, 2019-12-08:
Hi, 

How does SUPPORT_MQTT_EVENT work or what is it's function? I can't seem to find documentation about it on wiki. 

I'm trying to have client device (A) to send a MQTT message to client device (B) when (A)is turned ON. Where should I be look at to get this going?

Thank you.
JK