arendst/Tasmota #6285: Has Detected An IP Address Conflict

edgiru, 2019-08-23:
Hello!
firmware v6.6.0.6
Consider adding the IP address conflict detection function to the firmware. Faced a problem in a large enterprise network. After 10 seconds, the web interface hung. I thought for a long time to reflash the module 20 times, I thought the hardware problem. Then it occurred to me that maybe there is already a computer on the network or a device using the same IP address. changed the IP address for tasmota on the DHCP server, and the module began to work stably.

thanks

mprins, 2019-08-23:
the real question is why your dhcp server assigned an IP adress that was already in use. 
dhcp clients cannot "see" if an IP address is in use in a network until they have a valid IP address for that network, since they then have that ip address assigned - from an authorative source- there is no way to see if there is another device with the same IP address

what you ask seems impossible.

ascillato2, 2019-08-23:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-08-23 by ascillato2

edgiru, 2019-08-23:
> the real question is why your dhcp server assigned an IP adress that was already in use.
> dhcp clients cannot "see" if an IP address is in use in a network until they have a valid IP address for that [...]

the client had a static ip address for this DHCP server did not know that this ip was already taken, when he assigned the tasmota ip device, a conflict naturally arose. Somehow windows detects this conflict and informs.

Jason2866, 2019-08-24:
If you assign static IP addresses the only one who is responsible is you ;-)
ESP8266 is a very limited device in every aspect. There is no room for such features...

s-hadinger, 2019-08-24:
There seem to be an ARP conflict detection algorithm in Core Arduino, but I'm nor sure how it would surface in logs. Anyways the only option would be to log on serial, which is probably not convenient - you don't want to keep a pc connected by serial to all devices. 

Best option is always DHCP if you don't want headaches. Even for static IPs. My wifi router DHCP allows to define static IPS to Mac addresses.