arendst/Tasmota #6286: Migrate IRremoteESP8266-2.6.0 to IRremoteESP8266-2.6.4

hristo-atanasov, 2019-08-24:
There are newer HVACs included and it supports more than more than 64 bits code receiving

ascillato2, 2019-08-24:
Great Thanks!

Please, do a Pull Request.

Thanks!

closed 2019-08-24 by ascillato2

s-hadinger, 2019-08-24:
I'm preparing a PR to make it easier to add HVAC devices from IRREMOTEESP8266.  Stay tuned. 

@arendst Would you be Ok to publish a firmware dedicated to IR blasters, including (almost) all protocols from IRREMOTEESP8266? I know they use a lot of space but if we trim down the code space for IR it's manageable.

arendst, 2019-08-24:
Dedicated IR. Give it a try.

hristo-atanasov, 2019-08-24:
I'm not actually in need of all new HVACS. I'm interested in receiving more bits, as this way your software would become the most powerful tool I have ever seen.  This way everyone will not only have the ability to use Tasmota for controlling ACs ,but in fact could track their state change if the ACs status was changed by its original remote too. This will allow people to fully ingrate all their not so smart devices to smart home hubs like Home Assistant and Open Hab. :) Like my case - I wrote integration for my AC but I can track their state only when they were controlled by HA, if someone change their state with the remote, this change remains hidden from HA. I already have one dedicated NodeMCU for controlling the AC. Adding a second device only for tracking the state would be too much - no place where to hide it anymore .. :)

hristo-atanasov, 2019-08-24:
Well actually your software is already the most powerful for me .. But these additional bits, even if you can add them without migrating to new version of IRremote would be very nice. :) If I have to be honest, initially I decided to not bother you with this, and tried to change the code by myself, but the only thing I achieved was sh*ting my pants (and device) in C (as I don't know much about it) .. :D

Jason2866, 2019-08-24:
Since there are now and then people asking for support all kind of HAVACs it is a great
idea from @s-hadinger to do a special IR build supporting this.

s-hadinger, 2019-08-24:
Use case makes sense, I didn't think about capturing manual changes from the remote.

@arendst One more question. For Hex codes, each byte is sent MSB first which makes manual calculation a bit more complex. Most IR codes actually expect LSB first. I'm not talking about little-endian but how each bit is sent to the IR led. Is there any hard reason to have MSB first instead of LSB? I will had a flag to send LSB first too.

arendst, 2019-08-24:
I guess the only reason for MSB/LSB is the response you get from IrRemote and how it wants to have it's input.

arendst, 2019-08-24:
BTW regarding HVAC decoding you might want to look at the examples with IrRemote.

hristo-atanasov, 2019-08-24:
I still can't get the idea for special build just for HVAC .. It is already integrated .. Just capturing should allow more more bits. :) As for the other HVACS - the code is included in IRREMOTEESP266, but I guess it should be adapted for Tasmota ..

hristo-atanasov, 2019-08-24:
@arendst is it too hard to make it receive up to 128 bit codes with the current version (actually I need 104)?

s-hadinger, 2019-08-24:
It's one thing to capture the bits, it's another to decode/encode it. Recording all combinations of hvac ir codes wouldn't make sense because of combinatorial explosion. 

Please be patient. 

Also ESP8266 does not handle 128 bits integer so it's not a simple patch.

hristo-atanasov, 2019-08-24:
OK, I'm not too much in the ir codes thing, so I'll ask in different way .. Is it too hard to receive this:
0xC0A0000000002000402C60F9C3
instead of this:
                        0x2000402C60F9C3

s-hadinger, 2019-08-28:
@hristo-atanasov Thank you so much for finding an easy way to solve your issue. This is the beauty of open-source, you can solve your problem yourself.

This is not code I wrote so I apologies if I lead you to wrong conclusions. I will keep quiet next time you ask a question and will let you find your solution yourself.

Now the purpose of my discussion above is to find a solution for all users, not just for you.

Jason2866, 2019-08-28:
@hristo-atanasov  you know the fact that all people working on Tasmota are doing that in there free time as volunteers? Keep that in mind when writing here...

hristo-atanasov, 2019-08-28:
@s-hadinger As a person that does not have any knowledge  about C and C++ (and how Tasmota works internally), it took me 30 minutes (testing included), to read part the code of IRremoteESP8266, part of the code in Tasmota and change line 160 in 'xdrv_05_irremote.ino' from:
`snprintf_P(svalue, sizeof(svalue), PSTR("\"%s\""), hvalue);`
to:
` snprintf_P(svalue, sizeof(svalue), PSTR("\"%s\""), resultToHexidecimal(&results).c_str());`

And 'my_user_config.h' Line 429:
`#define IR_RCV_BUFFER_SIZE      100`
to:
`#define IR_RCV_BUFFER_SIZE      250` (222 is enough for me but I made it a little bigger)

Without migrating to the newest release.

You wrote so much excuses and self explanations, that (may be) it would took you way less time for you to search and see what is wrong and why Tasmota truncates the code.
I hope you get better in programming and thinking out-of-the-box.

@arendst Thank you for your great software! :)

hristo-atanasov, 2019-08-28:
@s-hadinger All I wrote is with good feelings and know you're looking for solution for all people.  I didn't ment to affect you. Just tried to explain in a different way what my problem is. Actually my solution is for all people too, because the issue with truncating the code affects all users that are using IRReceive functionality. I only expected help to solve this issue quicker (like where and probably how to I have to change the code), as I don't know anything for C and C++.

@Jason2866 When one conversations does not affect you, or if you can't help, just don't join in it. I can't believe that people working for free are not getting paid and that amaze me (just kidding). As Software Engineer from 7 years and developer from 12 years, I know what is to write opensource software, and I had such too. I can't believe how intelligent human being could make make conclusion from my previous post, that I'm Tasmota expert. Feel free to keep your expertise and knowledge for yourself (I haven asked for it). Please, don't even respond to me! Would be useless waste of words. Thank you!

@meingraham What? You're laughing that I have opened only one issue? Is this even funny?

@Jason2866 @meingraham
Why is that offending here? Do you treat same way other people having difficulties using Tasmota (or questions about it)?
I can't believe you guys are trolling here. I just opened enhancement request/issue. Are you so bored from your life, that you prefer to write this nonsense here? If you can't help, don't "intrude"!