arendst/Tasmota #6290: Lohas E14 R50 LED Bulb

Geoff571, 2019-08-24:
### ISSUE DESCRIPTION - TROUBLESHOOTING

After seeing a few people looking at Lohas LED bulbs previously, I took a plunge and bought a two pack of the Lohas E14 R50 Smart Spots from [Amazon](https://www.amazon.co.uk/Reflector-Changing-Controlled-Devices-Required/dp/B07VHCQGH8/ref=sr_1_5?keywords=lohas+e14+r50&qid=1566668154&s=gateway&sr=8-5). These are RGBW bulbs which I was able to flash out of the box using the Tuya Convert OTA process.

Now that\i have Tasmota installed, I'm struggling to get the correct template settings for the configuration. I have tried configuring the module with the default Sonoff B1 settings, but this doesn't allow me to switch the light. The AiLight module works to a point, I can control the RGB LEDs however, the `white` command doesn't work. 

Can anyone help me understand what I need to be looking at in order to tweak the template and get the white LEDs working? 

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ X] Device used (i.e. Sonoff Basic) : Lohas E14 R50 Smart Bulb
- [x ] Tasmota binary firmware version number used : 6.6.0 pre-compiled_ / (pre-compiled or self-compiled ?)
- [X ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE: 18:57:23 RSL: Lohas-LED-2/stat/STATUS = {"Status":{"Module":27,"FriendlyName":["Lohas-LED-2"],"Topic":"Lohas-LED-2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```
**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2019-08-24:
Perhaps https://blakadder.github.io/templates/lohas-a19.html

In any case, this not a bug or a feature request. as such, please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-08-24 by ascillato2

Geoff571, 2019-08-24:
Not quite a full solution, but better than I had and something for me to work with and tweak. 

Apologies for posting incorrectly, I thought that if I'd tagged it here any answers would help those that follow in my footsteps 

Thanks for you help.

Geoff