arendst/Tasmota #6291: Is it possible to Assign Command Result to Variable?

windischb, 2019-08-25:
Hi!
Is it possible to assign the result from a Command to a variable?
I would like to toggle the FanSpeed of in iFan03 on double press.
If FanSpeed is 0 set FanSpeed to 2, if Fanspeed other than 0 set FanSpeed to 0.

Rule1
on button1#state=2 var1 FanSpeed endon
on var1#state=0 FanSpeed 2 endon
on var1#state!=0 FanSpeed 0 endon

ascillato, 2019-08-25:
Sorry, not supported.

What is the behaviour you want to achieve?

windischb, 2019-08-25:
As i wrote above, i'll try to toggle the FanSpeed of an iFan03.

If the Fan is on (FanSpeed != 0) turn of the Fan.
If the Fan is off (FanSpeed == 0) turn the Fan on at Speed 2.

But for that i have the check the current value of Fanspeed...

In Javascript i would wrote something like that:
```Javascript
var currentFanSpeed = Fanspeed;
if (currentFanSpeed == 0) {
    FanSpeed 2;
} else {
    FanSpeed 0;
}
```

windischb, 2019-08-25:
Maybe the only possible solution is to compile with Scripting Feature...?

laurentdong, 2019-09-10:
@windischb You can try to use MQTT subscribe a topic and catch the value with RULE.

closed 2019-10-09 by ascillato2

richcarnell, 2019-11-16:
@windischb 
Did you have any luck with this?