arendst/Tasmota #6292: WS2812 always full brightness.

RudyFiero, 2019-08-25:
#4755 # ISSUE DESCRIPTION - TROUBLESHOOTING
I have a strip of five WS2812 type devices connected to an IO pin. The light output is always on full when the Tasmota firmware starts. Using the GUI I can select a different setting but it has no effect. The output is still on full. 

The WS2812  (possibly SK6812 device)  is powered by Vcc, 3.4 volts. I have tried using GPIO15 and GPIO13.  I have tried three different LED strips. 

I have used Adafruit Neopixel test code on the same boards and it works as expected. 

I used a second board running a test program, using the Adafruit library, and or-ed the control signal to the "Neopixel" input along side the Tasmota control signal. Either board can be made to send to the one Neopixel led board. When I change the value using the Tasmota Gui I always get full brightness. When I have the parallel board sending to the Neopixels the correct colors/brightness is displayed.

I have looked at the timing with a Salea logic probe and the timing seems to be within spec. (but on the very edge) The one value I have concerns of is  T0H 0.35uS +/- 150nS. When using the Tasmota T0H is sometimes 0.50uS, just on the edge of the specification. Otherwise I have not found anything else that I thought might be an issue.
  

- [*] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [*] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [*] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [*] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)     No account
- [ *] Device used  : ESP-07S and ESP-12E
- [*] Tasmota binary firmware version number used : 6.6.0 / pre-compiled
- [*] Development IDE - Compiler / Upload tools used : Arduino web updater sketch to load bin file
- [*] Provide the output of command ``status 0`` :
```
21:23:03 CMD: Status 0
21:23:03 MQT: stat/tasmoto/STATUS = {"Status":{"Module":18,"FriendlyName":["Breadboard","Light","Pump","Led"],"Topic":"tasmoto","ButtonTopic":"0","Power":40,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:23:03 MQT: stat/tasmoto/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:57:28","StartupUTC":"2019-08-25T19:25:35","Sleep":50,"CfgHolder":4617,"BootCount":58,"SaveCount":481,"SaveAddress":"FB000"}}
21:23:03 MQT: stat/tasmoto/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:23:03 MQT: stat/tasmoto/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MTS61752","HeatPad"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000000000000000000000","00000000"]}}
21:23:03 MQT: stat/tasmoto/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":13,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
21:23:03 MQT: stat/tasmoto/STATUS5 = {"StatusNET":{"Hostname":"Tasmota","IPAddress":"192.168.100.109","Gateway":"192.168.100.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.254","Mac":"5C:CF:7F:CF:6C:D9","Webserver":2,"WifiConfig":4}}
21:23:03 MQT: stat/tasmoto/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.100.135","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CF6CD9","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
21:23:03 MQT: stat/tasmoto/STATUS7 = {"StatusTIM":{"UTC":"Sun Aug 25 20:23:03 2019","Local":"Sun Aug 25 21:23:03 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:56","Sunset":"19:47"}}
21:23:03 MQT: stat/tasmoto/STATUS10 = {"StatusSNS":{"Time":"2019-08-25T21:23:03","Switch1":"ON","DS18B20":{"Temperature":26.4},"TempUnit":"C"}}
21:23:03 MQT: stat/tasmoto/STATUS11 = {"StatusSTS":{"Time":"2019-08-25T21:23:03","Uptime":"0T00:57:28","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":970,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"ON","POWER5":"OFF","POWER6":"ON","Dimmer":8,"Color":"141414","HSBColor":"0,0,8","Channel":[8,8,8],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"XXXXXXXX","BSSId":"60:FE:20:E3:A1:4E","Channel":1,"RSSI":96,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
21:24:54 CMD: weblog 4
21:24:55 MQT: stat/tasmoto/RESULT = {"WebLog":4}
21:24:55 CFG: Saved to flash at FA, Count 482, Bytes 3584
21:25:03 WIF: Checking connection...
21:25:03 WIF: Connected
21:25:23 WIF: Checking connection...
21:25:23 WIF: Connected

```

RudyFiero, 2019-08-25:
I just looked at the data sheet for the SK6812 and T0H is specified as 0.3uS +/- 150uS. And this would be a Maximum of 450uS. With the Tasmota firmware I measured a maximum of 500uS. I suspect that this is the reason for the problem I am seeing. 

One thing I forgot o mention above is that using a CPU speed of 160MHz made no difference to the timings. 

The bit timings I measures were, (in microseconds)

0.50 / 0.8
0.50 / 0.8
0.46 / 0.83
0.46 / 0.83
0.46 / 0.79
0.50 / 0.79
0.87 / 0.42
0.50 / 0.79
0.50 / 0.79
0.50 / 0.79
0.46 / 0.83
0.46 / 0.83
0.46 / 0.79
0.50 / 0.79
0.87 / 0.42
0.50 / 0.79
0.50 / 0.79

The LEDs I bought were sold as WS2812s but based on the dimensions of the packaging they matched the SK6812s. I have read that the SK6812s are regularly used instead of the WS2812 but still sold as the WS2812s. Also that the SK6812s are a more reliable part. 

I design hardware for a small electronics manufacturing company. I considered using these in the last product I designed.

s-hadinger, 2019-08-26:
You can take a look at `xplg_ws2812.ino`. There seem to be some options when initializing NeoPixelBus-2.2.9 library.

```
#ifdef USE_WS2812_DMA
  typedef Neo800KbpsMethod selectedNeoSpeedType;
#else   // USE_WS2812_DMA
  typedef NeoEsp8266BitBang800KbpsMethod selectedNeoSpeedType;
#endif  // USE_WS2812_DMA
  NeoPixelBus<selectedNeoFeatureType, selectedNeoSpeedType> *strip = nullptr;
```

then used here:

```
void Ws2812Init(void)
{
#ifdef USE_WS2812_DMA
  strip = new NeoPixelBus<selectedNeoFeatureType, selectedNeoSpeedType>(WS2812_MAX_LEDS);  // For Esp8266, the Pin is omitted and it uses GPIO3 due to DMA hardware use.
#else  // USE_WS2812_DMA
  strip = new NeoPixelBus<selectedNeoFeatureType, selectedNeoSpeedType>(WS2812_MAX_LEDS, pin[GPIO_WS2812]);
#endif  // USE_WS2812_DMA
  strip->Begin();
  Ws2812Clear();
}
```

I don't have a WS2812 so I can't test anything.

Jason2866, 2019-08-27:
You can try DMA Modus (needs self compile). 
Some WS2812 does only work with TASMOTA in this mode.

Jason2866, 2019-08-27:
Thx for your investigations in this! 
Are you going to open a issue in NeoPixelBus github?

RudyFiero, 2019-08-29:
The NeoPixelBus library has been modified to correct this problem. It is in the main branch. I don't know when he will put this into a release version.

Jason2866, 2019-08-29:
@arendst  Can you please update the NeoPixelBus library? 
Timing problem Makuna/NeoPixelBus#285 is fixed with https://github.com/Makuna/NeoPixelBus/pull/286
Thx!

arendst, 2019-08-29:
I'll have a look fearing code size increase ....

arendst, 2019-08-29:
Ah! it saves 4 bytes! 

Library updated. Over to @RudyFiero for testing. If a new bitbang driver is needed (like ..SK6812) pls let me know and I'll add it using a new define.

I hate the DMA version as it's implementation interferes with most other Tasmota interrupt drivers. I had to use defines to disable features like POW energy monitoring and Arilux RF control.

Jason2866, 2019-08-29:
A define for SK6812 would be good. I remember some problem questions (asked in Discord) for SK6812 stripes. Maybe with this optimized settings the stripes does work too?! 
I can understand your regrets against the DMA version but for WS2812 stripes it is the optimal one...

arendst, 2019-08-29:
Done.

Makuna, 2019-08-29:
@arendst Note that there other hardware assisted methods for the esp8266.  See [the wiki entry here](https://github.com/Makuna/NeoPixelBus/wiki/ESP8266-NeoMethods).
Also note, that the method naming is moving away from 400/800Kbps to more chip specific names, again [the wiki](https://github.com/Makuna/NeoPixelBus/wiki/NeoPixelBus-object#neopixel-led-model-specific-methods) has  an entry for this.  While generally not required right now, at least with the APA106s it is.

ameennihad, 2019-12-20:
I'm using SK6812 RGBW strip (manufacturer say it's similar ws2812b), it used to work fine in 6.5 by only setting USE_WS2812_CTYPE to NEO_GRBW, now after upgrading to 7.1.2 I'm getting strange results, the first 40 LEDs are white, others are always off, WW channel is always off. I tried setting USE_WS2812_HARDWARE  to NEO_HW_SK6812 and NEO_HW_WS2812X but nothing changed.
Are there other setting to change?

Makuna, 2019-12-20:
is USE_WS2812_DMA defined?
p.s.  please don't append a new issues onto an existing one that is closed.  You should create a new one and reference this one (put a link to it in the issue body).