arendst/Tasmota #6293: Windows 10 Tasmota Device Manager - config crashed TDM

scargill, 2019-08-25:
I have what I think is up to date TDM - 0.1.20 - running on Windows 10 64 bit - the Raspberry Pi 3 running Buster and Mosquitto is handling my Tasmota'd Sonoff BASICs (6.6.0) is running fine, 24-7. I can use TDM to turn the devices on and off very reliably, but if I hit either the top CONFIGURE or individual item CONFIGURE options in TDM, ultimately TDM crashes within seconds.

Pete

meingraham, 2019-08-25:
@scargill Pete,

TDM is a "related" project to Tasmota. Support for TDM can be obtained from @jziolkowski through his GitHub project. A newer version is under development. It is available at https://github.com/jziolkowski/tdm/tree/development. However, it is a work in progress and the Configure capability may not yet be fully fleshed out.

As your post is not a Tasmota bug or feature request, direct your questions to the TDM GitHub and please close this issue.

Mike

ascillato2, 2019-08-25:
Closing this issue as it is not Tasmota Software issue. Please, address this on TDM repository. Thanks

closed 2019-08-25 by ascillato2

scargill, 2019-08-26:
Sorry - wrong forum, thanks for feedback.