arendst/Tasmota #6294: wrong reply from baudrate command

futesta, 2019-08-26:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
reply of the baudrate command is not as expected. wrong number is replyed but hardware setup is correct.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : Sonoff Basic
- [X] Tasmota binary firmware version number used : 00:00:00 Project sonoff Sonoff Version 6.6.0(release-sonoff)-2_3_0
- [X] Development IDE - Compiler / Upload tools used : ____ / ____
- [X] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
21:40:57 CMD: status 0
21:40:57 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:40:57 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":460800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:51:50","StartupUTC":"2019-08-26T19:49:07","Sleep":50,"CfgHolder":4617,"BootCount":5,"SaveCount":19,"SaveAddress":"F9000"}}
21:40:57 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:40:57 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXX","YYY"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00C08009","280500000100060000000000000000000000","00000000"]}}
21:40:57 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
21:40:57 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7575","IPAddress":"192.168.0.224","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":":::::","Webserver":2,"WifiConfig":4}}
21:40:57 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_939D97","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
21:40:57 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Aug 26 20:40:57 2019","Local":"Mon Aug 26 21:40:57 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:58","Sunset":"19:46"}}
21:40:57 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-08-26T21:40:57"}}
21:40:57 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-08-26T21:40:57","Uptime":"0T00:51:50","Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXX","BSSId":":::::","Channel":4,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [X] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
21:11:22 CMD: Baudrate 9600
21:11:22 RSL: stat/sonoff/RESULT = {"Baudrate":9600}
21:11:41 CMD: Baudrate 115200
21:11:41 RSL: stat/sonoff/RESULT = {"Baudrate":115200}
21:11:57 CMD: Baudrate 1000000
21:11:57 RSL: stat/sonoff/RESULT = {"Baudrate":78000}
21:26:18 CMD: baudrate 230400
21:26:18 RSL: stat/sonoff/RESULT = {"Baudrate":230400}
21:26:28 CMD: baudrate 460800
21:26:28 RSL: stat/sonoff/RESULT = {"Baudrate":153600}

21:46:33 CMD: weblog 4
21:46:33 RSL: stat/sonoff/RESULT = {"WebLog":4}
21:46:33 CFG: Saved to flash at F8, Count 20, Bytes 3584
21:46:39 WIF: Checking connection...
21:46:39 WIF: Connected
21:46:47 CMD: baudrate 460800
21:46:47 SRC: WebConsole from 192.168.0.29
21:46:47 RSL: Received Topic /baudrate, Data Size 6, Data 460800
21:46:47 RSL: Group 0, Index 1, Command BAUDRATE, Data 460800
21:46:47 RSL: stat/sonoff/RESULT = {"Baudrate":153600}
21:46:59 WIF: Checking connection...
21:46:59 WIF: Connected

```

### TO REPRODUCE
setting baudrate reply issue, see above...

### EXPECTED BEHAVIOUR
correct reply of the set baudrate command

### SCREENSHOTS
none

### ADDITIONAL CONTEXT
While posting the status 0 command reply, i have seen that here the correct value for the baudrate is reported. So only reply of the baudrate command is wring? perhaps integer calc overflow or something?!?

The measured baudrate with the scope is corrected, so only reply seems to be wrong...

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-08-27:
As designed. Tasmota choose the possible speed. 
The speeds you try to set are not supported with Tasmota

arendst, 2019-08-27:
Baudrate in Tasmota is designed to be a multiple of 1200 bps and stored within 8 bits. 
- So 1200 bps is stored as 1200/1200 = 1 (*1200 = 1200)
- 115200 is stored as 115200 / 1200 = 96 (*1200 = 115200)
- 460800 is stored as 460800 / 1200 = 384 but this is too large for 8 bits (= hexadecimal 0x180) and will be stored as 0x80 = 128 (*1200 = 153600)

So it's a feature, not a bug.

arendst, 2019-08-27:
The code looks like this:
```
void CmndBaudrate(void)
{
  if (XdrvMailbox.payload > 1200) {
    XdrvMailbox.payload /= 1200;  // Make it a valid baudrate
    baudrate = XdrvMailbox.payload * 1200;
    SetSerialBaudrate(baudrate);
  }
  ResponseCmndNumber(Settings.baudrate * 1200);
}

void SetSerialBaudrate(int baudrate)
{
  Settings.baudrate = baudrate / 1200;
  if (Serial.baudRate() != baudrate) {
    if (seriallog_level) {
      AddLog_P2(LOG_LEVEL_INFO, PSTR(D_LOG_APPLICATION D_SET_BAUDRATE_TO " %d"), baudrate);
    }
    delay(100);
    Serial.flush();
    Serial.begin(baudrate, serial_config);
    delay(10);
    Serial.println();
  }
}
```
I now see where it fails; the recalculated baudrate in this case will be the same as input as the payload register is 32 bits. It then sets the requested baudrate (what is what @futesta registers) and finally stored the "crippled" version in 8 bits.

On reboot it will use the crippled version.

I'll change the code to also set the "crippled" baudrate. This implies the max baudrate Tasmota supports will be 255 * 1200 = 306000

New code will be
```
void CmndBaudrate(void)
{
  if (XdrvMailbox.payload > 1200) {
    XdrvMailbox.payload /= 1200;  // Make it a valid baudrate
    baudrate = (XdrvMailbox.payload & 0xFF) * 1200;
    SetSerialBaudrate(baudrate);
  }
  ResponseCmndNumber(Settings.baudrate * 1200);
}
```

futesta, 2019-08-27:
Folks... thanks for the activity on the issue! :-)

Do I understand this in the right manner... the 1MBit would not have worked after a reboot, right? hmmm, ok at the moment I can not test that, but I will test it in the next few days...

Tasmota stores the value for the baudrate settings in parts of 1200 baud, is this understanding correct? So availbale baudrates are 1200, 2400, 3600, etc... and the max in 255*1200=306000, right?

Damn, I thought I could build a bridge for my application. Unfortunately, the baudrate is fixed to 1MBit and not easily changeable... :-(

A little testing:
`
20:18:25 CMD: baudrate 306000
20:18:25 RSL: stat/sonoff/RESULT = {"Baudrate":306000}
20:18:25 RSL: tele/sonoff/RESULT = {"SerialReceived":"0d0a"}
20:18:29 CMD: status 1
20:18:29 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":306000,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T01:43:01","StartupUTC":"2019-08-27T17:35:28","Sleep":50,"CfgHolder":4617,"BootCount":17,"SaveCount":102,"SaveAddress":"F6000"}}
`

after "RESTART 1":
`
20:21:26 CMD: status 1
20:21:26 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":306000,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:02:02","StartupUTC":"2019-08-27T19:19:24","Sleep":50,"CfgHolder":4617,"BootCount":18,"SaveCount":105,"SaveAddress":"FB000"}}
`

Okay, lets test a higher baudrate, like 460800, this should results in a wrong reply and readback like 460800 / 1200 = 384 / 384-256 = 128 / 128 * 1200 = 153600, right?

Testing:
`
20:22:44 CMD: baudrate 460800
20:22:45 RSL: stat/sonoff/RESULT = {"Baudrate":153600}
`

So far... but "STATUS 1" says:
`
20:22:54 CMD: status 1
20:22:54 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":460800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:30","StartupUTC":"2019-08-27T19:19:24","Sleep":50,"CfgHolder":4617,"BootCount":18,"SaveCount":106,"SaveAddress":"FA000"}}
`

hmmm? Lets try "RESTART 1":
`
00:00:42 CMD: STATUS 1
00:00:42 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":153600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:42","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":19,"SaveCount":109,"SaveAddress":"F7000"}}
`

There it is! But why only now? I have no scope tangible, otherwise I could measure...

greets,
Michael

futesta, 2019-08-28:
ok, i was confused because you wrote 30600 not 306000 in your commit. now i think there is missing at the end "0" ;-)

are there any plans that the baudrate will be higher adjustable later on?

futesta, 2019-08-28:
this would be great, thanks! will habe a closer look of the ESP and max baudrate with tasmota FW when the new hardware is arrived!

thinkings: perhabs a baudrate scaler command (so it is solved with AVR microcontrollers to double the baud rate, halve or quadruple, etc...)

greets
Michael

PS: will close the issue this evening, currently no direct access to github

arendst, 2019-08-28:
Your findings are consistent with mine. 

It is solved with the fix released yesterday.

arendst, 2019-08-28:
Currently no plan but I will consider the impact of moving it from 8 bit to 16 bit

arendst, 2019-08-28:
Just changed the code to support baudrates from 300 to 19660500 bps in 300 increments. That should satisfy your need.