arendst/Tasmota #6295: new meter driver

gemu2015, 2019-08-27:
i am preparing a special driver for pr. 
below is a description of what the driver does.
i need advice for a few decisions.

1. the name is outdated and misleading (xsns_95_sml.ino) proposal ?
2. when not using scripts this driver requires a default meter to show
what should be the default  (possibly a most common or useful device, p1-decoder ?)
3. this driver is different from all others because if defines everything in the descriptor: 
web interface, scaling of signals, doing math with signals  , mqtt messages, gpiopins (which requires "None" in the corresponding setup of Tasmota) 
this driver should possibly have a library of descriptors to choose from, where everyone can contribute.
where to put it? 
@meingraham  will help me to set up a cookbook for the descriptor language.

=====================================================
Driver for various meters , heating devices and reed like contacts

supports the following:
 
1. ASCI OBIS telegrams emitted from many german smart meters and also from dutch P1 meter interface
2. binary SML OBIS telegram emitted from many german smart meters
3. binary EBUS telegram emitted by many heaters and heat pumps  (e.g. Vaillant, Wolf)
4. binary RAW telegram decodes all kinds of binary data eg EMS heater bus
5. counter interface (uses Tasmota counter storage) for e.g. reed contacts either in polling or IRQ mode.

there may be more then one software serial driven meter. this is accomplished by a modified Tasmota Serial which doesn't wait during IRQ.
(the limitation of this mode is that the telegrams must occur in intervals < 40 seconds)
 

there are dozens of variants and use cases for this driver
this requires a means of defining these variants

this is done through so called meter descriptors

the recent method which is still available is defining a meter by #define pragmas in the compiler
this requires recompiling the driver after each modification.

the new preferred method is using the script editor to define the descriptors in an ">M" section
for this only one software version is required and a modification can be easily made on the fly. 

the driver falls back to the default #define definition if no >M section is found in scripter or scripter is not compiled

descriptor defintion

in script editor define a section called M with the number of meters wanted (up to 5)
e.g. >M 3

next is defining the type of interface for each meter

e.g. +1,3,o,0,9600,OBIS

+1 adds the meter number 1
next is the source GPIO pin of signal => here 3 => RX pin
next is the type of meter => o=obis ASCI type of coding, s=sml binary smart message coding, c=counter type, e=ebus binary coding, r=raw binary coding (any binary telegram).
next is flag => 0=counter without pullup, 1=with pullup 
next is parameter => on o,s,e,r type serial baud rate, on type c positive value means counter poll interval, negative value irq driven counters with this debounce time (milliseconds)
next is MQTT JSON prefix up to 7 chars,this prefix is also shown in WEBUI

the list of meter entries start with the meter number to which this decoder belongs. (up to 16 entries, #define MAX_VARS 16)

e.g. 1,1-0:1.8.0*255(@1,consumption,KWh,Total_in,4

1. meter number, comma separated
2. then until @ char => sequence to decode OBIS as ASCI, or SML, EBUS, RAW as HEX ASCI
OBIS: asci OBIS code until the "(" char which indicated the start of the meter value
SML: SML binary OBIS as hex until FF indicating start SML decoded value
EBUS,RAW: hex values of EBUS,RAW block to compare  xx means ignore value, ss = extract this signed byte, uu = extract this unsigned byte, uuuu = extract this unsigned word, ssss = extract this signed word.
decoding a bit as 0,1 is done after the @ sign with bx: x=0...7 extracting the corresponding bit from a byte. e.g. 1,xxxx5017xxuu@b0:1,Solarpumpe,,Solarpump,0 extracts bit zero

3. scaling factor (divisor) (may also be negative or smaller then zero e.g. 0.1 => result * 10)
  when decoding a string result (e.g. meter serial) put a # char here (only in one line per meter) after the # char you need a termination char at OBIS a ')'
4. name of signal in WEBUI (max 23 chars)
5. unit of signal (max 7 chars)
6. name of MQTT signal (max 23 chars)
7. number of decimals (decimal precision) (if you add 16 here the signal is transmitted immediately else on teleperiod)

\# char terminates the list

=====================
special cmds

with the '=' char at the beginning of a line you may do some special decoding
=m does math  (+,-,*,/) a # before a number designates a constant number
example
=m 3+4+5/#3  add result of decoder entry 3,4,5 and divides by 3 (average)

=d difference calculated time difference within a decoder entry
example
=d 3 10  calculate difference of decoder entry 3 after 10 seconds each.

=h html text (up to 30 chars)
inserts a html line between entries (these lines do not count as decoder entry)

some examples

SML descriptor for Hager EHZ363
=====================
\>D

\>B
=>sensor95 r

\>M 1
+1,3,s,0,9600,SML

1,77070100010800ff@1000,Verbrauch,KWh,Total_in,4
1,77070100020800ff@1000,Einspeisung,KWh,Total_out,4
1,77070100100700ff@1,Aktueller Verbrauch,W,Power_curr,0
1,77070100000009ff@#,Zähler Nr,,Meter_number,0
\#

OBIS descriptor for Hager EHZ161
=====================
\>D

\>B
=>sensor95 r

\>M 1
+1,3,o,0,9600,OBIS

1,1-0:1.8.1*255(@1,Verbrauch,KWh,Total_in,4
1,1-0:2.8.1*255(@1,Einspeisung,KWh,Total_out,4
1,=d 2 10 @1,Aktuelle Einspeisung,W,Power_curr,0
1,1-0:0.0.0*255(@#),Zähler Nr,, Meter_number,0
\#

COMBO descriptor, Water,Gas,SML 
===========================
\>D

\>B
=>sensor95 r

\>M 3
+1,1,c,0,10,H20
+2,4,c,0,50,GAS
+3,3,s,0,9600,SML

1,1-0:1.8.0*255(@10000,Zählerstand,cbm,Count,4
2,=h==================
2,1-0:1.8.0*255(@100,Zählerstand,cbm,Count,3
3,77070100010800ff@1000,Verbrauch,KWh,Total_in,3
3,=h==================
3,77070100100700ff@1,Aktueller Verbrauch,W,Power_curr,2
3,=h -------------------------------
3,=m 10+11+12 @100,Ströme L1+L2+L3,A,Curr_summ,2
3,=m 13+14+15/#3 @100,Spannung L1+L2+L3/3,V,Volt_avg,2
3,=h==================
3,77070100240700ff@1,Verbrauch P1,W,Power_p1,2
3,77070100380700ff@1,Verbrauch P2,W,Power_p2,2
3,770701004c0700ff@1,Verbrauch P3,W,Power_p3,2
3,=h -------------------------------
3,770701001f0700ff@100,Strom L1,A,Curr_p1,2
3,77070100330700ff@100,Strom L2,A,Curr_p2,2
3,77070100470700ff@100,Strom L3,A,Curr_p3,2
3,=h -------------------------------
3,77070100200700ff@100,Spannung L1,V,Volt_p1,2
3,77070100340700ff@100,Spannung L2,V,Volt_p2,2
3,77070100480700ff@100,Spannung L3,V,Volt_p3,2
3,=h==================
3,77070100000009ff@#,Service ID,,Meter_id,0
3,=h--------------------------------
\#

WOLF CSZ 11/300 heater
================
\>D

\>B
=>sensor95 r

\>M 1
+1,3,e,0,2400,EBUS

1,xxxx0503xxxxxxxxxxxxxxxxss@1,Außentemperatur,C,Outsidetemp,0
1,xxxx5014xxxxxxxxxxuu@1,Raumtemperatur,C,Roomtemp,0
1,xxxx0503xxxxxxxxxxxxxxuu@1,Warmwasser,C,Warmwater,0
1,xxxx0503xxxxxxxxxxuu@1,Heizkessel,C,Boiler,0
1,03fe0503xxxxxxxxxxxxuu@1,Rücklauf,C,Returns,0
1,03fe0503xxxxuu@1,Status,,Status,0
1,03fe0503xxxxxxuu@b3:1,Brenner,,Burner,0
1,xxxx5017xxxxxxuuuu@16,Solarkollektor,C,Collector,1
1,xxxx5017xxxxxxxxxxuuuu@16,Solarspeicher,C,Solarstorage,1
1,xxxx5017xxuu@b0:1,Solarpumpe,,Solarpump,0
\#

arendst, 2019-08-27:
Gerhard,

as it needs scripter and updated TasmotaSerial I suggest to make a new firmware image selection, like ``sonoff-meter`` with scripter enabled and your meter driver enabled too.

To cater for your Serial code changes I suggest to add a new serial library with your changes to make sure backward compatibility is still preserved with other drivers using TasmotaSerial. Once settled I can try to integrate both Serial drivers.

So continue integration but do not expect scripter to be enabled in all images as it uses too much code space while it is also still expanding.

gemu2015, 2019-08-27:
@arendst 
hi Theo
the serial driver is completely backward compatible , there is an extra parameter enabling the new mode. (no wait mode)

TasmotaSerial::TasmotaSerial(int receive_pin, int transmit_pin, int hardware_fallback, int nwmode)

it uses bit change IRQs in stead of waiting. this has a drawback.
it  implies a problem for the last bit of course, if this is high  the byte is complete not until the next start bit.  this is no problem however if we receive continuous data streams.

using this driver i was able to receive 3 serial streams over software serial simultaneously.

ok ill make a pr for the meter driver with the updated software serial.

you may then decide how to call the driver ??

the default descriptor configuration would be subject to change.

it should be a device that a majority of user may need.

closed 2019-08-27 by gemu2015

meingraham, 2019-09-09:
@gemu2015 

Does `USE_SML_M` (commented out) need to be added to `my_user_config.h` as a placeholder?

gemu2015, 2019-09-10:
@meingraham 
we may do that with next pull request. i updated the driver for multiple MODBUS and PZEM devices. After some more testing ill do the next pr.