arendst/Tasmota #6302: Current, Voltage, frequency lost, power factor, etc lost in sonoff POW R2 (CSE7766)

pablozg, 2019-08-27:
### BUG DESCRIPTION

With version 6.6.0.8 (previously used 6.5.0.16) the voltage, current, power factor, frequency, etc. in sonoff pow R2 is not showed anymore after upgrade to last version.

Only Power, Energy Today, Energy Yesterday and Energy Total is showed.

### REQUESTED INFORMATION
- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : _____
- [X] Tasmota binary firmware version number used : 6.6.0.8 / self-compiled
- [X] Development IDE - Compiler / Upload tools used : Visual Studio Code / platformio
- [X] Provide the output of command ``status 0`` :
```
21:32:57 CMD: status 0
21:32:57 MQT: Casa/PB/monitorCasa/stat/STATUS = {"Status":{"Module":43,"FriendlyName":["monitorCasa"],"Topic":"Casa/PB/monitorCasa","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:32:57 MQT: Casa/PB/monitorCasa/stat/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:44","StartupUTC":"2019-08-27T19:31:13","Sleep":50,"CfgHolder":4617,"BootCount":57,"SaveCount":346,"SaveAddress":"3F9000"}}
21:32:57 MQT: Casa/PB/monitorCasa/stat/STATUS2 = {"StatusFWR":{"Version":"6.6.0.8(sonoff)","BuildDateTime":"2019-08-27T21:24:26","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(c0eb301)"}}
21:32:57 MQT: Casa/PB/monitorCasa/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Isabella_T","Isabella_H"],"TelePeriod":20,"Resolution":"558180C0","SetOption":["002A8009","28B400000100068000005AFF000000170D00","00000200"]}}
21:32:57 MQT: Casa/PB/monitorCasa/stat/STATUS4 = {"StatusMEM":{"ProgramSize":553,"Free":2516,"Heap":25,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A1","22B617CD","01001BC0","00000001"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
21:32:57 MQT: Casa/PB/monitorCasa/stat/STATUS5 = {"StatusNET":{"Hostname":"Casa/PB/monitorCasa-6274","IPAddress":"10.10.0.101","Gateway":"10.10.0.254","Subnetmask":"255.255.255.0","DNSServer":"10.10.0.254","Mac":"80:7D:3A:32:B8:82","Webserver":2,"WifiConfig":2}}
21:32:57 MQT: Casa/PB/monitorCasa/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.10.0.110","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_32B882","MqttUser":"htpc","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
21:32:57 MQT: Casa/PB/monitorCasa/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Aug 27 19:32:57 2019","Local":"Tue Aug 27 21:32:57 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"06:59","Sunset":"20:44"}}
21:32:57 MQT: Casa/PB/monitorCasa/stat/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
21:32:57 MQT: Casa/PB/monitorCasa/stat/STATUS10 = {"StatusSNS":{"Time":"2019-08-27T21:32:57","Epoch":1566934377,"ENERGY":{"TotalStartTime":"2019-05-25T00:00:00","Total":20.583,"Yesterday":24.744,"Today":20.583,"Power":1268}}}
21:32:57 MQT: Casa/PB/monitorCasa/stat/STATUS11 = {"StatusSTS":{"Time":"2019-08-27T21:32:57","Epoch":1566934377,"Uptime":"0T00:01:44","UptimeSec":104,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Isabella_T","BSSId":"D4:6E:0E:D0:43:EC","Channel":6,"RSSI":38,"LinkCount":1,"Downtime":"0T00:00:04"}}}
21:33:07 MQT: Casa/PB/monitorCasa/tele/STATE = {"Time":"2019-08-27T21:33:07","Epoch":1566934387,"Uptime":"0T00:01:54","UptimeSec":114,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Isabella_T","BSSId":"D4:6E:0E:D0:43:EC","Channel":6,"RSSI":42,"LinkCount":1,"Downtime":"0T00:00:04"}}

```
- [X] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
21:50:39 DMP: 55 5A 02 EC 70 00 03 5D 00 40 B5 00 6C 5C 52 0F D0 00 A4 3D F2 6E F6 E3
21:50:39 DMP: 55 5A 02 EC 70 00 03 5D 00 40 B5 00 6C 5C 52 0F D0 00 A4 29 F2 6E F8 D1
21:50:39 DMP: 55 5A 02 EC 70 00 03 5D 00 40 B5 00 6C 5C 52 0F D0 00 A4 29 F2 6E FB D4
21:50:39 DMP: 55 5A 02 EC 70 00 03 5D 00 40 B5 00 6C 5C 52 0F D0 00 A4 29 F2 6E FD D6
21:50:39 DMP: 55 5A 02 EC 70 00 03 5D 00 40 B5 00 6C 5C 52 0F D0 00 A4 29 F2 6F 00 DA
21:50:39 DMP: 55 5A 02 EC 70 00 03 5D 00 40 B5 00 6C 5C 52 0F D0 00 A4 29 F2 6F 02 DC
21:50:39 DMP: 55 5A 02 EC 70 00 03 5D 00 40 B5 00 6C 5C 52 0F D0 00 A4 29 F2 6F 05 DF
21:50:39 DMP: 55 5A 02 EC 70 00 03 5D 00 40 B5 00 6C 5C 52 0F D0 00 A4 29 F2 6F 07 E1
21:50:39 DMP: 55 5A 02 EC 70 00 03 5D 00 40 B5 00 6C 5C 52 0F D0 00 A4 29 F2 6F 0A E4
21:50:39 DMP: 55 5A 02 EC 70 00 03 5D 00 40 B5 00 6C 4E 52 0F D0 00 A4 21 F2 6F 0C D0
21:50:40 DMP: 55 5A 02 EC 70 00 03 5D 00 40 B5 00 6C 4E 52 0F D0 00 A4 21 F2 6F 0F D3
21:50:40 DMP: 55 5A 02 EC 70 00 03 5D 00 40 B5 00 6C 4E 52 0F D0 00 A4 21 F2 6F 11 D5
21:50:40 DMP: 55 5A 02 EC 70 00 03 5D 00 40 B5 00 6C 4E 52 0F D0 00 A4 21 F2 6F 14 D8
21:50:40 DMP: 55 5A 02 EC 70 00 03 5D 00 40 B5 00 6C 4E 52 0F D0 00 A4 21 F2 6F 16 DA
21:50:40 DMP: 55 5A 02 EC 70 00 03 5D 00 40 B5 00 6C 4E 52 0F D0 00 A4 21 F2 6F 19 DD
21:50:40 DMP: 55 5A 02 EC 70 00 03 5D 00 40 B5 00 6C 4E 52 0F D0 00 A4 21 F2 6F 1B DF
21:50:40 DMP: 55 5A 02 EC 70 00 03 5D 00 40 B5 00 6C 4E 52 0F D0 00 A4 21 F2 6F 1E E2
21:50:40 DMP: 55 5A 02 EC 70 00 03 5E 00 40 B5 00 6C 4E 52 0F D0 00 A4 21 F2 6F 20 E5
21:50:40 DMP: 55 5A 02 EC 70 00 03 5E 00 40 B5 00 6C 4E 52 0F D0 00 A3 C7 F2 6F 23 8D
21:50:41 DMP: 55 5A 02 EC 70 00 03 5E 00 40 B5 00 6C 4E 52 0F D0 00 A3 C7 F2 6F 25 8F
21:50:41 DMP: 55 5A 02 EC 70 00 03 5E 00 40 B5 00 6C 4E 52 0F D0 00 A3 C7 F2 6F 28 92
21:50:41 DMP: 55 5A 02 EC 70 00 03 5E 00 40 B5 00 6C 4E 52 0F D0 00 A3 C7 F2 6F 2A 94
21:50:41 DMP: 55 5A 02 EC 70 00 03 5E 00 40 B5 00 6C FE 53 D1 FE 55 5A 02 EC 70 00 03
21:50:41 CSE: Checksum failure
21:50:41 DMP: 55 5A 02 EC 70 00 03 5E 00 40 B5 00 6C 4E 52 0F D0 00 A3 C7 F2 6F 2F 99
21:50:41 WIF: Checking connection...
21:50:41 WIF: Connected
21:50:41 DMP: 55 5A 02 EC 70 00 03 5E 00 40 B5 00 6C 4E 52 0F D0 00 A3 C7 F2 6F 32 9C
21:50:41 DMP: 55 5A 02 EC 70 00 03 5E 00 40 B5 00 6C 4E 52 0F D0 00 A3 C7 F2 6F 34 9E
21:50:41 DMP: 55 5A 02 EC 70 00 03 5E 00 40 B5 00 6C 4E 52 0F D0 00 A4 1D F2 6F 37 F8
21:50:41 DMP: 55 5A 02 EC 70 00 03 5E 00 40 B5 00 6C 4E 52 0F D0 00 A4 1D F2 6F 39 FA
21:50:41 DMP: 55 5A 02 EC 70 00 03 5E 00 40 B5 00 6C 4E 52 0F D0 00 A4 1D F2 6F 3C FD
21:50:42 DMP: 55 5A 02 EC 70 00 03 5E 00 40 B5 00 6C 4E 52 0F D0 00 A4 1D F2 6F 3E FF
21:50:42 DMP: 55 5A 02 EC 70 00 03 5E 00 40 B5 00 6C 8A 52 0F D0 00 A4 1D F2 6F 41 3E
21:50:42 DMP: 55 5A 02 EC 70 00 03 5E 00 40 B5 00 6C 8A 52 0F D0 00 A4 1D F2 6F 43 40
[... 77 lines cut ...]
21:50:52 DMP: 55 5A 02 EC 70 00 03 5B 00 40 B5 00 6B A8 52 0F D0 00 A2 8F F2 70 43 CB
21:50:53 DMP: 55 5A 02 EC 70 00 03 5B 00 40 B5 00 6B A8 52 0F D0 00 A2 8F F2 70 46 CE
21:50:53 DMP: 55 5A 02 EC 70 00 03 5B 00 40 B5 00 6B E0 52 0F D0 00 A2 8F F2 70 48 08
21:50:53 DMP: 55 5A 02 EC 70 00 03 5B 00 40 B5 00 6B E0 52 0F D0 00 A3 14 F2 70 4B 91
21:50:53 DMP: 55 5A 02 EC 70 00 03 5B 00 40 B5 00 6B E0 52 0F D0 00 A3 14 F2 70 4D 93
21:50:53 DMP: 55 5A 02 EC 70 00 03 5B 00 40 B5 00 6B E0 52 0F D0 00 A3 14 F2 70 50 96
21:50:53 DMP: 55 5A 02 EC 70 00 03 5B 00 40 B5 00 6B E0 52 0F D0 00 A3 14 F2 70 52 98
21:50:53 DMP: 55 5A 02 EC 70 00 03 5B 00 40 B5 00 6B E0 52 0F D0 00 A3 14 F2 70 55 9B
21:50:53 DMP: 55 5A 02 EC 70 00 03 5B 00 40 B5 00 6B E0 52 0F D0 00 A3 14 F2 70 57 9D
21:50:53 DMP: 55 5A 02 EC 70 00 03 5B 00 40 B5 00 6B E0 52 0F D0 00 A3 14 F2 70 5A A0
21:50:54 DMP: 55 5A 02 EC 70 00 03 5E 00 40 B5 00 6B E0 52 0F D0 00 A3 14 F2 70 5C A5
[... cut ...]
21:50:54 DMP: 55 5A 02 EC 70 00 03 5E 00 40 B5 00 6B E0 52 0F D0 00 A3 61 F2 70 6B 01
21:50:54 DMP: 55 5A 02 EC 70 00 03 5E 00 40 B5 00 6B E0 52 0F D0 00 A3 61 F2 70 6E 04
21:50:54 DMP: 55 5A 02 EC 70 00 03 5E 00 40 B5 00 6B E0 52 0F D0 00 A3 61 F2 70 70 06
21:50:54 DMP: 55 5A 02 EC 70 00 03 5E 00 40 B5 00 6B E0 52 0F D0 00 A3 61 F2 70 73 09
21:50:55 DMP: 55 5A 02 EC 70 00 03 5E 00 40 B5 00 6B E0 52 0F D0 00 A3 28 F2 70 75 D2
21:50:55 DMP: 55 5A 02 EC 70 00 03 5E 00 40 B5 00 6B E0 52 0F D0 00 A3 28 F2 70 78 D5
```
### TO REPRODUCE
Upgrade to last version

arendst, 2019-08-27:
As a workaround try SetOption44 1 and SetOption45 1

pablozg, 2019-08-28:
Fixed, thanks

closed 2019-08-28 by pablozg