arendst/Tasmota #6306: Troubleshooting kinetic tuya device

armenchester, 2019-08-28:
I have a tuya device(not dimmer) with remote kinetic(without battery) switch and flash it with Tasmota, here is a device link https://www.aliexpress.com/item/32953812650.html?spm=a2g0s.9042311.0.0.694d4c4daVRBBq. 
I followed this two links without success https://blakadder.github.io/templates/ERC309.html, 
https://github.com/arendst/Sonoff-Tasmota/issues/6056 . 
I guess my and Zemismart device in the wiki can have the same functionality. I set configuration as tuya dimmer. The kinetic switch pairs but it don't trigger the relay, when I press on the remote switch the blue led blinks on the device and that's all.

What am I doing wrong? How can I make it work? Thanks

Jason2866, 2019-08-28:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

armenchester, 2019-08-29:
I have already addressed this issue there but no one answered.