arendst/Tasmota #6307: Step by step relay with double commutation

durante-ap, 2019-08-28:
Hi guys!
Before start, let me to thank all community and developer abot Tasmota. It's a very great firmware!

This days, i'm trying to create with a `SCRIPT` code, a step by step relay with double commutation, is some words i've two lamp, and i would to power their with following sequence:

LAMP1 ON, LAMP2 ON
LAMP1 ON, LAMP2 OFF
LAMP1 OFF, LAMP2 ON
LAMP1 OFF, LAMP2 OFF

I realized the following script, to do that:

>D
sw=0
prevButt=0
>B
spinm(5,0)

>F
sw=pin[5]
if sw==0
and prevButt!=0 {
prevButt=1
if pwr[1]==0 
and pwr[2] == 0 {
=>power1 1
=>power2 1
}else {
if pwr[1]==1 
and pwr[2]==1 {
=>power1 1
=>power2 0
}else{
if pwr[1]==1 
and pwr[2]==0{
=>power1 0
=>power2 1
}else{
=>power1 0
=>power2 0
}
}
}
}else{
if sw > 0 {
prevButt=0
}
}

I flash my NodeMCU board with a latest sonoff firmware (version 6.6.0) and core 2.4.2 with `USE_SCRIPT` flag enabled.

Moreover, i configure the board as following:
[image]

And I connect a button (on the GPIO 5) following the schema that i found in this thread: 
[https://github.com/arendst/Sonoff-Tasmota/issues/508](url)

My problems with this configuration and this code are the following:

1. If i try to pring the status of pin[5] with command `=>pring %pin[5]%` it will return always `1.00` value. Is it possible to configure sonoff to read the value as 0? I tried to connect the button the GND instead of VCC, but when i press the button, the status remain to 1 and after 40sec the board is reset
2. Every kind of code that i put into `>F` section is never executed also for example the simple code:
>D
sw=0
>B
=>print BOOT executed
spinm(5,0)

>F
sw=pin[5]
if sw>0{
;=>print pin status %int(sw)%
}

Can someone help me in this "simple" project?
I used the `SCRIPT` because for me is more intuitive and simple, but if someone know how to resolve this issue with Rules it can be fine!

Thanks a lot for your help!

gemu2015, 2019-08-28:
you need a pullup, either in hardware or per software
the script below works without problems

>D
sw=0

>B
; input with pullup
spinm(5,2)

>F
if pin[5]>0 {
print 1
} else {
 print 0
}

; this works also
print %pin[5]%

Jason2866, 2019-08-28:
Dont define GPIO 5 as a button. Define it as a switch.
See https://github.com/arendst/Sonoff-Tasmota/wiki/Buttons-and-Switches

durante-ap, 2019-08-28:
> you need a pullup, either in hardware or per software
> the script below works without problems
> 
> >D
> sw=0
> 
> >B
> ; input with pullup
> spinm(5,2)
> 
> >F
> if pin[5]>0 {
> print 1
> } else {
> print 0
> }
> 
> ; this works also
> print %pin[5]%

Hi gemu2015. I tried your code but it doesn't work in >F section. It works only in >S section.
Also, i follow also the suggest 

> Dont define GPIO 5 as a button. Define it as a switch.

by Jason2866, but it still doesn't work. 

I paste here the new configuration:
[image]

And the code that i tried
>D
sw=0

>B
=>print input with pullup
spinm(5,2)

>F
=>print %pin[5]%

It's seems to work only if I replace `>F` with `>S` section, and the Serial log show me the following line:
00:05:34  input with pullup
00:05:35  1.00
00:05:36  1.00
00:05:37  1.00
As you can see, the log is every ONE second

gemu2015, 2019-08-28:
Do you use an outdated Version?
F was introduced a few weeks ago

durante-ap, 2019-08-28:
> Do you use an outdated Version?
> F was introduced a few weeks ago

I use Tasmota V.6.6.0 with core version 2.4.2

gemu2015, 2019-08-28:
Ok, outdated
Use last dev Version

ascillato2, 2019-09-01:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-09-01 by ascillato2

nicolo-ribaudo, 2020-01-30:
If anyone else finds this issue with Google and needs something to copy-paste, I managed to make it work using this script:
```
>D
p:sw=0

>B
spinm(9 0)

>F

if (pin[9]!=sw) {
  print changed %sw% %pin[9]%
  if (pwr[1]==0) {
    if (pwr[2]==0) {
      => power1 1
      => power2 1
    } else {
      => power1 0
     => power2 0
    }
  } else {
    if (pwr[2]==0) {
      => power1 0
      => power2 1
    } else {
      => power1 1
      => power2 0
    }
  }
}

sw=pin[9]
```

Pin 9 is button 2 of the Sonoff Dual R2, and it's connected to a physical switch.