arendst/Tasmota #6310: Custom start bit and delay for RF bridge

corteXnzt, 2019-08-28:
I had this RF remote for a fiber optic display for ceiling/wall lights and can be learned by RF bridge. But when I send the learned signal it is not working. The remote is an RF controller with no label on chip. So I snoop the signal via SDR and Logic analyzer. 

The receiver is a SYN470R RF module.
The receiver has lots of noises when I tapped the logic analyzer when no signal is present. So to my understanding upon inspection as you can see on this screenshots it has like a start bit which indicates in the receiver that there is a command. So I copied the start bit and the delay and use the rc-switch library for arduino in sending the 24bit command after. And it works well after.  So I can't integrate it on my sonoff RF bridge.

If others had encounter this or might be useful to other users it might be good to have this feature to add a start bit and delay options or any possible solutions?

Thanks!

[image]
[image]

Jason2866, 2019-08-28:
Do you use the alternative RF Firmware "Portisch" for the RF bridge?
If not flash it, and try if it solves your issue.

Jason2866, 2019-08-28:
There are different versions of Portisch firmware. Which one do you have tried?

Jason2866, 2019-08-28:
Latest can always found at [Portisch](https://github.com/Portisch/RF-Bridge-EFM8BB1)

corteXnzt, 2019-08-28:
Thank you for the fast reply. I had flashed the Portisch version but after that it is not reading the remote afterwards, so I returned back to the stock firmware of the RF.

corteXnzt, 2019-08-28:
> There are different versions of Portisch firmware. Which one do you have tried?
> Stock firmware is very limited...
> Try this one. It works for me with a SYN470
> [RF-Bridge-EFM8BB1.zip](https://github.com/arendst/Sonoff-Tasmota/files/3551462/RF-Bridge-EFM8BB1.zip)

Oh I see, sorry I taught is just one version. I had tried the one that is included on the latest firmware release that I downloaded and assumed it was the latest?

corteXnzt, 2019-08-28:
> Latest can always found at [Portisch](https://github.com/Portisch/RF-Bridge-EFM8BB1)

Ok noted with that. I will test this tomorrow and give you a feedback. Thanks again!