arendst/Tasmota #6312: HX711 loses zero point after a reset

sheppy99, 2019-08-28:
### BUG DESCRIPTION
After a reboot HX711 when connected to a load cell resets the zero point to whatever it is currently measuring.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used: SOnOff Basic, HX711, AliExpress Load Cell
- [x] Tasmota binary firmware version number used : V6.6.0 / 2.3.0 Core pre-compiled
- [x] Development IDE - Compiler / Upload tools used : none
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

`23:01:39` MQT: stat/Oven/STATUS = {"Status":{"Module":1,"FriendlyName":["Oven"],"Topic":"Oven","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:01:39 MQT: stat/Oven/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:27:44","StartupUTC":"2019-08-28T21:33:55","Sleep":50,"CfgHolder":4617,"BootCount":29,"SaveCount":52,"SaveAddress":"F9000"}}
23:01:39 MQT: stat/Oven/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
23:01:39 MQT: stat/Oven/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SR2",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008008","280505000100060000000000000000000000","000000C0"]}}
23:01:39 MQT: stat/Oven/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
23:01:39 MQT: stat/Oven/STATUS5 = {"StatusNET":{"Hostname":"Oven-5883","IPAddress":"192.168.1.87","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"68:C6:3A:CD:B6:FB","Webserver":2,"WifiConfig":4}}
23:01:39 MQT: stat/Oven/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.69","MqttPort":1886,"MqttClientMask":"Oven","MqttClient":"Oven","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
23:01:39 MQT: stat/Oven/STATUS7 = {"StatusTIM":{"UTC":"Wed Aug 28 22:01:39 2019","Local":"Wed Aug 28 23:01:39 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:01","Sunset":"19:42"}}
23:01:39 MQT: stat/Oven/STATUS10 = {"StatusSNS":{"Time":"2019-08-28T23:01:39","HX711":{"Weight":0}}}
23:01:39 MQT: stat/Oven/STATUS11 = {"StatusSTS":{"Time":"2019-08-28T23:01:39","Uptime":"0T00:27:44","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SR2","BSSId":"38:10:D5:A7:DE:CF","Channel":2,"RSSI":20,"LinkCount":1,"Downtime":"0T00:00:09"}}}
23:01:45 MQT: tele/Oven/STATE = {"Time":"2019-08-28T23:01:45","Uptime":"0T00:27:50","Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SR2","BSSId":"38:10:D5:A7:DE:CF","Channel":2,"RSSI":22,"LinkCount":1,"Downtime":"0T00:00:09"}}
23:01:45 MQT: tele/Oven/SENSOR = `{"Time":"2019-08-28T23:01:45","HX711":{"Weight":0}}

'''
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

`23:03:06` WIF: Connected
23:03:15 MQT: tele/Oven/STATE = {"Time":"2019-08-28T23:03:15","Uptime":"0T00:29:20","Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SR2","BSSId":"38:10:D5:A7:DE:CF","Channel":2,"RSSI":22,"LinkCount":1,"Downtime":"0T00:00:09"}}
23:03:15 MQT: tele/Oven/SENSOR = `{"Time":"2019-08-28T23:03:15","HX711":{"Weight":0}}`

'
### TO REPRODUCE
0) Connect Load Cell to SOnOff taking care to ground the 0V rail for safety
1) Apply weight to load cell
2) Read correct weight in console / over MQTT
3) Reboot
4) Discover Weight shows zero, removing weight also shows zero
5) Discover adding extra weight to original weight only reads extra weight and NOT total weight

### EXPECTED BEHAVIOUR
After reboot it shows the same weight as before reboot

### ADDITIONAL CONTEXT
I'm trying to continuously weigh a gas bottle so I can see when its almost used up. As I reboot my devices every month to prevent them locking up and the gas bottle lasts for many months I need a way of preserving the calibration between reboots

ascillato2, 2019-08-28:
Closing this issue as it is **duplicated**. Sorry.

You have to update to latest Tasmota version 6.6.0.9

http://thehackbox.org/tasmota/020300

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-08-28 by ascillato2

sheppy99, 2019-08-29:
Thanks very much for the prompt response, updating has solved the problem. Can you point me to where I can find the latest fix information so I can check in future.

ascillato, 2019-08-29:
Yes, of course. 

All changes are in the changelog file: https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/_changelog.ino

lalo-uy, 2019-09-07:
Same problem with 6.6.0.11

tele/balanza/LWT Online
tele/balanza/INFO1 {"Module":"Generic","Version":"6.6.0.11(2e9f06f-sensors)","FallbackTopic":"cmnd/DVES_6E979A_fb/","GroupTopic":"sonoffs"}
tele/balanza/SENSOR {"Time":"2019-09-07T20:22:00","HX711":{"Weight":0.000}}
tele/balanza/SENSOR {"Time":"2019-09-07T20:22:10","HX711":{"Weight":0.000}}
weight added
tele/balanza/SENSOR {"Time":"2019-09-07T20:22:20","HX711":{"Weight":1.003}}
tele/balanza/SENSOR {"Time":"2019-09-07T20:22:30","HX711":{"Weight":1.217}}
tele/balanza/SENSOR {"Time":"2019-09-07T20:22:40","HX711":{"Weight":1.217}}
tele/balanza/LWT Offline

reboot without changing weight

tele/balanza/LWT Online
tele/balanza/INFO1 {"Module":"Generic","Version":"6.6.0.11(2e9f06f-sensors)","FallbackTopic":"cmnd/DVES_6E979A_fb/","GroupTopic":"sonoffs"}
tele/balanza/SENSOR {"Time":"2019-09-07T20:23:27","HX711":{"Weight":0.000}}
tele/balanza/SENSOR {"Time":"2019-09-07T20:23:37","HX711":{"Weight":0.002}}