arendst/Tasmota #6313: Interlock delay

vzorglub, 2019-08-28:
Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

**Have you looked for this feature in other issues and in the wiki?** Yes

**Is your feature request related to a problem? Please describe.**
I need a clear and clean interlock to avoid relay burn and possible damage to the motor device. Inductive load.

**Describe the solution you'd like**
A new command called Interlockdelay n. n in 10ms delay

**Describe alternatives you've considered**
A rule with a delay but not sure how to achieve that

meingraham, 2019-08-28:
@vzorglub 

Please provide pseudo logic for what you need. This will help to understand the feature request better or if there is an alternate means of accomplishing what you require (Rules, Backlog, Delay, Interlock, etc.).

Mike

vzorglub, 2019-08-28:
On interlock mode only one relay is ON at a time.
If I send the command Relay2 ON, Relay1 will turn OFF immediately.
I want to define a delay between the two events.

Relay2 ON command sent.

* Relay1 turns OFF
* 10ms (or multiple as define by Interlockdelay) passes
* Relay2 turns ON

Does that help?

meingraham, 2019-08-28:
@vzorglub 

To clarify, the `Interlock` command allows more than one relay on at a time.

For example `Interlock 1,2 3,4`
1&2 are interlocked (only one on at a time) and 3&4 are interlocked. This means that 1&3, 1&4, 2&3, or 2&4 could be on at the same time.

@arendst may already have a delay built in between turning off interlocked relays and turning their "partners" on.

As an alternative...
Let's say it is Switch2 that triggers the request for Relay2 to come on.

`Rule1 ON Switch2#State=1 DO Backlog Power1 0; Delay 2; Power2 1 ENDON`
`Rule1 1`
The only thing is that the smallest the delay can be is 0.2s. You can use the Scripting language could reduce that time.

Mike

meingraham, 2019-08-29:
@vzorglub - in Tasmota, the relay GPIO are controlled via an internal bitmask named `power`. The logic runs through and sets the bits in the `power` variable according to the requested action and the `Interlock` settings. Once the new state of the relay `power` bits has been set, the GPIO are set from that variable.

@arendst - does the logic run through the `power` bits to set the GPIO bit-by-bit or are all the GPIO updated in a single "output"? If sequentially, is there a possibility that two interlocked relays are on simultaneously for an instant as the code runs through each bit. In other words, might it turn on a lower numbered relay before it turns off it's interlocked partners which may be in higher order bits?

Mike

vzorglub, 2019-08-29:
Does the interlock ensures that relays are turned OFF **before** the other ones are turned ON
That is very important for the safety of some devices.
And if that's the case what is the delay beween the ON event and the OFF event?

arendst, 2019-08-29:
Looking at the code I once wrote ;-)
```
    if (Settings.flag.interlock && !interlock_mutex) {  // Clear all but masked relay in interlock group
      interlock_mutex = true;
      for (uint32_t i = 0; i < MAX_INTERLOCKS; i++) {
        if (Settings.interlock[i] & mask) {             // Find interlock group
          for (uint32_t j = 0; j < devices_present; j++) {
            power_t imask = 1 << j;
            if ((Settings.interlock[i] & imask) && (power & imask) && (mask != imask)) {
              ExecuteCommandPower(j +1, POWER_OFF, SRC_IGNORE);
              delay(50);                                // Add some delay to make sure never have more than one relay on
            }
          }
          break;                                        // An interlocked relay is only present in one group so quit
        }
      }
      interlock_mutex = false;
    }
```
I think only one relay in an interlock group is on at once. It even has a delay of 50mS.

closed 2019-09-01 by ascillato2