arendst/Tasmota #6316: Multiple MCP230xx modules

ridikuel, 2019-08-29:
Hello, I am currently working on extending the support of the MCP230xx modules with [my Fork](https://github.com/ridikuel/Sonoff-Tasmota-stepper-MultiMCP). My current requirement is 2 modules on the same Tasmota, but I'd like to design for more.

There are some decisions to take. So, If some of you could kindly share ideas on that.
First are
- [How many possible Modules?](https://github.com/ridikuel/Sonoff-Tasmota-stepper-MultiMCP/issues/8)
- [Instantiate general Settings for each separate module?](https://github.com/ridikuel/Sonoff-Tasmota-stepper-MultiMCP/issues/7)

Maybe there also would be some people interested in this topics that might help me.

**Have you looked for this feature in other issues and in the wiki?**
Yes. ;-)

**Is your feature request related to a problem? Please describe.**
The Problem is: Too few IOs ;-)

**Describe the solution you'd like**
Mainly modify xsns_29_mcp230xx.ino to multiply the working variables to represent more than one module. Doing this with arrays.

**Describe alternatives you've considered**
- Setting up a new xsns_<foo>_mcp230xx_2nd and having a second sensor to talk to. But this will add tons of code and waste flash space in device. Also this is limited to 1 module per new xsns.

**Additional context**
Please use issues and wiki sections in my fork repo.

**(Please, remember to close the issue when the problem has been addressed)**
Maybe you can leave this issue open for some days. Thank you.

arendst, 2019-08-29:
I won't ask why you need more than the already provided 16 GPIO's by the current driver.

I just say it's enough. If you need more then add a second esp8266 device. I didn't write the code but a quick glance shows it needs every 50mSeconds attention servicing something. I do not like such an approach as it stalls other services.

ascillato2, 2019-09-01:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-09-01 by ascillato2

ridikuel, 2019-09-03:
Thank you for that valid points. I think I will keep an eye on the thing with the frequent calls. And I will limit it to two MCPs.
Maybe at the end it will turn out, that using a 2nd ESP will be the thing. But for now, I think I will try further.

nccchirag, 2023-03-29:
@ridikuel are you still developing the feature?

nccchirag, 2023-03-29:
Thank you for the prompt response @arendst .. I have successfully integrated 1 Nos MCP23017 into the project however, I also have the requirement of using 2 Nos MCP into the project. Currently the only alternative I have is to use 2nos PCF8574 instead of the second MCP ... for using PCF I need to modify the PCB artwork so was trying to find if anyone has implemented the same (support for dual MCP23017)

I can try to modify the source however; I need some help ... 
If you could guide me, how can I add custom settings to the TSettings? example first MCP23017 settings are stored at

Mcp230xxCfg   mcp230xx_config[16];       // 6F6
uint8_t       mcp230xx_int_prio;         // 716
uint16_t      mcp230xx_int_timer;        // 718

For the second MCP I need to define the additional 19 bytes of storage. My question is how can I add additional settings without disturbing the current layout / memory map? Please guide

the max allowed size is 4096

**Intent**
I'm building 1nos hardware using ESP32S3 which can control 28 relays powered using Tasmota firmware.

thank you

arendst, 2023-03-29:
Did you read the new mode2 posibility/functionality? It supports several MCP23xxx chips so no need to change any code. Just define your template data as shown in the examples.

ridikuel, 2023-03-29:
> @ridikuel are you still developing the feature?

Sorry, nope. Didn't come together for me. I took another solution. Would have needed to mess a lot in the conf structure.

nccchirag, 2023-03-29:
> > @ridikuel are you still developing the feature?
> 
> Sorry, nope. Didn't come together for me. I took another solution. Would have needed to mess a lot in the conf structure.

No problem ... thank you for the response
Btw, do check mode2 suggested by @arendst .. it seems to work well .. 
I have tested it with 2 MCP23017 connected via I2C bus on ESP32S3