arendst/Tasmota #6317: Dimmer integration with Home Assistant broken in 6.6.0

bigcookie, 2019-08-29:
I updated from 6.5.0 to 6.6.0 and the light control for a Tuya dimmer was broken.
HomeAssistant: 0.98.0 (tried also 0.97.1)

Original successful setup:
I used auto-discovery from Tasmota (SetOption19 1), which worked like a charme. In HomeAssistant a light entity appeard, which I could use with the light card and turn on/off plus dim the light. State was reported correctly.

Issue:
After updating to Tasmota 6.6.0, the status report didnt work anymore: I can set the brightness via HomeAssistant and Tasmota receives it, but the status is not anymore reflected in HA. Also I could only turn the light on via HA button, but it never toggles to "on", thus I cannot turn it off.
It seems that the status is not properly updated/reported to HA. 
Nevertheless when checking MQTT, the ID/tele/STATE (I switch topic and prefix!) was still sent as it seems - checked with MQTTSPY.
A new switch was discovered in home assistant, which let me turn the light on/off and also provided correct state -> but dimming would be missing.

I noticed that my dimmer with Tasmota 6.6.0 was suddenly reporting 6 relays/switches, and in HA a light component PLUS a switch component for each relay appeared -> but there is only one light to switch. It reported:
light.schlafzimmer_6
switch.schlafzimmer
switch.schlafzimmer_2
switch.schlafzimmer_3
switch.schlafzimmer_4
switch.schlafzimmer_5
whereby switch.schlafzimmer was toggling the light and the light.schlafzimmer_6 component allowed to dim, but without status feedback.

Reverting to 6.5.0 only one component was discovered:
light.schlafzimmer
no numbers and the webUI only showed one "toggle" button instead of 6 buttons. light.component in HA worked, I can toggle and dim the light while seeing correct state.

Anymore information needed: please let me know :-)

shantur, 2019-08-29:
@bigcookie : If you are using `Tuya Dimmer` module, then it might be worth to have a look at https://github.com/arendst/Sonoff-Tasmota/wiki/Tuya-Dimmer-and-Tuya-Switches-(based-on-serial-MCU)
There are lots of fixes / changes related to Tuya Dimmer (based on MCU) in latest.

bigcookie, 2019-08-29:
@shantur : thanks for the insights - and yes, I am using a Tuya module. The dimming itself works fine - via web frontend. Thus the hardware configuration is correct (during the first investigations I mapped all GPIOs out etc).
The new functions to adapt the dimming range (I was missing that) as well as some ID settings is great, and I will try it out.

But the problem I am having is with the integration in HA -the dimmer works fine as stated. I assume that a few things have been changed and the auto discovery might have not followed up.
I do have a one channel dimmer, thus I would expect it to be discovered as one dimmable light in HA as prior. With the latest version, actually 1 light (dimmable) and 5 switches are discovered and the state is only identified correctly for the switches, but not the "light" component.

Again, to stretch this out: the dimmer works fine via the web UI. The issue seems to be in the HA integration.
-> nevertheless I can still have made a mistake of course, but I used different HA versions to check and at the end downgrading Tasmota to 6.5.0 solved the auto-discovery and the integration. 

If anybody knows how to provide better debug logs, I am happy to do so, but I might need a bit of help with the steps.

bigcookie, 2019-08-29:
Based on the new guide, I can disable the non existent relays which are shown in the web UI using SetOption41 0. I assume this will make the other faulty switches disappear in the HA auto-discovery.
Not sure if this will help. Will try later. If it helps, then something seems to be wrong in the standard setup: Flashing Tasmota 6.6.0 assumed 6 relays instead of one channel (web UI presented 6) and the auto-discovery potentially pushed wrong values...

I will try again later to use 6.6.0, but I still dont expect the fine tuned dimmer configuration to be fixing the auto-discovery... I will of course report my results.

ascillato, 2019-08-29:
Try the latest version that have all the fixes: 6.6.0.9

bigcookie, 2019-08-29:
Hi, I tested 6.6.0.9 - this one works. I didnt see anything noticeable in the changelog, but it works now. Issue can be closed.

closed 2019-08-29 by bigcookie