arendst/Tasmota #6318: tasmota 6.6.0 on sonoff s31 randomly rebooting on its own

evanrich, 2019-08-30:
### BUG DESCRIPTION
After replacing my router (PF Sense) with a Palo Alto networks PA-2050, I noticed my sonoff S31s (I have like 8 of them) all started rebooting non stop.  They reboot, come up for ~3-5 minutes, then reboot again.   I tried running status 0, it doesn't say anything.  All swiches were working fine, but now give exceptions (see below).   They are all running tasmota 6.6.0 (sonoff.bin)

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Device used (i.e. Sonoff Basic) : S31
- [ ] Tasmota binary firmware version number used : ____ / (pre-compiled or self-compiled ?)
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
I can't get it to output any output for this command

```

```
CONSOLE OUTPUT HERE:
00:00:00 CFG: Loaded from flash at F8, Count 840
00:00:00 APP: Set Baudrate to 4800
00:00:00 Project sonoff Stereo Rack Version 6.6.0(release-sonoff)-2_3_0
00:00:00 WIF: Connecting to AP1 IOT-WIFI in mode 11N as sonoffS31-3-3815...
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on sonoffS31-3-3815 with IP address 192.168.3.17
00:00:04 RSL: tele/sonoffS31-3/SENSOR = {"Time":"1970-01-01T00:00:04","ENERGY":{"TotalStartTime":"2019-02-20T09:19:30","Total":60.180,"Yesterday":0.402,"Today":0.574,"Power":3,"ApparentPower":5,"ReactivePower":5,"Factor":0.57,"Voltage":123,"Current":0.044}}
21:00:57 MQT: Attempting connection...
21:00:57 MQT: Connected
21:00:57 MQT: tele/sonoffS31-3/LWT = Online (retained)
21:00:57 MQT: cmnd/sonoffS31-3/POWER = 
21:01:12 RSL: tele/sonoffS31-3/INFO1 = {"Module":"Sonoff S31","Version":"6.6.0(release-sonoff)","FallbackTopic":"cmnd/DVES_EAAEE7_fb/","GroupTopic":"sonoffs"}
21:01:17 RSL: tele/sonoffS31-3/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoffS31-3-3815","IPAddress":"192.168.3.17"}
21:01:22 RSL: tele/sonoffS31-3/INFO3 = {"RestartReason":"Fatal exception:28 flag:2 (EXCEPTION) epc1:0x4022c7ee epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000004 depc:0x00000000"}
21:01:27 RSL: stat/sonoffS31-3/RESULT = {"POWER":"ON"}
21:01:32 RSL: stat/sonoffS31-3/POWER = ON

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

evanrich, 2019-08-30:
FYI, they all seem to be doing this, although not at the exact same time.  I've pulled them all from the wall, and then plugged them back in.  Still randomly reboot.

curzon01, 2019-08-30:
Try using a binary with another core, e.g. [2.4.2](http://thehackbox.org/tasmota/) or [2.5.2](http://thehackbox.org/tasmota/020502/).

As this might be a wifi router issue please address this to [Support's Community](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md#supports-community).
The community channels are a better and more dynamic for helping you.
Github issues are mainly used for Tasmota software feature requests and bug reporting. 

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

<hr>

#### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and links if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

arendst, 2019-08-30:
From your reboot message it seems you receive an UDP message (which might well be from your new router) which does contain data leading to an exception 28. This could mean that Tasmota receives a number where a string is expected or the other way round.

For debugging purposes try disabling any emulation on one device and see what happens.

ascillato2, 2019-09-01:
@evanrich Hi, any news on this?

ascillato2, 2019-09-02:
Closing this issue as there is no feedback. If you need further assistant, please ask in the Tasmota support chat. Thanks.

closed 2019-09-02 by ascillato2