arendst/Tasmota #6320: PN532 doesnt work with 6.6.0.9

Jason2866, 2019-08-30:
### BUG DESCRIPTION

No results receiving when rfid chip attached to reader

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : wemos mini D1
- [x] Tasmota binary firmware version number used : 6.6.0.9 core pre2.6 latest self-compiled
- [x] Development IDE - Compiler / Upload tools used : VSC / OTA
- [x] Provide the output of command ``status 0`` :
```

13:10:37 CMD: Status 0
13:10:37 SRC: WebConsole from 192.168.2.137
13:10:37 MQT: stat/sonoff-4514CFCF/STATUS = {"Status":{"Module":18,"FriendlyName":["MGS"],"Topic":"sonoff-4514CFCF","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:10:37 MQT: stat/sonoff-4514CFCF/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.2.20/data/firmwares/sonoff-minimal.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:56","StartupUTC":"2019-08-30T11:08:41","Sleep":50,"CfgHolder":4617,"BootCount":105,"SaveCount":184,"SaveAddress":"F9000"}}
13:10:37 MQT: stat/sonoff-4514CFCF/STATUS2 = {"StatusFWR":{"Version":"6.6.0.9(sonoff)","BuildDateTime":"2019-08-30T13:05:08","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
13:10:37 MQT: stat/sonoff-4514CFCF/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jason_Home_WLAN",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000005AFF000000170700","00100000"]}}
13:10:37 MQT: stat/sonoff-4514CFCF/STATUS4 = {"StatusMEM":{"ProgramSize":493,"Free":508,"Heap":25,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8F8AE387","040003A1","00441497","001022C0","00000000"],"Drivers":"1,2,3,4,5,6,8,9,10,20","Sensors":"1,3,4,5,6,9,10,15,19,35,40"}}
13:10:37 MQT: stat/sonoff-4514CFCF/STATUS5 = {"StatusNET":{"Hostname":"sonoff-4514CFCF-5327","IPAddress":"192.168.2.130","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"B4:E6:2D:45:14:CF","Webserver":2,"WifiConfig":5}}
13:10:37 MQT: stat/sonoff-4514CFCF/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.114","MqttPort":1883,"MqttClientMask":"sonoff-4514CFCF","MqttClient":"sonoff-4514CFCF","MqttUser":"openhabian","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
13:10:37 MQT: stat/sonoff-4514CFCF/STATUS7 = {"StatusTIM":{"UTC":"Fri Aug 30 11:10:37 2019","Local":"Fri Aug 30 13:10:37 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"06:26","Sunset":"20:00"}}
13:10:37 MQT: stat/sonoff-4514CFCF/STATUS10 = {"StatusSNS":{"Time":"2019-08-30T13:10:37","Epoch":1567163437,"MGS":{"NH3":1.02,"CO":0.39,"NO2":0.90,"C3H8":1049.51,"C4H10":668.34,"CH4":0.08,"H2":0.02,"C2H5OH":0.07}}}
13:10:37 MQT: stat/sonoff-4514CFCF/STATUS11 = {"StatusSTS":{"Time":"2019-08-30T13:10:37","Epoch":1567163437,"Uptime":"0T00:01:56","UptimeSec":116,"Vcc":2.984,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":9,"RSSI":94,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```

### TO REPRODUCE
Connect PN532 

### EXPECTED BEHAVIOUR
getting RFID readings from PN532

arendst, 2019-08-30:
Verified. Using TasmotaSerial 2.3.1 it's working OK.

Investigating...

Jason2866, 2019-08-30:
Thx, does work again.

closed 2019-08-30 by Jason2866