arendst/Tasmota #6321: Is there any 868 MHz radio support (TX)?

maximevince, 2019-08-30:
Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

**Have you looked for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
I'd like to be able to transmit 868 MHz packets, like I can at 433 using the Sonoff RF bridge with Tasmota firmware. 

**Describe the solution you'd like**
If there are any supported devices, where can I find them. If not, I am willing to integrate one, e.g. a CC1101 over SPI. Some pointers to get me started on how to integrate / interface with Tasmota would be appreciated.

**Describe alternatives you've considered**
Not sure. I'd like to integrate it with one of my existing ESP-based Tasmota devices (RF bridge, Sonoff S26, NodeMCU ...)

Many thanks!

meingraham, 2019-08-30:
@maximevince 

Use this as a coding guide - https://github.com/arendst/Sonoff-Tasmota/wiki/Sensor-API

You can use the other sensor drivers (xsns---.ino) as examples. Just adapt the CC1101 protocols per the module's data sheet.

Jason2866, 2019-08-30:
@maximevince  Take a look in Gemu Tasmota fork. He has already a driver for cc1101
https://github.com/gemu2015/Sonoff-Tasmota/blob/universal7/sonoff/xsns_97_moritz.ino

closed 2019-09-01 by ascillato2