arendst/Tasmota #6322: Dimming for ifan03?

mcaskey, 2019-08-30:
Is it possible to have dimming for the light module in your fan?

meingraham, 2019-08-30:
Insufficient information. Our ability to provide assistance is greatly hampered since you did not take a few minutes to complete the  [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) with the requested information. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

As instructed in the template when you create a new issue, this kind of question should be directed to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). Please close this issue. In any case, a moderator will perform this maintenance issue momentarily.

The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

mcaskey, 2019-08-30:
Ok thanks.

mkormendy, 2019-08-31:
The iFan03 and 02 models do not support light dimming. I can confirm this because the hardware inside only has a relay for on or off.

closed 2019-09-01 by ascillato2