arendst/Tasmota #6323: Homeassistant: add device_class in MQTT discovery for more sensors

rejong, 2019-08-30:
Home Assistant(HA) uses device_class to show the correct icon in the UI and I use it when HA publishes a state_change for a sensor to select the correct collection in MongoDB.

It is already partly implemented.

For humidity & pressure in: **xdrv_12_home_assistant.ino**
```
const char HASS_DISCOVER_SENSOR_HUM[] PROGMEM =
  ",\"unit_of_meas\":\"%%\","                         // %
  "\"val_tpl\":\"{{value_json['%s'].Humidity}}\","    // "SI7021-14":{"Temperature":null,"Humidity":null} -> {{ value_json['SI7021-14'].Humidity }}
  "\"dev_cla\":\"humidity\"";                         // humidity

const char HASS_DISCOVER_SENSOR_PRESS[] PROGMEM =
  ",\"unit_of_meas\":\"%s\","                         // PressureUnit() setting
  "\"val_tpl\":\"{{value_json['%s'].Pressure}}\","    // "BME280":{"Temperature":19.7,"Humidity":27.8,"Pressure":990.1} -> {{ value_json['BME280'].Pressure }}
  "\"dev_cla\":\"pressure\"";                         // pressure
```
(dev_cla = device_class)

Homeassistant supports the following device classes
- None: Generic sensor. This is the default and doesn’t need to be set.
- battery: Percentage of battery that is left.
- humidity: Percentage of humidity in the air.
- illuminance: The current light level in lx or lm.
- signal_strength: Signal strength in dB or dBm.
- temperature: Temperature in °C or °F.
- power: Power in W or kW.
- pressure: Pressure in hPa or mbar.
- timestamp: Datetime object or timestamp string.
[source](https://www.home-assistant.io/components/sensor/)

With humidity & pressure already implemented the only other relevant for Tasmota imho are: 
- power
- temperature

I presume these would be the places to make changes in **xdrv_12_home_assistant.ino**:
add for temperature:
`"\"dev_cla\":\"temperature\""`
to
```
const char HASS_DISCOVER_SENSOR_TEMP[] PROGMEM =
  ",\"unit_of_meas\":\"°%c\","                        // °C / °F
  "\"val_tpl\":\"{{value_json['%s'].Temperature}}\""; // "SI7021-14":{"Temperature":null,"Humidity":null} -> {{ value_json['SI7021-14'].Temperature }}
```
and for power add
`"\"dev_cla\":\"power\""`
to
```
const char HASS_DISCOVER_SENSOR_WATT[] PROGMEM =
  ",\"unit_of_meas\":\"W\","                          // W
  "\"val_tpl\":\"{{value_json['%s'].%s}}\""; // "ENERGY":{"TotalStartTime":null,"Total":null,"Yesterday":null,"Today":null,"Power":null,"ApparentPower":null,"ReactivePower":null,"Factor":null,"Voltage":null,"Current":null} -> {{ value_json['ENERGY'].POWER }}
```

Sorry, I'm unable to make a pull request for this myself, but I hope to have provided enough detail.

Thanks in advance!

pablozg, 2019-08-31:
Try to change your code with this new definitions and compile it:

`
const char HASS_DISCOVER_SENSOR_TEMP[] PROGMEM =
  ",\"unit_of_meas\":\"°%c\","                        // °C / °F
  "\"val_tpl\":\"{{value_json['%s'].Temperature}}\"," // "SI7021-14":{"Temperature":null,"Humidity":null} -> {{ value_json['SI7021-14'].Temperature }}
  "\"dev_cla\":\"temperature\"";                      // temperature
`

`
const char HASS_DISCOVER_SENSOR_WATT[] PROGMEM =
  ",\"unit_of_meas\":\"W\","                          // W
  "\"val_tpl\":\"{{value_json['%s'].%s}}\"," // "ENERGY":{"TotalStartTime":null,"Total":null,"Yesterday":null,"Today":null,"Power":null,"ApparentPower":null,"ReactivePower":null,"Factor":null,"Voltage":null,"Current":null} -> {{ value_json['ENERGY'].POWER }}
  "\"dev_cla\":\"power\"";                            // power
`

If works like you expected, I will open a PR for you.

ascillato2, 2019-09-01:
Hi, any news on this?

rejong, 2019-09-01:
I've never build Tasmota before, but I'll try. Might take me a couple of spare hours next week:-)

DeviousPenguin, 2019-09-04:
It would be nice to add units to the distance sensor also, next time I try building Tasmota I can try out these changes and see if they work OK. :)

effelle, 2019-09-09:
@ascillato2 I'm working on a PR right now. 

I've Included:
- Illuminance Class (mainly for Analog LDR), 
- Temperature Class (both Analog and Digital sensors)
- Power Class for kW/h, W, V, A

I have not added `signal_strength`, `battery` and `timestamp`, since I can't see a real use here. 
Normally SENSOR doesn't show that kind of information (correct me if I´m wrong Adrian). 

@DeviousPenguin `units` and `distance` are not selectable classes on Home Assistant, can't add them sorry.

pablozg, 2019-09-09:
> @ascillato2 I'm working on a PR right now.

Thanks, I offered me to do it, but I don't use Home Assistant, so is better if you can made it and test it.

effelle, 2019-09-09:
Sure @pablozg and you deserved your credits on my PR.

pablozg, 2019-09-09:
> Sure @pablozg and you deserved your credits on my PR.

I'm glad to hear this, but It is not necessary, I only intend to help as much as I can, and in this case you made all the needed. 👍 😃